luc-github/ESP3D #217: Problem with ESP-12E and MKS-GEN v1.2

onekk, 2018-04-10:
The documentation lacks of my board, but it is has RAMPS compatible AUX connectors.

In my board the pins are clearly marked on the rear silkscreen and I've to swap the RX -TX pin, 
The RX pin is near the GND connection and the TX is obviously those at the end of the row.

Plus a little hints, following the schematics of the Adafruit Huzza Board the RX pin of the ESP-12E is correctly working if you connect a 1N4148 diode with the cathode toward the printer Boards. following thid explanation on adafruit forum

`also called a diode clamping level shifter - there's an internal pullup on the ESPRX pin. when the the input is 5V volts, the diode is reverse biased, and the ESPRX is pulled up to 3V. When the input level is 0 volts, the diode is forward biased, and ESPRX is at 0.7V or so, which is below the Vlow and registers as low `

with this tweak there is no need of two resistors and the circuit is even more simple.

Regards

Carlo D.

luc-github, 2018-04-10:
Thank you for sharing, fill free to add a page in wiki for your mms 1.2

onekk, 2018-04-16:
I read only today the comment, I will try to add some information about MKS gen v1.2 asap.

Thank you and Regards

Carlo D.

luc-github, 2018-04-16:
Thank you ^_^