luc-github/ESP3D #219: After I connect through Wi-Fi and enter the password it tells me my index.HTML.gz file is outdated

PETEIOT, 2018-04-24:
After I connect through Wi-Fi and enter the password it tells me my index.HTML.gz file is outdated

luc-github, 2018-04-24:
what version do you use ?

PETEIOT, 2018-04-25:
Are you asking what version of arduino,  if so I have the most recent version. I  do have a screenshot but not sure how to send it.

PETEIOT, 2018-04-25:
I was using the mobile browser I switch the desktop and it me add the screenshot 

[image]

luc-github, 2018-04-25:
index.html is not outdated is missing as stated in screenshot

check point 4 of readme which explain that you must upload files from data directory, you may noticed screen is same as yours
[image]

luc-github, 2018-04-26:
I guess this issue can be closed

closed 2018-04-26 by luc-github

PETEIOT, 2018-04-26:
Sorry just got around to it and that worked. Thanks for your help!