luc-github/ESP3D #221: Creality CR-10 / Ender support?

roger-, 2018-05-07:
Is this printer supported?

luc-github, 2018-05-08:
I do not know the technical specs of this printer - seems also several version of main board
if you are able to find/connect rx/tx + 3.v or 5v + gnd should be ok as it seems Marlin can be flashed 
for original FW it may work also but I do not know what it is

roger-, 2018-05-09:
Not sure about the tech specs either but it looks like this: https://www.aliexpress.com/item/3D-Printer-Part-Mainboard-Screen-for-Creality-Cr-10-Cr-10s-3D-Printer-Mainboard-Control-Broad/32827096365.html

It does use Marlin, but I’m not sure how to get to a free UART.

jRam84, 2018-05-13:
That Crealty board has 2 rx/tx pins, Pin 8 D0 RXD, Pin 9 D1 TXD and Pin 10 D2 BTN_EN2, Pin 11 D3 BTN_EN1. The last 2 are for the LCD encoder so if the first 2 are not used then those are the ones you need to use.

roger-, 2018-05-13:
Thanks, good to hear!

Where are you getting those pin names from? Are they all in the EXP header (that goes to the LCD)?

The rear of the LCD board has two unconnected headers (EXP1 and EXP2). Perhaps those provide access to the TX/RX pins?

jRam84, 2018-05-13:
The pin names are in the Marlin firmware, specifically on the file pins_MELZI_CREALTY.h. The Marlin devs did an amazing job at mapping the pins in that closed source board. 

For D0 and D1 I do not think Crealty provided physical connectors so the only way is probably by soldering directly to the Arduino. I tried tracing them with a Multimeter and they are not in the EXP header, only D2 and D3 are.

Here is the link for the Atmega1284 pinout used in the Crealty board [https://camo.githubusercontent.com/68d25863f87140bbb2d4dbbad407991ee064dbae/68747470733a2f2f692e696d6775722e636f6d2f4b51574c4a75322e6a7067](url)

luc-github, 2018-05-21:
@jRam84 you can update wiki with the 2 creality boards you modified - 
for picture to upload - upload them here I will update repo accordigly for wiki or you can just link pictures from this issue to wiki - up to you ^_^ 

Let me know if you need any assistance for the wiki

jRam84, 2018-05-21:
Page added for the Sanguino based CR-10 and Ender 3. I will add another page for the CR-10S (Ramps based board) later

https://github.com/luc-github/ESP3D/wiki/Creality-CR-10-Ender-3

luc-github, 2018-05-21:
Excellent ^_^  thanks a lot for this valuable contribution 👍

luc-github, 2018-06-03:
@roger- I think you have all info now, I think we can close issue 😸 

@jRam84 - thanks again - feel free to complete wiki anytime ^_^

closed 2018-06-03 by luc-github