luc-github/ESP3D #222: connecting to repetier and ramps

ekaggrat, 2018-06-04:
i can get it to work out of the box with marlin by just plugin it in on aux1 on the ramps. no changes to firmware. but on repetier it just wont connect and keeps reseting the board when i try to connect via wifi to the esp. i am using a nodemcu

luc-github, 2018-06-04:
Some Firmware/board  do a board reset when serial connection is established - this not related to ESP3D

luc-github, 2018-06-04:
I have a due/radds/repetier and davinci (due like) with repetier without issue

Did you have issue on same board or 2 boards: one with repetier and one with marlin ?

ekaggrat, 2018-06-04:
Same board same setup. Also it just keeps resetting but doesn’t connect to the printer. In marlin it works effortlessly

luc-github, 2018-06-04:
Well in that case it is something in repetier Fw with avr board

luc-github, 2018-06-07:
Did you find the solution ?

luc-github, 2018-06-12:
I have a mega/ramps with repetier 1.0 and I do not have any issue

ekaggrat, 2018-06-12:
I suspect the 5v on the ramps is running low and the moment the esp connects to wifi it resets it

luc-github, 2018-06-13:
May be , my system is connected by 12V on ramps - and tried several connection without reset

ekaggrat, 2018-06-13:
so u are using a voltage convertor to change 12v to 5 from the ramps separately? I can try that!

luc-github, 2018-06-13:
no - I just have my ramps/mega connect to 12V and my ESP connected to RX/TX of Aux 1 on ramps and 5V/G come from EXP1/EXP2  connectors

ekaggrat, 2018-06-13:
thats how i have connected... 5v comes from the ramps board on aux1 and rtx
tx also on aux1 .. okay so i am now sure it is due a fault on the 5v of the
arduino mega somehow. i am using a nodemcu .. which esp board are you using?

luc-github, 2018-06-13:
my 5V/Gnd does not come from AUX1 but EXP1/EXP2

I did not tried nodeMCU yet, can try later

it is a prototype board not yet on market - but soon I think

luc-github, 2018-06-17:
I have tried with node mcu without issue

luc-github, 2018-06-30:
Can we close issue as it is not related to ESP3D actually

ekaggrat, 2018-07-07:
okay i finally got time to check it using a external power for the nodemcu and now it works. sorry to waste your time because of my flaky arduino. One last thing that remains. I am not able to get the sd card listing on the web interface although i refresh it several times. Is there something else i need to do to get it working? thanks

luc-github, 2018-07-08:
Just set repetier as target firmware
What is the output for M20 command on webui ?
What the output of  sd content on printer screen?

ekaggrat, 2018-07-08:
yes i set it repetier and correct baud rate and all motor commands work. but on M20 there is no response. the sd card shows the file list on the lcd . But on the webui the command has no response!

luc-github, 2018-07-08:
What the output of [ESP800] and M115

luc-github, 2018-07-08:
Can you connect using putty and data port (port 8888, raw connection) or and Send M20 and M115
,or direct serial to your ramps, and give the output ?

luc-github, 2018-07-08:
ok i can reproduce both issues
M115 and M20. Weird was ok before and also M20 is working on 2.0
M115 is because repetier send ok then send answer...
Will work on it today...

luc-github, 2018-07-08:
I have created 2 issues for better tracking
https://github.com/luc-github/ESP3D/issues/242
https://github.com/luc-github/ESP3D/issues/241

So I close this topic which is not same

closed 2018-07-08 by luc-github