luc-github/ESP3D #223: [question] Serial ports

DotBreno, 2018-06-07:
Can I connect my ESP8266 on any serial port or it has to be AUX1(RXD0,TXD0)?

I have an ESP8266 nodemcu module like [this one ](https://www.alibaba.com/product-detail/ESP8266-NodeMCU-V3-LUA-CP2102-ESP_60754334824.html?spm=a2700.7724838.2017115.1.21ac62da6IoOWH&s=p) and I would like to connect to a [Rumba board](https://www.alibaba.com/product-detail/High-Quality-3D-Rumba-Board-with_60457481904.html?spm=a2700.7724838.2017115.31.64314d08gMNBJO&s=p). The problem is the only serial pins available from the atmega2560 are the TXD3 and RXD3 wired at the EXT3 connector on the board. Do I need to change anything else to connect the ESP8266 to serial3(TXD3,RXD3) port or it will work if I just wire it?
Sorry for my poor english and electronics knowledge.
Also I would like to thank you for making your project open source. This is really a great work!

luc-github, 2018-06-07:
Any serial port is valid but you need to be sure your printer FW use this port to communicate, usually need to changes some settings in Marlin/Repetier and  reflash

DotBreno, 2018-06-07:
Thanks so much.

I am using Marlin. What exactly should I change to make sure the serial3 is being used to communicate? Should I check something on the pins configuration file inside the firmware? If the file says nothing about that specific set of pins that means I can just use them?

luc-github, 2018-06-07:
I think the documentatio is pretty clear 
http://marlinfw.org/docs/configuration/configuration.html

Some people also use #define BLUETOOTH as a trick to duplicate serial and use the serial used by  bluetooth for ESP3D, to keep main serial available

luc-github, 2018-06-12:
@DotBreno  did you managed to connect to Serial3 ?

DotBreno, 2018-06-13:
Yes. Thanks so much! I tried changing the main serial port to serial3 and it worked with the ESP8266 but then I couldnt connect trough usb using repetier or cura. So I changed it back to serial0 and did serial3.begin(115200). Not sure if its the right way but it seems to work so far.
Sorry for the delayed reply

luc-github, 2018-06-13:
yes if you change main main serial to Serial 3 it is normal you cannot access using usb after

just doing Serial3.begin will allow you to send command from ESP to printer but you must not get any answer - like no temperature - not position - no feedback if send M115 in terminal and no files in sd panel
I am correct ?

DotBreno, 2018-06-21:
> just doing Serial3.begin will allow you to send command from ESP to printer but you must not get any answer - like no temperature - not position - no feedback if send M115 in terminal and no files in sd panel
I am correct ?

Actually, you are correct. What is the best way for me to use both USB and ESP8266 to communicate (not necessarily simultaneously)? Is there a way to "mirror" all data from serial0 to serial3? Or should I just wire the ESP8266 to serial0 (same pins as usb)?

luc-github, 2018-06-22:
Repetier allows to mirror main serial to another one when enabling BLUETOOTH  
On Marlin seems not this feature but I am not sure

I am not expert of Marlin sorry, it is better to ask on Marlin forum I think 

On my side I use ESP on Serial 0 as I only print from SD and never connect my USB

DotBreno, 2018-06-25:
Thanks so much @luc-github. You really helped me more than enough so far.

luc-github, 2018-06-30:
I guess issue is closed now

closed 2018-06-30 by luc-github