luc-github/ESP3D #224: How can I faster the  transmission speed of the sd-card (upload) without using the wifi-sd.

wrh1995, 2018-06-11:
Thank a lot. I wanna to make the speed approach 1M/s . Could you give me some advice. 10K/s....now

luc-github, 2018-06-11:
the only solution is to connect your SD reader directly to ESP sharing it with your main board and not use the upload that go thrue the printer board FW

This is what Azteeg do http://www.panucatt.com/azteeg_X5_mini_reprap_3d_printer_controller_p/ax5mini.htm

wrh1995, 2018-06-12:
That means the ESP doesn't work...I wanna  use the WiFi to upload the gcode. Shall I use the DMA or IIS make this project into reality.The reference are as follows:
1. https://github.com/lhartmann/esp8266_reprap

luc-github, 2018-06-12:
ESP works very well, upload are fast if ESP is connected to SD card by SPI like I said , if it is a daughter board, or like in project  you linked, if it is a main board for 3D printer

But ESP by serial connected to 3D printer board => upload are slow like hell - no improvement using the M28/M29 upload protocol

Per my experience ESP8266 as main mcu for 3D printer is not best choice because it has only one core, or you need to develop a very optimized FW like GRBL 

For 3D printer, ESP32 is a better choice, enough power and 2 cores,  and have already marlin support

luc-github, 2018-06-12:
I close issue as not related to ESP3D

closed 2018-06-12 by luc-github

asmallbee, 2018-06-22:
Hi, I am going to build a new RAMPS-like PCB with SD Card built-in as well as the ESP8266. Would you please advise how should I connect the SD Card on the ESP8266? The default SPI with a CS? Can you please share your schematic when you were testing upload Gcode file to the SD card directly connected to ESP8266? Thank a lot!

luc-github, 2018-06-23:
I do not have any schematics - I am not electronician - connecting ESP to a SD card is following normal way: cs ck miso mosi 
EP8266 provide code for this : https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer/examples/SDWebServer

asmallbee, 2018-06-23:
Thanks! Is that already implemented in the current version of ESP3D? How can I try upload gcode to the SD card connected to ESP8266 with ESP3D?

luc-github, 2018-06-23:
No it is not, because once on SD it is useless ESP3D act as host and this is for ESP3D 2.x I a currently rewriting and will need esp32 as esp8266 has not enough power for this

Connected like this SD is not accible from Printer FW

asmallbee, 2018-06-23:
I will be patient to wait for ESP3D 2.x with ESP32. Thank you very much for your efforts.

probonopd, 2018-09-08:
> will need esp32 as esp8266 has not enough power for this

I have been using https://github.com/probonopd/WirelessPrinting to output GCODE from a SD card attached to the ESP8266 to the printer, and have not experienced "not enough power". So maybe the part that takes the GCODE and sends it to the printer could be taken from my code?