luc-github/ESP3D #229: SPIFFS writing broken esp-01

spider0404, 2018-06-27:
There is a new batch of esp-01 whit different chip, whit them you can't write on SPIFFS, can't upload web-ui. You can upload web-ui with arduino ide but when you try to write on it put macro or something else. Web-ui is gone blank page or page saying you are missing index......gz

There is a patch so everything is working as expected.

https://github.com/esp8266/Arduino/issues/4061#issuecomment-368273656

This is not ESP3D bug but without this patch ESP3D doesn't work :)

This is written here so somebody else doesn't lose time.

luc-github, 2018-06-30:
Thanks for pointing out -yes  the puya chip is not yet well supported by default

probonopd, 2018-09-08:
Here is a patch: https://github.com/esp8266/Arduino/issues/4061

closed 2018-12-09 by luc-github

luc-github, 2019-03-13:
To enable the patch , search for esp.h in core, there is the switch to enable the patch with latest git of the core
```
#ifndef PUYA_SUPPORT
  #define PUYA_SUPPORT 0
#endif
```
Change to
```
#ifndef PUYA_SUPPORT
  #define PUYA_SUPPORT 1
#endif
```

luc-github, 2019-05-09:
another solution:
Create platform.local.txt  in esp8266 core directory and add  these lines:
```
compiler.c.extra_flags= -DPUYA_SUPPORT=1
compiler.cpp.extra_flags={compiler.c.extra_flags}
```
like explained here https://github.com/esp8266/Arduino/pull/5504#issuecomment-490111960

luc-github, 2020-01-04:
Now patch is enabled in core by default