luc-github/ESP3D #230: Help

hackintoshlover, 2018-06-28:
How do I save jog commands manually by moving respective axes and then executing them
So what it does is when I click an axis it moves and when I press record button it saves the recent gcode
When stop is pressed gcode is saved as a file and then sent

luc-github, 2018-06-28:
there is no "record" neither "excecute last command" feature so far

but you can use a macro to execute a gcode file you saved on SPIFFS or on SD

hackintoshlover, 2018-06-28:
How do I save file on spiffs either
Btw is there any way I could add record and execute last command

luc-github, 2018-06-28:
SPIFFS is ESP file system:
https://raw.githubusercontent.com/luc-github/ESP3D-WEBUI/master/images/SPIFFS.PNG

what is puprpose of that ?

also a jog is actually 3 GCODE commands (relative mode + position command + absolute mode) not 1  so save last command is not waht you want I guess, it is more "last action"

hackintoshlover, 2018-06-28:
Ya I mean last action

hackintoshlover, 2018-06-28:
Any help on saving recent action

luc-github, 2018-06-30:
I guess issue is closed

closed 2018-06-30 by luc-github