luc-github/ESP3D #232: ESP3D Async on ESP32 is Crashing with following Exceptions

Amtho, 2018-06-30:
It was working two weeks ago or so... now i realy dont have a clue what could be the problem.

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4013179a  PS      : 0x00060930  A0      : 0x800df4f8  A1      : 0x3ffb1e90  
A2      : 0x3ffc63c0  A3      : 0x3ffb1eea  A4      : 0x00000000  A5      : 0x00000400  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb1e90  
A10     : 0x3ffb239c  A11     : 0x3ffb239c  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000138  LBEG    : 0x400012e5  LEND    : 0x40001309  LCOUNT  : 0x800f9d39  

Backtrace: 0x4013179a:0x3ffb1e90 0x400df4f5:0x3ffb1ec0 0x400dcc11:0x3ffb1ee0 0x400d6680:0x3ffb1f10 0x400d7ea7:0x3ffb1f30 0x400d21af:0x3ffb1f80 0x4014b2be:0x3ffb1fa0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
E (322) spiram: SPI RAM enabled but initialization failed. Bailing out.

**And decode:** 
PC: 0x4013179a
EXCVADDR: 0x00000138

Decoding stack results
0x400df4f5: WiFiSTAClass::macAddress(unsigned char*) at C:\Users\Amtho\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src\WiFiSTA.cpp line 415
0x400dcc11: WIFI_CONFIG::get_default_hostname() at C:\Users\Amtho\Documents\Arduino\libraries\ESP3D\src\wificonf.cpp line 111
0x400d6680: CONFIG::reset_config() at C:\Users\Amtho\Documents\Arduino\libraries\ESP3D\src\config.cpp line 692
0x400d7ea7: Esp3D::begin(unsigned short, unsigned short) at C:\Users\Amtho\Documents\Arduino\libraries\ESP3D\src\esp3d.cpp line 167
0x400d21af: setup() at C:\Users\Amtho\Documents\Arduino\libraries\ESP3D\examples\basicesp3d/basicesp3d.ino line 29
0x4014b2be: loopTask(void*) at C:\Users\Amtho\Documents\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 15

luc-github, 2018-06-30:
Seems a known issue and this should be fixed soon if it is not : 
https://github.com/espressif/arduino-esp32/issues/1540

Amtho, 2018-06-30:
Yeah but the tipp with just ignoring it is not working. Becuase the esp32 is not booting at all. When powered on the chip will restart after like 6 sec and this goes on and on... i cant get the ap running for configuration anymore.

luc-github, 2018-06-30:
Did you updated the ESP32 core ?
they do IDF update 3 days ago and some others recently

Amtho, 2018-06-30:
yeah i updated the core today. Ok would try to test it with old core but i cant find where i can download an old core...

luc-github, 2018-06-30:
Also reflash with erase all flash content to clean the flash if it is corrupted

luc-github, 2018-06-30:
ho you mean this happen with recent core only not with old one ?
so you can go before big update then just nagivate  in commit history : 
this one was in may : https://github.com/espressif/arduino-esp32/tree/7b243adcfdfdbddc76e571052e7becf310cde757

luc-github, 2018-06-30:
I did not tested latest commits on ESP32 - thanks for pointing this out - now I am aware ^_^

closed 2018-06-30 by luc-github