luc-github/ESP3D #236: ESP32 How to connect a grbl board with esp32?

Amtho, 2018-07-02:
So i have configurated the esp3d firmware to use the serial2 and connected this serial 2 with a arduino mini running grbl this is not working at all. it looks like i get no output when i send a command from esp3d web interface to the serial port. The grbl board is working fine when im connection to it directly.

luc-github, 2018-07-02:
why using Serial 2 of ESP32 and not Serial ? Any reason ?

I remember on early tests I did, serial 2 has issue on ESP32 core , but now it is suposed to be fixed, but I never test since

Amtho, 2018-07-02:
yeah using serial2 because on the most dev boards serial is connected with the usb part.

luc-github, 2018-07-02:
is baud rate same in grbl and esp3d ?

Amtho, 2018-07-02:
yes 115200. at least this is what i set in the esp3d ui. for the grbl im sure this is correct.

luc-github, 2018-07-02:
did you connect Serial2 to a serial to usb adapter to verify Serial2 is doing proper communication ?

Amtho, 2018-07-02:
yeah. and this is the reason i believe something in the esp3d software or maybe in the esp32 core could be the problem. because i dont get any reaction on this port. i only have to define in config.h "use serial 2" or do i have to do more?

luc-github, 2018-07-02:
No only this should be enough - and it is ok for Serial so should work with serial 2

does a serial2 test sketch doing output ?

```
void setup(){
  Serial2.begin(115200);
}
void loop() {
delay(1000);
 Serial2.println("Hello Serial 2");
}

Amtho, 2018-07-02:
i will try this later, right now im not on my dev computer. also will try to use serial0 and see if the issue is also there.

luc-github, 2018-07-02:
Serial0 ? there  no such port 
Serial Serial1 Serial2 only so  far  - I only tested Serial - but need to ground GPIO 15 to GND to avoid output garbage at boot

Amtho, 2018-07-02:
i mean serial

luc-github, 2018-07-02:
I can check Serial2 tomorrow also

luc-github, 2018-07-02:
what ESP32 board do you use ?

Amtho, 2018-07-02:
right now this board. on my own pcb. ESP32-WROOM-32U but also tried the one of this dev boards ESP-WROOM-32 Development Board

luc-github, 2018-07-02:
Ok I have a  ESP-WROOM-32 Development Board will try with this one

Amtho, 2018-07-02:
ok just tested this: 
```
#include "HardwareSerial.h"
HardwareSerial Serial2(2);

void setup(){
  Serial2.begin(115200);
}
void loop() {
delay(1000);
 Serial2.println("Hello Serial 2");
}
```
this example is working and gives me the expected output on serial2.

luc-github, 2018-07-02:
It seems for Serial2 compilation missed a reference - I just added it on https://github.com/luc-github/ESP3D/tree/asyncwebserver

For what I can test at boot I got the M117 192.168.0.1 from serial2 once and after no communication 
I will test more

Amtho, 2018-07-02:
Also what i got as a result now. When im connection just my usb to serial to serial2 or serial i think i get the right signs. but after i connect my arduino mini with grbl just then no serial communication after all. swaped the lines and using a 3.3v powered arduino so this should be not the problem...

luc-github, 2018-07-02:
may be it is linked to this github.com/espressif/arduino-esp32/issues/854
but I am also guessing Serial2 is not for normal usage : 
github.com/espressif/arduino-esp32/issues/1314
github.com/espressif/arduino-esp32/issues/1189

luc-github, 2018-07-02:
Each time I have Serial + Serial2 connected at boot I have the ` M117 192.168.0.1` from serial2 then communication seems not working

Looks like core issue definitly

Amtho, 2018-07-02:
thanks for investigating this. but i also cant get the connection established with serial only. next try would be to resolder my board so that gpio15 is always gnd so that i dont get boot msg intefering thinks.

luc-github, 2018-07-03:
Hi 2 update : 
1 - now you can use  https://dl.espressif.com/dl/package_esp32_dev_index.json
I have updated  https://github.com/luc-github/ESP3D/tree/asyncwebserver to support it

2 - I have tested a sketch with Tx/Rx
```
#include "HardwareSerial.h"
HardwareSerial Serial2(2);

void setup(){
  Serial2.begin(115200);
}
void loop() {
delay(1000);
 Serial2.println("Hello Serial 2");
 int len = Serial2.available();
 if (len > 0){
		char sbuf[len+1];
        sbuf[len] = '\0';
        Serial2.readBytes ( sbuf, len);
		Serial2.print("You typed :");
		String s = String(sbuf);
		Serial2.println(s);
 }
}
```
and so far it is Ok -  so issue is in ESP3D code but linked to API usage or to Async web server 
FYI : On ESP8266 the Async webserver make the serial loosing data so I am rewriting the ESP3D 2.0 with sync server,  Async seems to starve serial in some way so it may affect ESP32 on Serial2 - need more tests

I am currently ending the ESP2.0 sync (which is suppose to be also compatible async  in theory) for ESP8266, will start to test the ESP32 today or tomorrow
So keep this issue open until root cause is known

Amtho, 2018-07-03:
Thanks for your help. I have also tried this https://github.com/AlphaLima/ESP32-Serial-Bridge and i can use the serial2 over the esp32 and can communicate with my grbl board. so looks like you said to an error in the esp3d code

Amtho, 2018-07-03:
Ok now im confused. I just updated your lib on my pc with the fix for the crashes. then configurated in the printer settings: output msg: tcp and now everything looks ok. I can see the comunication on the serial2 port and also get answer from my grbl board.

Amtho, 2018-07-03:
OK it looks like the msg are queued up and not sent instantly. when the buffer is then full the first msg is sent instead the actuall msg.

luc-github, 2018-07-03:
Seems the bridge you share use different begin parameter
But in read me pins for serial 2 are not 17/16 . what pins do you use ?

Amtho, 2018-07-03:
pin 17/16 are the pins im using. for the wlan bridge it is defined as serial1 with pins 16/17. so used then serial1. he just redefines it. but according to esp core "hardwareSerial.cpp" Serial2 is defined for pin 16 and 17.

luc-github, 2018-07-03:
Ok thank you I will do more test when back home

Amtho, 2018-07-04:
Just for your information: IM using now the relased 2.0 branch with sync webserver. And now the Serial2 Communication is working.
I needed to uncomment this in config.cpp Line 57: //HardwareSerial Serial2 (2);
it looks like it is now correctly defined in the esp32 core.

luc-github, 2018-07-04:
Yes they just update hardware serial 23h ago ...

luc-github, 2018-07-04:
Thank you for test feedback - so I close issue

closed 2018-07-04 by luc-github

Amtho, 2018-07-04:
It is not working with the AsyncServer Version! Only with the SyncServer Version!