luc-github/ESP3D #238: ESP3D 2.0 Feedbacks

luc-github, 2018-07-04:
Please feedback on https://github.com/luc-github/ESP3D/tree/2.0

Thank you ^_^

Amtho, 2018-07-04:
run into compilation errors: webinterface.h:46:23: fatal error: WebServer.h: No such file or directory
in webinterface.h line 43: `#if defined(ASYNCWEBSERVER)
#include <AsyncTCP.h>
#else
#include <WebServer.h>  --> should be: syncwebserver.h
#endif`

luc-github, 2018-07-04:
You must use latest git from ESP32 it has this library but better to use AsyncWebserver there is a #define in config.h

Amtho, 2018-07-04:
Thanks for pointing this out. Next problem is when i try to use #define SDCARD_FEATURE i get the next compilation error: config.cpp:34:21: fatal error: IniFile.h: No such file or directory

luc-github, 2018-07-04:
#define SDCARD_FEATURE is not supported yet

luc-github, 2018-07-04:
I have updated the 2.0 branch to avoid confusion - thanks for pointing this out

luc-github, 2018-12-09:
No real update in 6 month so let's close this topic - if anything just open new ticket

closed 2018-12-09 by luc-github