luc-github/ESP3D #239: ESP 3D 2.1 Status

luc-github, 2018-07-04:
Ok time to prepare 2.1

What is the plan ? Do what was not done in 2.0 :wink: 

1 - Nextion Screen support ((ESP3D + Nextion FW)
2 - M408 support according target FW support
3 - Reprap support
4 - MKS TFT WiFI support (Simulate MKS-WiFi module behavior to use the connector)
5 - ESP3D as Library as part of main board FW (eg Marlin)
6 - UI improvement :
      Follow disconnection in authenticate page (Authentication is currently disabled in 2.X)
add toshiba flash air support
7 - Add ethernet support for ESP32 (like olimex)
8 - Add optional translation file on SPIFFS for oled ESP3D message like (connected, update, etc...)
9 - Monitor temperatures on Oled in autonomous way (no web browser connected)
10 -Embedded Host support for ESP32 and FWvGRBL (OLED/SD card/rotary encoder ?)

Amtho, 2018-07-04:
Nice list of work. Sorry i think  i cant help you much to accomplish all. Im more from the electrician site...
But as you my have noticed ;) SD Card support directly on the ESP32 would be my top wish ;)

luc-github, 2018-07-05:
No worry, testing and feedback are great help, so you help ^_^

luc-github, 2018-07-22:
PR for sync webserver (esp8266/ESP32) have beed done - first one is in waiting list for 2.5 and second one is merged already

I am spending some time on UI to get consistent UI across browsers and devices - especially mobile devices - seems response behvior of bootstrap 3.0 is little bit messy sometimes - so need to be managed 
- unfortunatly fix depend of each control and position - so need some time - but results are promising so should be ok next week

luc-github, 2018-08-21:
About Marlin library code is under early development here : https://github.com/luc-github/Marlin/tree/esp32 because hard to separate both code    
Latest Status is here:  https://github.com/luc-github/Marlin/issues/1   
it is alpha state and need code cleaning as I currently I focus on 'feasability' first - once everything is working / code will be cleaned for sure

vivian-ng, 2018-09-10:
Another feature request (wishlist?): display the interface on a touch screen (not just the Nextion) so that it can be used with generic TFT screens which we usually use with Arduino or STM32 dev kits. This is probably quite a major undertaking with a total design of the menu tree plus touch interface. Maybe start small without touch (use buttons or rotary encoder for inputs). Just an idea that makes the ESP3D effective the "OS" for MKS TFT alternative.

luc-github, 2018-09-10:
Currently Oled screen is supported 
Having TFT or Nextion make sense only if ESP3D become a Host or you have something else in mind ?

So first step is to add full host feature - which is planned then TFT 

Do you have TFT screen to recommend ? One you already used and happy with 
Nextion is nice for easy customization but I am not really happy with touch responsiveness - so TFT support should happen  before Nextion

vivian-ng, 2018-09-12:
@luc-github Yes, full host feature is what I was thinking about. The web interface is great already, but without a local interface, it is still somewhat hard to use. So a screen and input (rotary encoder or buttons) would be great. But that only comes after full host feature.

As for screens, I have a few lying around, using ILI9341 (need additional 8 pins), ILI9225 (SPI), ILI9486 (need additional 8 pins), and ST7735S (SPI). I also have a M5Stack, which comes with its own screen and 3 buttons (so it already has all the stuff required to be a host). I am not a very good programmer, but I hope to help by testing and debugging.

luc-github, 2018-09-12:
@vivian-ng thanks for the feedback - I have ordered a couple of ILI9341 seems well supported by ESP32
I will received them in 2 weeks I guess - I will start to check then

luc-github, 2018-09-20:
[GRBL-ESP32](https://github.com/bdring/Grbl_Esp32/tree/WebUI)  and [Marlin-ESP32](https://github.com/luc-github/Marlin/tree/eps32_webui) are now using using  [ESP-WEBUI 2.1](https://github.com/luc-github/ESP3D-WEBUI/tree/2.1)
I will create an ESP3D 2.1 branch soon to sync latest bug fixes and features to ESP3D itself

2.1 is mostly ESP32 focused because more and more possibilities / features come with ESP32
the top 2 I am exited to integrate :
 1 - Touch Screen TFT for ESP32 as host (got ILI9341 screen to play with)
2 - Camera (just got  couple of OV2640 modules and M5Stack ESP32 camera modules )

luc-github, 2018-11-15:
Ok I have re-ordered TFT as the ones I got were without touch :disappointed: 
this time the Touch is confirmed to be XPT2046 spi port ....

luc-github, 2018-11-29:
I rename topic 3.0 - I am rewriting the complete core because I want to add more cross platform functions and simplify the code by splitting files by functions
So it will be easier to add new features or modify existing ones
because it is big rewrite - I thing it make sense to change the major version number

vivian-ng, 2018-12-12:
> Hi Luc, the M5 stack camera modules look great, I also found these, and was wondering what you think of them as candidates of ESP3D. The thing that catches my eye is the built in SD card. This opens a few possibilities for host function, or at least file upload to sd card.
https://www.aliexpress.com/item/1pcs-ESP32-CAM-WiFi-WiFi-Module-ESP32-serial-to-WiFi-ESP32-CAM-Development-Board-5V-Bluetooth/32955484091.html
These cameras are not as easy to flash as they do not come with USB port.

Even the new M5Cam with PSRAM is working quite hard to stream the video. Without PSRAM, video streaming will have lag. I am not sure if it can handle the additional load of running ESP3D if video streaming is already such a burden. Instead, I intend to use my ESP32CAM and M5CAM modules to monitor my prints. They can be configured to connect to the home router, and then it is just a matter of inserting "http://192.168.x.x/jpg_stream" as the link for camera streaming in Octoprint or ESP3D.

luc-github, 2018-12-12:
@vivian-ng ok you confirm my fears

luc-github, 2018-12-20:
there a recent sample add : https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/Camera/CameraWebServer
it may worth to try it

just-jason, 2018-12-24:
Thanks for the tip, I am still waiting for my board to come from china, could be a while still

luc-github, 2018-12-29:
Quick note, 3.0 have several settings simplifications:
- no more sleep mode configuration : always set to none as connected to printer
- no more PHY configuration : esp8266 looks have now hard coding N mode so no need to bother with useless setting
- AP IP is only static and the only settings that can be configured are IP and channel, masque is set as 255.255.255.0 because only 1 client is supported at once.
- AP is always visible and security will be set to WPA if password is defined or OPEN if no password, so no more settings for SSID visibility neither authentication type.

Feel free to comment if any issue

luc-github, 2018-12-30:
I am starting working on Ethernet support - I have a board for ESP32 (olimex gateway) but none for ESP8266 .
* Anyone know an existing solution for ESP8266 or one that does not need to solder component ? - I am not good at it :cry:
* Does another board with ethernet exist for ESP32 ? I know there is a POE version but looks expensive.

So currently only the ESP32  Olimex gateway should be supported.

Ethernet  will have DHCP / Static IP and can be act as DHCP server, by default not
So network modes will be :
1 - Nothing
2 - Wifi AP (static IP only)
3- Wifi STA (static or DHCP IP) 
4 - Ethernet Client (static or DHCP IP) 
5 - Ethernet Server (static IP only)
6 - Serial Bluetooth
I think it should cover all needs, to clarify acting as gateway (WiFi / Ethernet) is out of current specs so far.
Feel free to comment

luc-github, 2019-01-18:
I see several libraries for ILI9341 and Touch XPT2046 SPI 
-> TFT_eSPI (esp8266/esp32)  https://github.com/Bodmer/TFT_eSPI
-> ESP32_TFT_library (esp32) https://github.com/loboris/ESP32_TFT_library

I did not play with any yet

Looks like TFT_eSPI  is multiplatform which would be easier to implement
Anyone used these libraries or any others ones ? Any comment ?

vivian-ng, 2019-01-19:
@luc-github 
I can see merits of both, so I am quite torn over which to suggest too.
-> TFT_eSPI (esp8266/esp32) [https://github.com/Bodmer/TFT_eSPI](https://github.com/Bodmer/TFT_eSPI)
Bodmer seems to be very active in development of drivers for displays, so it is likely that this library is going to be stable and capable, and can probably rely on Bodmer to help if problems are encountered.

-> ESP32_TFT_library (esp32) [https://github.com/loboris/ESP32_TFT_library](https://github.com/loboris/ESP32_TFT_library)
I thought I recognized the name loboris, then I saw the repos and yup, this is the same person who has a fork of MicroPython working on the ESP32 with PSRAM. So it is again someone who can be relied on to help if any problems are encountered.

Similar to you, I think TFT_eSPI seems to be easier to implement, since it is an Arduino library (ESP32_TFT_library doesn't have any Arduino examples, so it is going to be a bit harder to try an figure it out). It also looks to be easier to configure for different displays through the User_Setup_Select.h header file. This makes the implementation more high-level, and we just need to worry more about how to implement features, and let the library handle the backend low-level driver portion. End users will then just need to change the config file to select their specific display, and can then compile.

DerKrawallkeks, 2019-01-20:
I just wanted to tell you how great this project is, since I just installed it with an esp8266 on my printer.
Again, is there any chance to increase the speed of upload to SD?

luc-github, 2019-01-21:
@DerKrawallkeks I have just created a FAQ for this question : https://github.com/luc-github/ESP3D/issues/292, please have a look

luc-github, 2019-01-23:
I am now rewriting the web events part to use same api as async with sync webserver - this would simplify the ESP3D and ESP3D-WebUI code not using the Websocket for sync instead, it will also be implemented in embedded page for communication control

luc-github, 2019-01-24:
this year for sure, ASAP I can do , now an exact date - no sorry - I do others projects in parallele and 
3.0 is a big rewrite which also include the WEB-UI, currently I am working on web events, then will implement WebSockets for printer feedback , once done I will work on Embedded Web page to support new communication way, then update webUI to support new communication way also.
Then will add SD support, and also time server for file upload, then will be back to webUI to check if everything is Ok, then will add host support and again will have to add this feature to WebUI

So as you see many to be done before something usable - no mention DHT support and TFT screen support and some others minor features I need to work on

About ESP32 any would fit I think,  I have several ones - and will I try all during test - do not use the `too cheapest chinese ones` - I got some really cheap ESP8266 / ESP32 which were actually really crappy  and not working...
Choose the one you are confortable with , with SD reader is convenient butyou may need the check what SPI pins are used, if you buy without SD you need to buy a SD card reader but this one can have a SD detect feature which is may not the case of embedded ones, but with external SD read it is also easier to use the right pins for connection

IInickII, 2019-01-24:
Thank you for reply. Can you give me some link on ebay or somewhere for type of board example which you use?
Thanks

luc-github, 2019-01-24:
https://www.aliexpress.com/wholesale?spm=2114.search0104.8.11.2951ba86EQdEqQ&initiative_id=QRW_20190124063635&SearchText=t8+v1.7+ttgo+esp32&productId=32853967655

They also have ESP32 with 128Mb flash (16MB) which allow to use FAT ( instead of SPIFFS 11MB)

I usually buy from them : https://www.aliexpress.com/store/1983387?spm=2114.search0104.8.7.46a71368F1Pioi

luc-github, 2019-01-24:
But now it is Chinese New year so I think nothing will be sent before 3 weeks

luc-github, 2019-01-30:
Testing ILI9143 + ESP32, so far so good ^_^

Just need to solder the SD card reader header, connect all wires, and compile sample accessing JPEG from SD and render
[image]
[image]
Everything work out of the box - very nice - back screen is little bit too hot (the LED part) almost burned my fingers...
Touch screen reactivity is better than nextion IMHO, but precision is just Ok - need big buttons I think - the keypad sample and calibration sample are working as expected.
The embedded SD reader is working well - just miss an SD detect pin, but can use without it
 
Now need to read  ESP32_TFT API and write some code :smile_cat:

luc-github, 2019-01-30:
Another change in 3.0 as be a library seems not really a need - be back to single app - and as I got several people not able to install external libraries - everything will be included 
files organization has been also modified to become simple (1 ino file, 1 configuration file, all others files are in subdirectories - including necessary libraries
Current structure as of today (still miss telnet, websocket, oled, tft, host, etc...) :
[image]
This should avoid end user installation issues (I hope) :crossed_fingers:

rikaige, 2019-02-03:
Hi Luc,
I'm writing you about Nextion display integration...
In these days I'm starting to develop something new for my RADDS+Due running MalinKimbra firmware, I have a Nextion display on my shelf but don't really like the MK graphics so I'm drawing a new interface and then I'll make them communicate... 
Looking for some WiFi implementation I found your project and your will to implement this display that sounds really cool to me :) because this will be totally cross-firmware&hardware compatible, I have never worked with ESP mcu but I will be happy to help with the graphics and Nextion coding ;)
In the meanwhile I ordered a couple of ESP (8266 & 32) boards to test the WiFi connectivity and look into the code!

Rikaige

luc-github, 2019-02-03:
Thank you - I won't recommend to use ESP8266 as host + screen support + wifi with all ESP3D feature, not enough power - ESP32 is the board I will test only 
Not acting as host would be a solution for esp8266 but lot of people feel disapointed when they try the SD upload which is slow like hell because of current protocole not really file oriented, acting as host would avoid this.

Sure every  help is welcomed - I am definitly not a good designer - I started to write Nextion UI some time ago :
[image]
[image]
[image]
[image]
[image]
[image]
[image]

 but result is not really nice, so I moved on to other topics until I get an UI I like and so work on its implementation.

I am also starting to work on ili9143 Touch screen implementation, and design will become soon the problem, I am just at reading API and testing code, I  wish to use same UI for Nextion and TFT

What Nextion screen do you have ?

rikaige, 2019-02-03:
I have a 3,5" 480x320 basic display now, but it was meant to be used in an other project so when I'll have defined the graphics I'll think about the proper screen size, even if it this seems big enough to me...
The cool thing about Nextion, or similar displays, is it is self-sufficient and needs only serial communication to interact with it, thus I was thinking , but not verified yet, about retrieving data using GCode commands as any other host, but leaving all the printing control on the printer hardware, more or less as the ESP is doing I guess... 

Do you think having the ESP wired to the printer hardware on Serial and the Nextion on another ESP Serial (HW or SW) with the ESP that handles the display informations will be too demanding for ESP?

As soon as I have something ready I'll share some screenshots of the idea.

luc-github, 2019-02-03:
No Nextion is jus at display - it cannot be used as standalone unlike MKS-TFT which is actually a M0 +TFT 
* Nextion use it's own protocol using Serial - so it cannot send pure GCODE, https://www.itead.cc/wiki/Nextion_Instruction_Set, it means ESP must be the MCU handling the code
you can check Nextion library 
* its memory is very low, the minimum to display output - for example file listing cannot be handled directly by nextion, it need to be splited list like 5 names by 5 names
* Nextion SD card is only for flashing screen, no direct access

The Nextion strength is the ease of use to change UI - you can add code the UI to simplify the back end and so the communication with ESP but not so much actually.

Yes the ESP must be between Printer and Nextion - ESP32 have Serial + Serial 1 (serial 2 seems having hardware bug for what I read)
if no host ESP8266 must use Software Serial but I would prefer to start with ESP32 to be sure is working first and see if it is can be optimized for ESP8266 after

just-jason, 2019-02-22:
Hey Luc , how about some beeps from the printer eg, when commanded to start printing, file upload starts / ends? I am sure you can think of a few more good ideas .

luc-github, 2019-02-23:
Well could be an idea. But need to be an option as not all Printers/FW support M300 command. Also these commands are time consuming so if use music instead of few beep. Another solution would be to add buzzer on ESP, it won t bother printer with additional command and will work as stand alone what you think ?

just-jason, 2019-02-23:
I agree , options are always great. I would be happy to have the option for the printer buzzer. Every one can turn in on/ off according to their setup.

luc-github, 2019-03-09:
Just did a sync with my repo to add support for DHT ,  Reset Pin and Time management https://github.com/luc-github/ESP3D/commit/b0375a3541eca974f88302e5568d71782a92651f
Next modules on a way :
* Notifications pushover and email
* Basic host for Macro and autostart
* oled + TFT 
* rotary encoder + touch

dremeier, 2019-03-17:
sorry, but where do I find the ESP3D 3.0? I´ve no luck with 2.0.

luc-github, 2019-03-17:
3.0 is in 3.0 branch but not yet usable, not in pre alpha  as I did not rewrote the webui yet - current webui is not compatible as I have rewritten lot of things. I am first working on pure FW part
I have recently pushed the  notification port for pushover/Line/Email
Next step is the basic host and autostart, then Oled support
if you have issue with 2.0 please open issue

dremeier, 2019-03-17:
Ok, thank you will fight with 2.0 :-)

luc-github, 2019-03-17:
3.0 will be nice I think, but I need time to work on it,  sorry

aganov, 2019-03-26:
@luc-github Do you have any plans for the web ui. My proposal is to use webpack for the build process and then we can use https://github.com/developit/preact (3kb) or https://github.com/choojs/choo (4kb) to organise the UI into components and bring some reactivity. Also we can replace glyph icons with some svg based icons like https://feathericons.com/ for example... What do you think?

luc-github, 2019-03-26:
Not yet I was planning to reuse the existing javascript I have wrote as I do not want online dependency in case printer is not connected to internet - but I am open to new solutions
I never used these projects do you ?
about glyph currently I use svg extracted from bootstrap, but I am also Ok to change - and there was a request to be able to add new ones so I am fine as soon the size is kept small

To sum up I open to everything but I am not sure I can study all solutions by my own, especially because I do not know them so any proposal is welcome

luc-github, 2019-03-26:
@aganov I have created https://github.com/luc-github/ESP3D-WEBUI/issues/47 - which explain also size constraint

just-jason, 2019-04-11:
> I am starting working on Ethernet support - I have a board for ESP32 (olimex gateway) but none for ESP8266 .
> 
> Anyone know an existing solution for ESP8266 or one that does not need to solder component ? - I am not good at it 😢

If you are still thinking about this , then take a look here

https://esp8266hints.wordpress.com/2018/02/13/adding-an-ethernet-port-to-your-esp-revisited/

luc-github, 2019-04-11:
Yes I know this one - it need to do some soldering ^_^, I addition of not beeing good at it - I am lazy, that is why I was looking for a ready solution 😉

luc-github, 2019-05-25:
LittleFS is now in esp8266 core git - it should allow time/date unlike current esp8266 SPIFFS, but need to check if current implementation allow it - TBC

luc-github, 2019-05-25:
about **USB host support using arduino usb host shield** I have ordered a couple of shields to check - they will arrive in few weeks in theory

schorhr, 2019-07-13:
> about **USB host support using arduino usb host shield** I have ordered a couple of shields to check - they will arrive in few weeks in theory

Hello :-) I am just curious if it works, no rush or anything. Would be amazing. Thank you for your hard work!

luc-github, 2019-07-13:
@schorhr I have tested a little but seems my hardware is a clone - no wonder it was cheap, and USB connector is not really reliable, so I got a lot of random disconnection , undetected connection, etc...
 I have ordered an uno shield usb host too compare behavior but did not have time yet.
during my test I realized the code need to match the targeted device, which will make task little bit more complex than I expected.
I have also received today some UDisk boards (USB connector with CH376s chip) to read USB drive like SD card, I plan to support it also

schorhr, 2019-07-13:
Thank you for taking the time to update on the progress!
Too bad you're having troubles with the clones. But I suppose it's a good thing, as many people would use clone boards as well, so it's no surprise later on.
Great! Good luck!

ithinkido, 2019-07-16:
Hello Luc,

Would it be possible to get the actual flow, feed, ( and fan speed ? ) values? At the moment , if for any reason you reconnect, then the value shown will be 100% even when the machine is running a different value

luc-github, 2019-07-16:
Which gcode get these values ?

luc-github, 2019-08-11:
Tft screen have colors support and better resolution than oled so I will integrate littlevgl support. This great library allows very nice UI, for oled I will keep current simple UI
I have already wrote littlevgl drivers to access Flash (Fat and SPIFFs) and SD for images. But in case of flash image storage,  I think 8M  to 16M flash will be necessary, TBC

labotecno, 2019-08-29:
hello what stuff do you use for your new esp3d 3  ? 
to plain to byy it soon 
i use esp3d 2.1 
what is the last version of your firmware esp3d 2.1 
thank you

luc-github, 2019-08-29:
any ESP should fit but better have at least 4M flash
depending on needs ou choose ESP32 or ESP8266, some feature like host or even Touch/tft ESP32 is better choice more pins and 2 cores, more feature need more power = esp32, limited/specific feature need less power = esp8266.

luc-github, 2019-09-29:
Ok after nice POC of https://github.com/eben80/ESP32-CAM-Print-Monitor I took some time to check and was able to make the camera streaming server to coexists with webserver and  ESP3D.
decreasing the task priority and disabling the face recognition helped a lot ^_^
Still need to do some stress tests to verify - but I was able to upload a small file on SPIFFS when video was streaming
There video is streamed using httpd server which require 2 ports available, the commands to modify camera settings are done using [ESP172] command
WebUI is not done yet but camera stream is visible using browser `<ip>:<camera port>/stream`

vivian-ng, 2019-09-29:
@luc-github Interesting idea for the camera.

I actually have a ESP32-CAM module (actually, a few) which I configured to do a MJPEG stream for the "Camera" tab in ESP3D now (running on Marlin-ESP3D). Of course, the camera module is not connected to my MRR ESPA board except for power, but it would be interesting if I can modify the "webcam" sketch to monitor the serial port for similar [ESP172] codes. Your `Camera::command` method looks like a good place to start.

eben80, 2019-09-29:
Glad that this device is getting a second look :-)
I'm still using it with my basic functionality, monitoring, manual gcode entry, abort sequence and remote power down. Following you with interest!

luc-github, 2019-09-30:
@vivian-ng using current 3.0 should be ok out of the box for what you describe , for adding ui command should need to modify the html page of webcam to change the commands to send to webserver for camera- I will try to do it this week as sample.
@eben80 yes thanks you for showing the  possibilities ^_^, I still need to do a compatible WebUI with and without overlay info so it is not currently as convenient as your solution.

vivian-ng, 2019-09-30:
@luc-github Let me play around with @eben80's sketch and see what I can do with it.

@eben80 I see that your cpp file has a cmd_handler which seems to allow the camera settings to be changed on the fly. If you can expand a bit on how to do that in the README, that would greatly help. Of course, one can always read the code... 😅

[99 more replies]

closed 2020-12-11 by luc-github