luc-github/ESP3D #240: ESP2.0 Serial2 with AsyncServer not working.

Amtho, 2018-07-04:
Sorry for reopening it, but the case is still not really fixed.
When you use Asyncserver Serial2 is still buggy. When using SyncServer everything is fine.

luc-github, 2018-07-05:
Actually there is a missing fix for Serial and async, I forgot to implement it because it was not enough for esp8266, but as serial is ok for esp32 it may solve serial2 issue
Will add it in 2.0 asap

luc-github, 2018-07-05:
@Amtho I just pushed some commits - they should solve issue as on my side Serial2 is now Ok
https://github.com/luc-github/ESP3D/tree/2.0
Got also some fixes pushed by @me-no-dev that would help a lot on Serial too

Please check and let me know

Amtho, 2018-07-05:
Yes looks good. Now Serial2 is also working with the AsyncServer. Thanks to you both!

closed 2018-07-05 by Amtho

luc-github, 2018-07-05:
Cool ! thanks for the confirmation