luc-github/ESP3D #244: Excessive EEPROM writes

ESS-1, 2018-07-26:
The firmware does EEPROM.commit() after changing any setting, including situations when multiple settings changes sequentially.
E.g. CONFIG::reset_config causes 42 erases of the same flash sector.
Considering typical endurance of 100K erases and no load leveling, this is rather unacceptable.

luc-github, 2018-07-26:
when settings are set, you are not suppposed to change them, unless it is user major activity ...
This is same for reset - I have hard time imagine people keeping doing reset (more than 2000 times in our case)

Commit each change was the easiest and safest way.  IMHO

Could it be optimized ? yes, but what will the gain ? increasing code complexity  and so possible errors because of  the different user scenario. So I am not sure it worth it. again IMHO 
  
Now saying it is "unnacceptable" !! Please develop, because the 100K is not the only "good" reason, technical is one thing - user case is also very important.

ESS-1, 2018-07-26:
OK, it is not so critical as it would be :)
Anyway, some SPI flash IC's have only 10K cycles, and those used on cheap modules from China may have unpredictable endurance.

Why not to add a "Save" button, which is shown only if there if there is unsaved EEPROM data?

luc-github, 2018-07-27:
Of course it could be done - as many other improvements
About cheap modules, even 200 reset won't be the critacal path IMHO - something else will break before

ESS-1, 2018-08-01:
There is another source of flash writes: the SDK from Espressif automatically saves WiFi parameters, but the firmware overwrites these when loading. Did you investigated this?

PS: what "delay(8000)" after "WiFi.mode(WIFI_OFF)" does?

luc-github, 2018-08-01:
Hmm not sure what you mean - flash is flash : SPIFFS,program area, ota area and EEPROM all area are subject to same limitation of wearout.

the 8 s are to let printer FW starting first to not load too much current, and also for reset.
This timing is split in 2 delay that can be defined in 2.0 ` void begin(uint16_t startdelayms = 8000, uint16_t recoverydelayms = 8000);`
recoverydelayms is also now ignored if feature is disabled

ESS-1, 2018-08-02:
Thank you for the explanation of the delay.

Regarding WiFi and flash, I mean that ESP core for Arduino saves WiFi configuration to the flash automatically, e.g. when we call "WiFi.mode(...)", the module saves the new mode and automatically restores it after restart (system_restore() is supposed to clear saved configuration).
Considering that ESP3D re-configures WiFi at startup, I think we can disable this by calling "WiFi.persistent(false)" at the beginning of "setup()".

I've made this change in the fork I created recently, and this works fine so far:
https://github.com/ESS-1/ESP3D/commit/55e742e05cc88674c5958bcc3f8b8f2a5176d8bf

luc-github, 2018-08-02:
Persistent write only is parameters used in begin are differents than the one already saved so not updated each strart
And these settings impact the autoreconnect feature lasttme I tested
The 2.0 use events  but auto reconnect still rely on embedded

luc-github, 2018-08-02:
Ok seems that was modified since last time I checked : https://github.com/esp8266/Arduino/issues/3641
http://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/generic-class.html#persistent

but autoconnect feature is lost if set to false before settings are saved to flash .
So need to handle auto reconnect in that case ...

ESS-1, 2018-08-03:
Yep, WiFi::begin() compares configurations and likely it doesn't modify the flash memory in our case.

But ESP3D also call WiFi.mode(WIFI_OFF) followed by WiFi.mode(WIFI_STA) or WiFi.mode(WIFI_AP).
WiFi.mode() is a wrapper for wifi_set_opmode(), which saves new mode to flash (see https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf
).
So when are at least two writes when ESP3D is turning on.

Why do you think these settings may affect auto-reconnect?
I've made full flash erase, then uploaded the firmware, SPIFFS, and turned STA mode on. The module gracefully restores STA connection when I'm turning my WiFi router off and on.

I added system_restore() to CONFIG::reset_config(), so it is called only when other settings are resetting.

luc-github, 2018-12-09:
I am rewriting the FW for 3.X and so I will implement the persistence settings to false
But I have no plan to change the atomic  update process - so I guess Issue can be closed

closed 2018-12-09 by luc-github