luc-github/ESP3D #247: Platform IO

ModMike, 2018-08-05:
Is anyone using Platform IO to compile?  Would love to leave Arduino IDE completely behind.

luc-github, 2018-09-08:
Did you succeed ?

ModMike, 2018-09-08:
I got it all setup but I didn't pursue, seeing as i had the tool chain setup.  I will try again next week and update you.

ModMike, 2018-09-08:
Hi Luc,

You need to enable the framework, which is pretty easy:

[image]

Then pick board:

[image]

After that you can create new project for home screen:

[image]

That will get you going.

ModMike, 2018-09-08:
Edited comments to keep it concise.

luc-github, 2018-09-09:
Thanks for sharing  ^_^

luc-github, 2018-12-09:
I have labeled as FAQ - so issue can be closed - thank you

closed 2018-12-09 by luc-github