luc-github/ESP3D #248: Socat connection

Zefram88, 2018-08-07:
As title, i'm asking if it is possible to create a virtual tcp-to-com connection using socat or something similar. This is beacuse i'm using OSX and repetier version is really outdated, so i can't take advantages from network connections.

Thanks

Zefram88, 2018-08-12:
UPDATE:
I'm able to successfully open a connection through this command:

`socat -d -d pty,raw,echo=0,wait-slave, tcp:[IP_ADDRESS]:[PORT]`

Now the problem is how to select the interface from the slicer (Cura, Repetier etc..)

luc-github, 2018-08-12:
you may need to set a Serial port in command like : `PTY,link=/tmp/ttyS0`

Zefram88, 2018-08-16:
With the command `socat -d -d pty,raw,echo=0,link=/tmp/ESP,wait-slave, tcp:[IP_ADDRESS]:[PORT]` work but only on Proterface because you can set every address you want. The problem is on Repetier host/Cura because you are not able to select the interface manually and it is not auto-detected.

The connection itself works just fine

luc-github, 2018-10-02:
@Zefram88 did you solved your issue ?
In repetier host you should be able to  select TCP connection and enter IP/port

Zefram88, 2018-10-03:
> @Zefram88 did you solved your issue ?
> In repetier host you should be able to select TCP connection and enter IP/port

Well, yes and no.
Basically the problem is not of ESP-3D itslef i suppose. I can't use repetier beacuse i own a macbook, the repetier version is outdated and doesn't have the IP connection function.
Cura also doesn't detect the connection.
I'm able to use it only on Proterface

The best would be to make ESP3D "findable" by Cura

luc-github, 2018-12-09:
I guess issue can be closed then

closed 2018-12-09 by luc-github