luc-github/ESP3D #249: Using with MKS TFT Wifi - Any Success?

ZhuDaHai, 2018-09-11:
Issue #206 was closed back in March concerning using the MKS TFT Wifi board with this ESP3D firmware. 
I also have the MKS TFT28 display board and the MKS TFT Wifi board, and the default closed-source firmware from MKS is very limited. So I'd like to see if this could work.
And so was wondering if you'd made any progress on this and/or if there was anything I could do to help.
Thanks!
P.S. As a side note, I updated the MKS TFT WIFI board with the ESP3D firmware, but now I cannot go back. I don't know why, but while I can load the original MksWifi.bin file and set the mks_config.txt file on the SD card to update, and it does appear to update, the MKS TFT WIFI board will not switch out of AP mode no matter what I try. Thoughts? 
Thanks!

ZhuDaHai, 2018-09-12:
That worked!
I flashed 'EraseFlash', then the stock MksWiFi.bin, then the mks_config.txt file and its back to stock.
Thanks! I can go back and forth now while I test.
Let me know if I can help with your firmware for the MKS TFT WiFi module. I'm a retired programmer of 30+ years. I don't know jack about this module, but I can still learn :)
Maybe point me to the correct code block...
Thanks again!

ZhuDaHai, 2018-09-12:
I littered your code with LOG command at every ESP_SERIAL_OUT.readBytes command I could find, but nothing I added appeared in the log.txt file - just your stuff. So apparently I haven't a clue where to look.
I did find these in the MKS TFT firmware. Maybe they'll help. Just wish I knew where to look for where that would be coming into your code:
```
at+netmode=%d
 at+dhcpc=%d
   at+wifi_conf=%s,auto,%s
   at+wifi_conf=%s,wpawpa2_aes,%s
    at+net_ip=%s,%s,%s
    at+net_dns=%s,8.8.8.8
 at+remotepro=tcp
  at+mode=server
    at+remoteport=8080
    at+uart=%d,8,n,1
  at+net_commit=1
   at+net_wanip=?
    at+wifi_ConState=?
    at+reconn=1
   at+dhcpd=%d
   at+dhcpd_ip=%s,%s,%s,%s
   at+dhcpd_dns=%s,8.8.8.8
```

luc-github, 2018-09-12:
Yes as I suspected after few test MKS use AT commands - to query / set wifi parameters
ESP3D do es not know AT commands so do not answer to any query send by MKSTFT so TFT think there is no wifi

Need to connect RX/ TX /GND to RX/GND of Serial to USB adapter to visualize the exchange on a Serial Monitor
once the communication exchange is known  - we can simulate the answers from ESP3D

luc-github, 2018-09-12:
Yes `ESP_SERIAL_OUT.readBytes` just ignore the command as it does not know them
need to code proper answer like:
if it receive `at+net_wanip=?` it need to answer for example `192.168.1.4` but currently I do not know what command it could receive and what format it need to use to answer - it could be standard AT or custom AT

luc-github, 2018-09-27:
Ok I have some time and checked what TFT is requesting:
```
at+netmode=2
at+dhcpc=1
at+wifi_conf=NETGEAR_2GEXT_OFFICE2,auto,blabla
at+net_ip=192.168.1.240,255.255.255.0,192.168.1.1
at+net_dns=192.168.3.1,8.8.8.8
at+remotepro=tcp
at+mode=server
at+remoteport=8080
at+uart=115200,8,n,1
at+net_commit=1
at+net_wanip=?
at+net_wanip=?
at+net_wanip=?
at+net_wanip=?
```

luc-github, 2018-09-27:
This is the doc for the wifi module 
[HLK-RM04 user manual.pdf](https://github.com/luc-github/ESP3D/files/2425257/HLK-RM04.user.manual.pdf)

luc-github, 2018-09-28:
But my TTF screen seems not communicating well with the WiFi module the TX seems OK but not RX  
I can see the above command sent by TFT

Modules communicated once got connected and right IP, then no more (Only got exception) so hard to continue

ZhuDaHai, 2018-09-28:
Thanks for the updates! I plan on working on it this weekend and let you know if I make any progress or not.

luc-github, 2018-11-15:
Hi any update on this ?

luc-github, 2018-12-09:
My MKS TFT seems not well communicate and no update for a while 
I close topic - feel free to re-open if any update

luc-github, 2019-03-09:
I have ordered a new MKS TFT a v4 this time,I hope this time it will work 
https://www.aliexpress.com/item/MKS-TFT32-v4-0-touch-screen-MKS-Slot-module-extended-SD-card-reader-splash-lcds-touching/32946735899.html

Valcob, 2019-03-22:
just posting in case it may help. 
So it looks like the MKS WIFI has 2 UARTs setup for communication the one is a normal one UART0 setup on pin TX(label on the wifipcb) goes to RX0(IO3 of the ESP12s) and RX(label on the wifipcb) goes to TX0(IO01 of the ESP12s) and another one UART1 is on TX(mainboard <-> IO0 wifiboard) goes to IO0(ES12s) RX(mainboard <-> IO1 wifiboard) goes to IO4(ESP12s) looks like sw serial here.

So on UART0 when just wifi is connected to  usbToSerial converter is 
[image]
on RX pin of the wifipcb; on TX is nothing
, and on UART1 is some rubbish  
[image]

And when connected to main board which is and MKS TFT 3.5 I have this on UART0 from RX wifiboard
[image]
the two black rectangles are SSID and pass
and at the same time the responses are from the TX pin on wifiboard
[image]

And while still connected to main board on UART1 TX pin of the main board
[image]

And for UART1 RX pin of the main board
[image]
with something being continuously written 
also played with baud rates to get something useful but with no luck

If there is something I can help I would be glad to do so.

luc-github, 2019-03-23:
@Valcob interresting what FW version do you have and what settings for wifi ?
I just got my MKS TFT32_L V4.0 looks like it has the latest FW 3.03

connector Aux1   

 ?  |  ? |  ? | ?  
---|---|---|---
 TX | RX | GND | +5V

connector WiFI   

 RST | RX | TX | NC    
---|---|---|---
 TX | RX | GND | VCC   

Connector SD   

CS | DI    
---|---
GND | V
CLK|GND
DO|CD

Can you clarify which TX/RX you mentioned ? Thank you

Valcob, 2019-03-23:
@luc-github 
so WIFI pcb is MKS wifi module, TFT is MKS TFT 3.5 with 1.0.2 firmware connector used is WIFI 
[image]

luc-github, 2019-03-23:
Ok

luc-github, 2019-03-23:
Ok using MKS TFT WIFI and updating with latest FW it seems working and now I even have a webserver which I did not have before
[image]

Communication with board according config is 250000 and I can see the M105 on Arduino serial
so my previous issues were FW and not hardware ...

[image]

I will dump the MKS WIFI firmware to have a backup - just in case

Valcob, 2019-03-23:
can you please share your config for tft ?
```
#########  Printer type setting  ##################################

#mainboard firmware setting(marlin:1; repetier:2; smoothie:3)
>cfg_firmware_type:1

#machine setting (Normal:1; Delta:2)
>cfg_machine_type:1

#baud rate (9600:1; 57600:2; 115200:3; 250000:4)
>cfg_baud_rate:3
```

luc-github, 2019-03-23:
But I cannot do any telnet action, I have tried different port and connection failed 
I connected to 192.168.1.25:8080 - connection is accepted but no effect /feedback of any command
[CONFIG.txt](https://github.com/luc-github/ESP3D/files/2999309/CONFIG.txt)

luc-github, 2019-03-23:
So I am not sure using wifi bridge is working or need additional setting

Valcob, 2019-03-23:
also can you share what wifi binary you are using I cant get  a normal output(M105) for some reason (

luc-github, 2019-03-23:
I use the one present in https://github.com/makerbase-mks/MKS-TFT

luc-github, 2019-03-23:
Version 3.03

luc-github, 2019-03-23:
Looks like they build a cura plugin. Will check it https://github.com/makerbase-mks/Software/blob/master/MKS%20Plugin/MKS%20Plugin3.5↑.rar

Valcob, 2019-03-23:
yeah I saw that too and it works I can confirm
[image]

luc-github, 2019-03-23:
So cura can send commands using wifi
So hope to see in plugin code how to do

luc-github, 2019-03-23:
Ok I was able to bridge RX/TX wire and I saw the communication at 115200 baud same as you @Valcob 
but when checking wires to see which wire is mandatory. the VCC was disconnected and after reconnect fully the MKS TFT WIFI - I loose the wifi connection - but same wifi module was working on other TFT
so I have applied the config file again but still not success
I could recover only after flashing again the 3.03 FW which means the FW can easily get corrupted

Valcob, 2019-03-23:
Yeah have the same issue with the FW the text on UI buttons is gone and it doesn't matter if I reflash the config or both config and TFT bin :(

luc-github, 2019-03-23:
I have dumped the MKS WIFI FW 
so actually the MKS WIFI module is like this:
I needed to put IO0 to GND to connect, so IO0 is GPIO0 
and I need to connect my Serial adapter RX -> WIFI RX (which is TX) and TX -> WIFI TX (which is RX)
IO1 is GPIO4 - verified with multimeter like all others pins

GPIOO |GPIO4 | GND | VCC
-- | -- | -- | --
RST | __TXD__ | __RXD__| NC

not sure what GPIO4 is used for
[MKSWIFI.zip](https://github.com/luc-github/ESP3D/files/2999714/MKSWIFI.zip)

I did not flash back to verify yet, but as I was able to dump the FW. there is no reason flash the dump won't work

luc-github, 2019-03-23:
if you want to try, the command to dump esp8266 FW is like this :
`esptool.py --port COM59 --baud 115200 read_flash 0x00000 0x400000 backpack.bin` under windows
and `esptool.py --port /dev/ttyUSB0 --baud 115200 read_flash 0x00000 0x400000 backpack.bin` under linux/OSX

Valcob, 2019-03-23:
@luc-github please be aware that the dump has your wifi creds ;)

luc-github, 2019-03-23:
It is test network so it is ok

jamesarm97, 2019-04-12:
Not sure if this helps, it is the "protocol" used when connecting to the ESP over TCP but don't know if it is the same that goes out the serial port. The ESP may just be passing the data straight through to the TFT or may be interpreting it and sending other data. http://osoyoo.com/wp-content/uploads/2016/07/MKS-TFT-Protocol.pdf

luc-github, 2019-04-15:
Thank you for sharing -interresting indeed, I was wondering what 997 and 998 commands stand for

IrishMoss, 2019-04-20:
> yeah I saw that too and it works I can confirm
> [image]
Just curious, have you been able to get this plugin working in cura 4.0?

luc-github, 2019-05-29:
I have sent email to MKS team (github one) to get help on protocol, let see what they answer.

luc-github, 2019-06-17:
I was able to dump communication between TFT and MKS-HLKWIFI module (not esp based) and between TFT and MKS wifi TFT (ESP based) 
1 - both use different protocols :
MKS-HLKWIFI use at commands
MKS -WIFI-TFT use binary and text commands

2 - they use differents pins for RX/TX which means we cannot use MKS-HLKWIFI protocol with MKS-WIFI-TFT module, but MKS-HLKWIFI should be easily replaced by any ESP module

3 - MKS-HLKWIFI protocol looks easy but slower to connect than MKS-WIFI-TFT protocol
4 - Once connected, MKS-HLKWIFI protocol is quiet, only commands are bridged, unlike  MKS-WIFI-TFT which is pretty verbose
5 -  I was not able to use MKS-WIFI-TFT as a bridge, unlike the MKS-HLKWIFI, I suspect it is only to be used on specific cloud server when MKS-HLKWIFI as no restriction.

Next step will be to add the MKS-HLKWIFI protocol support in ESP3D and test with a standard esp8266.

here the  MKS-HLKWIFI communication for reference
```
[TFT]	at+netmode=2\r\n
[WIFI MODULE]	at+netmode=2\r
[WIFI MODULE]	\r\n
[WIFI MODULE]	ok\r\n
[TFT]	at+dhcpc=1\r\n
[WIFI MODULE]	at+dhcpc=1\r
[WIFI MODULE]	\r\n
[WIFI MODULE]	ok\r\n
[TFT]	at+wifi_conf=YOURSSID,auto,YOURPASSWORD\r\n
[WIFI MODULE]	at+wifi_conf=YOURSSID,auto,YOURPASSWORD\r
[WIFI MODULE]	\r\n
[WIFI MODULE]	ok\r\n
[TFT]	at+net_ip=192.168.3.100,255.255.255.0,192.168.3.1\r\n
[WIFI MODULE]	at+net_ip=192.168.3.100,255.255.255.0,192.168.3.1\r
[WIFI MODULE]	\r\n
[WIFI MODULE]	ok\r\n
[TFT]	at+net_dns=192.168.3.1,8.8.8.8\r\n
[WIFI MODULE]	at+net_dns=192.168.3.1,8.8.8.8\r
[WIFI MODULE]	\r\n
[WIFI MODULE]	ok\r\n
[TFT]	at+remotepro=tcp\r\n
[WIFI MODULE]	at+remotepro=tcp\r
[WIFI MODULE]	\r\n
[WIFI MODULE]	ok\r\n
[TFT]	at+mode=server\r\n
[WIFI MODULE]	at+mode=server\r
[WIFI MODULE]	\r\n
[WIFI MODULE]	ok\r\n
[TFT]	at+remoteport=8080\r\n
[WIFI MODULE]	at+remoteport=8080\r
[WIFI MODULE]	\r\n
[WIFI MODULE]	ok\r\n
[TFT]	at+uart=115200,8,n,1\r\n
[WIFI MODULE]	at+uart=115200,8,n,1\r
[WIFI MODULE]	\r\n
[WIFI MODULE]	ok\r\n
[TFT]	at+net_commit=1\r\n
[WIFI MODULE]	at+net_commit=1\r
[WIFI MODULE]	\r
[WIFI MODULE]	\r\n
[TFT]	at+net_wanip=?\r\n
[WIFI MODULE]	at+net_wanip=?\r
[WIFI MODULE]	\r\n
[WIFI MODULE]	192.168.1.59,255.255.255.0,192.168.1.1\r\n
...
[TFT]	at+net_wanip=?\r\n
[WIFI MODULE]	at+net_wanip=?\r
[WIFI MODULE]	\r\n
[WIFI MODULE]	192.168.1.59,255.255.255.0,192.168.1.1\r\n
[TFT]	at+wifi_ConState=?\r\n
[WIFI MODULE]	at+wifi_ConState=?\r
[WIFI MODULE]	\r\n
[WIFI MODULE]	Connected\r\n
[TFT]	at+reconn=1\r\n
[WIFI MODULE]	at+reconn=1\r
[WIFI MODULE]	\r\n
[WIFI MODULE]	ok\r\n
```
so it should be easy to reproduce in theory

luc-github, 2019-06-17:
Support of MKS TFT is now added : https://github.com/luc-github/ESP3D/commit/1d7b3ff83f72d5bc5cc8bc0a525ceda4f2333c02

need to enable 
```
//MKS TFT WIFI support see Wiki for wiring
//#define MKS_TFT_FEATURE
```
because it is disabled by default
boot is slow due to TFT Firmware, it start from disconnected to exception then go to connected like standard MKS-HLKWIFI module, bridge is working - I did not checked web UI yet but it should be ok based on telnet tests I did

I have used a nodemcu connected according :

NC | NC| NC| NC
---|---|---|---
 TX | RX| GND| VCC

if someone could confirm it is working - it would be great

luc-github, 2019-06-18:
OK tested with webUI and some features does not work due to MKS TFT FW filter some commands
what does not work so far:
1 - positions display
M114 does not reach printer so cannot get position
2 - smoothieware may have issue as console commands are filtered (not tested yet)
3 - Printer configuration is not working as M500/M503 etc .. commands are blocked

not sure if this can be handled, may be push issue about filer in MKS as issue with some commands 
like M114

luc-github, 2019-06-24:
Close with limitation due to tft fw

closed 2019-06-24 by luc-github

ETE-Design, 2019-08-28:
So it is possible to make ESP3D work on MKS TFT2.8 with Wifi module? Can you then make some sort of autodetect, so it only show working stuff on the GUI? And is it possible to print direct from Cure with ESP3D

AnHardt, 2019-08-28:
Likeliy not [The Cure](https://www.youtube.com/watch?v=xik-y0xlpZ0), but [cura](https://ultimaker.com/software/ultimaker-cura) could be possible. (too slow ;-) )

ETE-Design, 2019-08-28:
And how to connect it to Arduino Uno to flash it? Have looked at the video - but can't figure it out...

atsage, 2019-09-25:
@luc-github Any news on this?  I got one of these modules a while ago and just now getting around to using it.  It's pretty horrible on stock firmware and ESP3D would be a huge upgrade.

luc-github, 2019-09-25:
It is working but with limitations as I wrote above

atsage, 2019-09-25:
Thanks, I thought you meant that you meant nothing worked due to firmware limitations.  I'm planning on using Marlin, so the only things that won't work are positions display and printer configuration?  If so I'll give it a shot.  I'm planning on only using the control panel and sd card panel.

slm4996, 2019-10-02:
Good day,
I have successfully flashed  ESP3D on the MKS WiFi Robin V1.0 module (https://www.aliexpress.com/item/32816882694.html)
 used by the Robin Mini board:
[image] However after enabling MKS_TFT_FEATURE the printer does not see the module as active, and as such is ignoring the commands sent. This module normally runs similiar to the MKS TFT WiFi module in that it opens a wifi to serial bbridge on <ip>:8080.

Output from the stock firmware on the TX pin to the printer board is as follows:
`
Mordor	<SanitizedWifiPassword>www.baizhon12345678f'	HJNLMuserCS1.0.2_180327⸮⸮
M27
M997
M115
⸮⸮
`

Maybe this module outputs different responses to the board to tell it that it is connected?

I am fairly new at this ESP stuff, but with a little guidance I am willing to gather any information needed to assist in getting this to work.

luc-github, 2019-10-02:
You need to set the module as  MKS-HLKWIFI module in MKS config, as I explained above  https://github.com/luc-github/ESP3D/issues/249

The protocol you use is the MKS WIFI TFT but I implemented the  MKS-HLK-WIFI protocol

slm4996, 2019-10-02:
I do not see a way to do that in the MKS_Config.txt. Maybe I just do not know the definition to add...

Here is what I have:
`
#Wifi mode (0:sta;1:ap)
>cfg_wifi_mode:0

#Wifi name
>cfg_wifi_ap_name:MySSID

#Wifi password
>cfg_wifi_key_code:password

#Cloud service enable (0: disabled; 1: enabled)
>cfg_cloud_enable:0
#Cloud service address
>cfg_wifi_cloud_host:www.baizhongyun.cn
#Cloud service port
>cfg_cloud_port:10086
`

slm4996, 2019-10-02:
Found it, here what you need ofr anyone else looking:

#wifi model(MKS WIFI TFT:0;  MKS HLK-WIFI:1)
`>cfg_wifi_type:1`

slm4996, 2019-10-02:
Setting the cfg_wifi_type to 1 made no difference, the printer still shows no info under the WiFi reporting and ESP3D commands are ignored.

Command triggered on the LCD (Home, Fan speed, Temp) are shown in the ESP3D console though...

luc-github, 2019-10-02:
Using hk protocol means also you use hk wiring

luc-github, 2019-10-02:
Tx and rx are not in same position depending protocol
https://github.com/luc-github/ESP3D/issues/249#issuecomment-475857956

slm4996, 2019-10-02:
The wiring is not alterable, the ESP module plugs directly into headers on the main board and after digging into the firmware and trying to change the config with stock module firmware, this board only runs in HLKWIFI mode.

This is the schematic for the board / module interface:
[image]

Since I am seeing printer commands in the ESP3D interface, I can assume that the RX pin was defined correctly (I was using -1 to let the board definitions override). I am now setting those manually and retrying.

Anyway I can have ESP3D display the full serial info it is receiving from the printer, or does the console do that by default?

slm4996, 2019-10-02:
Same results when defining the TX and RX pins for the board. I see the LCD issues commands in ESP3D, but the printer shows blanks for WiFi info and ignores all commands sent by ESP3D.

slm4996, 2019-10-02:
I have confirmed that the pin definitions / wiring are correct by flashing the printer with stock marlin firmware. ESP3D works great!

So it seems the printer ignoring the commands is due to it thinking WiFi is not connected.

luc-github, 2019-10-02:
do you have picture of your tft wifi connector and the wifi module itselft ?

atomphil, 2019-10-02:
[connector](https://www.lesimprimantes3d.fr/forum/uploads/monthly_2018_07/1289311414_mksrobin.jpg.f86625094dcdd4573d75f865820f33e1.jpg)

[the module](https://de.aliexpress.com/item/32816882694.html)

pictures: 
[top](https://55.img.avito.st/640x480/5053181655.jpg)
[side](https://56.img.avito.st/640x480/5053181656.jpg)
[bottom](https://54.img.avito.st/640x480/5053181654.jpg)

it looks like a bare ESP

luc-github, 2019-10-03:
So I suspect it ignore
```
#wifi model(MKS WIFI TFT:0; MKS HLK-WIFI:1)
>cfg_wifi_type:1
```
and only the MKS WIFI TFT protocol is enabled, I think you can confim it by checking the TX output from your board with MKS HLK-WIFI enabled  it should be same as the one you show at the top of thread

but you mentioned 
>this board only runs in HLKWIFI mode.

which is confusing because :
```
��Mordor <SanitizedWifiPassword>www.baizhon12345678f' HJNLMuser�CS1.0.2_180327⸮⸮�� M27 M997 M115 ⸮⸮
```
is not HLKWIFI  but the MKS WIFI TFT

slm4996, 2019-10-03:
That may be mistake in protocol vs module naming I guess. MKS calls the 16 pin esp12 based module MKS-HLKWIFI and the smaller 6? Pin package the mks-tftwifi.

slm4996, 2019-10-03:
Also, I'll add a little tidbit here, if you export the binary from Arduino IDE and rename it mkswifi.bin you can upload it to the stock firmware web interface or put it on your SD card and reboot the printer and it will flash it. Going back to stock is the same thing just give it the stock MKSWiFi.bin and it will revert back.

No need to take it out of the printer and wire it all up just to flash.

[56 more replies]