luc-github/ESP3D #250: ESP3D esp32 OLED compile problem

shaddow501, 2018-09-19:
Hello

I am trying to compile in order to upload to the module and I have some issues.
"sketch\wificonf.cpp:54:21: fatal error: NetBIOS.h: No such file or directory"

What am I missing here?

Is there a compiled version that I can upload to the board?

luc-github, 2018-09-19:
NetBIOS library is part  of ESP32 core, trry to update from git repo

shaddow501, 2018-09-19:
Ok managed to compile and update the Lolin-esp32 - oled.
but... it doesnt boot anymore:

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4

Any suggestions?

shaddow501, 2018-09-19:
This is the error I get when I open the com port to see if the esp32 booting.
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4

The board crash...

luc-github, 2018-09-19:
Where do you see error ? Looks like normal ESP32 output at start
Type [ESP444]RESET to be sure serial output is on

shaddow501, 2018-09-19:
I see this error after I flash the module and opened serial terminal.

luc-github, 2018-09-19:
What the output at boot  if you connect gpio15 to gnd ?

shaddow501, 2018-09-19:
If I program on the board other software the board works,
With your software I should see something right (on the LCD or serial output).
The LCD stay black..

luc-github, 2018-09-19:
Did you the 2 things I wrote ?

shaddow501, 2018-09-19:
You mean to connect GPIO15 to GND? What do you would like to know what should I get?

luc-github, 2018-09-19:
Read what I wrote
Do the 2 things I asked you to do
Answer the question I have asked

luc-github, 2018-09-19:
I just compile and flashed and got : 
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
M117 
M117 Connecting.   
M117 Connecting..  
M117 Connecting... 
```

luc-github, 2018-09-19:
if I connect GPIO15 to GND
8s after boot/reset I got : 
```
M117 
M117 Connecting.   
M117 Connecting..  
M117 Connecting... 
M117 Connecting....
M117 Connecting.   
M117 Connecting..  
M117 Connecting... 
M117 Connecting....
```
No more boot loader

shaddow501, 2018-09-19:
To make things more clear I will explain what I do.
I use Lolin-esp32 board - as like this one - https://diyprojects.io/unpacking-wemos-esp32-lolin-clone-0-96-ssd1306-monochrome-oled-display/

I have downloaded the ZIP of the ESP 2.0 branch and extract the libs into Arduino libs folder.
Opened the basicesp3d.ino file, selected the WEMOS Lolin32 and checked the compilation before uploading to see if something failed.
at first it didnt find all the cpp and h files so I have moved all the cpp and h files to be in the same folder as the basicesp3d.ino was, than it compiled till the first issue of the NetBIOS.h that was missing and you offered to update the ESP32 hadware.
So I went to boards in the arduino and updated the esp32 to the latest.
After that it compiled successfully. That was the time to upload to the module. So now it was uploaded and what I get that the display is black no image showing, and when I open the serial console I see this:

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
€ן¿��x�Vֹqets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4

I dont see anything else...

luc-github, 2018-09-19:
2.0 has switches to disable output OLED / Serial etc
typing after boot (8s)
[ESP444]RESET 
ensure the Serial Switch output is on

so you confirm you do not ground the GPIO15 and you confirm you did not type [ESP444]RESET
I am correct ?

shaddow501, 2018-09-19:
Dear Luc the GPIO15 is connected to a MicroSD reader for future development... when I shortcut the GPIO 15 to the GND nothing happens, also nothing happen when I send [ESP444]RESET via the serial console.

Where are those switches that disable OLED and Serial?
Also found one issue with the oled Pins -  SSD1306 esp_display(OLED_ADDR, OLED_PIN_A, OLED_PIN_B);
Which in my board they are - 5,4 - #define OLED_PIN_A  5
#define OLED_PIN_B  4 instead of the originally in the ino that was -  #define OLED_PIN_A  4
#define OLED_PIN_B  15.

luc-github, 2018-09-20:
1 - so you see the first part is bootloader and not an error 
2 - as you are not able to reset try 
https://github.com/luc-github/ESP3D/blob/2.0/src/config.cpp#L150 
```
bool  CONFIG::is_locked(byte flag){
    return ((CONFIG::output_flag & flag) == flag);
}
```
change to 
```
bool  CONFIG::is_locked(byte flag){
   return false;
 //return ((CONFIG::output_flag & flag) == flag);
}
```
so no output will be blocked including serial

3 - about oled PINS well Pinout depend of each manufacturer I just had one embedded and one external
so yes you may need to adjust oled pinout

4 - on some first boot at least on esp8266 the SPIFFS formating can take a while, so be patient, if you change code above you will see output for sure

shaddow501, 2018-09-20:
Dear Luc

after the change it works -

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
M117 Brand
M117 Connecting.   
M117 Connected
M117 Connecting.  
M117 Connecting.. 
M117 Connecting...
M117 Connecting.   
M117 10.0.0.197

Does this change affect the disable serial function or can I still able to disable serial?

luc-github, 2018-09-20:
Now type The ESP reset command then you can undo the changes and reflash

shaddow501, 2018-09-20:
Dear Luc

In the old esp3d application I have made this argument:

 if (digitalRead(12) == LOW)
      {
        web_interface->blockserial = true;
  while (Serial2.available() >0) {
    char inByte = Serial2.read();
    ESP_SERIAL_OUT.write(inByte);
  }
    }
    else
    {
    web_interface->blockserial = false;

this allowed me to block the esp3d serial and enable the printer serial port.

What should I do with the new code?

shaddow501, 2018-09-20:
So to put all of this under the process?

luc-github, 2018-09-22:
I guess isue can be closed

closed 2018-09-22 by luc-github