luc-github/ESP3D #251: Unable to print from SD card and corrupted gcode

theman00011, 2018-09-22:
I've been trying to upload and print a gcode file in 8.3 format, but every time I upload a file, it never shows up in the files list. If I run a M20 command, it shows the file on the SD card but I'm unable to run it. Furthermore, when I pull out the SD card from the printer and I compare the SHA-1 hashes of the file I tried to upload and the one I pull from the SD card, they are completely different. It's also not a small error either, the original file is 317,267 bytes and the one I pulled from the SD card is 312,050 bytes. I've tried both the 1.0 firmware and the 2.0 dev firmware without success on Marlin printer firmware.

theman00011, 2018-09-22:
Yes, right now it is an uploading issue but the couple times I did get a print to start, it was very erratic and just grinded the filament at max speed until I shut it down. I am almost certain the gcode is good but we can cross that when we get there.

I tried 1.0, 2.0, 2.1 and 2.0 with async & async UI without success. 

I believe the picture was on 1.0 but it's possible I'm wrong.

Marlin version is 1.0.0

Baud rate I use is 250000

The only thing plugged into the printer is the SD card.

luc-github, 2018-09-22:
For printing ESP3D has nothing to do with printing process - it just launchs SD  print command to Marlin - print is 100% handled by Marlin 

if you says GCODE is correct on SD then it is not ESP3D issue - the Upload can be, but printing no, once print is launched it is out of ESP3D.

You can verify that by manually copy GCODE file to SD and print from SD from printer then launch another print from ESP3D to compare.

Can you give me the exact version of Marlin you use?

I will try to see if there was some change since I tested and try again - Marlin code is always changing and I do not really always follow the changes in real time as I do not used Marlin

theman00011, 2018-09-22:
This is the output of M115
`FIRMWARE_NAME:Marlin V1; Sprinter/grbl mashup for gen6 FIRMWARE_URL:https://github.com/ErikZalm/Marlin/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:ROBOR1PLUSV2 EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000`

I reformatted the SD card as FAT instead of FAT32 to see if that makes any difference. I'm currently doing a upload test with Simplify3D to make sure that works as expected. (It's so slowwww) If so, I'll plug in the ESP8266 and try it again.

luc-github, 2018-09-22:
This FW is pretty old I am surprised ESP3D can list files properly - I will do some tests on a system to check

theman00011, 2018-09-22:
That bug you linked in the Marlin repo looks very similar to my problem but I am on a much older version and that is a problem with a recent commit. I dug into my Marlin firmware and found the only place that error is called is the get_command function in Marlin_main.cpp if you want to see the code that throws the error. The line that throws that error is `SERIAL_ERRORPGM(MSG_ERR_LINE_NO);`

luc-github, 2018-09-22:
Yes but at that time I had no info about your FW version. I was guessing with info I had

theman00011, 2018-09-22:
Right. Is it possible to put the g-code in the SPIFFS filesystem and just stream the commands to the printer from the ESP8266? It would save a lot of time from not having to put the g-code on the printers SD card.

theman00011, 2018-09-22:
Also, the problem picture I linked to in the original post happened again with a fresh install of 2.0. I started ESP3D, made sure the SD was clear, started the upload, it stayed at 100% for about a minute and a half (Way too short for the full upload), and then it started trying to erratically print automatically and the files box was filled with temperature data.

theman00011, 2018-09-22:
Update: I upgraded to Marlin 1.1.9 and the problem is still there. Uploaded a file, progress bar started at 70%, progressed to 100% in a minute or so before starting printing automatically and erratically so it isn't unique to that firmware version. This is the new output of M115.

`FIRMWARE_NAME:Marlin bugfix-1.1.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Robo R1+ EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:THERMAL_PROTECTION:1`

luc-github, 2018-09-23:
The print start automaticaly ? that is weird, it means the commands are not saved to SD but send to printer

Can you share the file you said it is only  310KB

theman00011, 2018-09-23:
Sure, it's attached. That's why I have the theory that ESP3D thinks it is still uploading but in reality, it has exited save mode for some reason and is in normal mode.

[openlock.zip](https://github.com/luc-github/ESP3D/files/2408523/openlock.zip)

luc-github, 2018-09-23:
thank you I have setup a system - just need to flash the Marlin FW on it - so I will have a look today

theman00011, 2018-09-23:
Sounds good. I thought the firmware update would fix it but I guess now. (Also, the new auto bed leveling is so awesome, I'm glad I decided to upgrade Marlin. The 1.0.0 bed leveling didn't do much.)

luc-github, 2018-09-23:
Can reproduce issue with ESPD2.0  : 
after 3min upload I got :  
```
Error:No Checksum with line number, Last Line: 0
Resend: 1
```
so no line was accepted

[image]

what ever I type I got same message:
```
M29
Error:No Checksum with line number, Last Line: 0
Resend: 1
```

need to check how to reset this - a part of restart printer board

luc-github, 2018-09-23:
and of course :
```
M20
Begin file list
OPENLOCK.GCO 0
End file list
```

theman00011, 2018-09-23:
Great, at least it is reproducible. Not sure what is causing it though and why it's only a problem I've run into.

luc-github, 2018-09-23:
I think it is not same issue as you got initialy 
if I comment lines as suggested in link above I get :
[image]

Now I will check file integrity

theman00011, 2018-09-23:
Comment which lines? Because in the Marlin link you sent, that is for a recent build and they suggest to comment some lines in queue.cpp which doesn't exist in either version of Marlin that I tested with.

luc-github, 2018-09-23:
yes not same file:

[image]

Stripping look correct at first glance :

[image]

but actually some parts are corrupted:
[image]

I will try to see if corruption come from upload part or serial transfert part

theman00011, 2018-09-23:
Ah, okay. How hard would it be to implement a checksum compatible with Marlin that could be toggled? Since it appears the checksum is needed or it is possible to get a corrupt print.

luc-github, 2018-09-23:
Implement check sum is not really hard:
1 - before upload launch M110 N0
2 - for each line do checksum
This will slow down more the transfert

but before that I need to be sure upload is correct, because if upload is corrupted checksum will be Ok as done on ESP3D side with wrong content

luc-github, 2018-09-23:
On your side I suggest you to check your slicer - line 27 is not correct

luc-github, 2018-09-23:
Ok Upload is correct 
so issue is either in stripping method or in Serial transfert - I check

luc-github, 2018-09-23:
So far my stripping is correct (I am relieved ^_^)

So time to implement checksum and line numbering for test

luc-github, 2018-09-23:
Ok I have a test code working but I need to fine tune several parts
the issue is not generating checksum
the issue is catch error properly
Unlike Repetier, Marlin validate command with `ok` without line numbering
So need to guess which `ok` is for what command or you get discrepancy when  check sum error happen
for example : checksum error is on line 250 but you already sent 260 lines..., hard to rollback ...

Anyway I got on download Ok - may be by luck so I need to test more and adjust some serial buffer purge

Will continue on it tomorrow

theman00011, 2018-09-23:
`{travel_speed}` is still shown as available in the Cura definitions for some printers so I would assume it is still a valid variable unless some of the docs just aren't updated.

Were you able to successfully implement the checksum and line numbering?

luc-github, 2018-09-24:
Well if variable is not replaced then there is an issue - printer does not understand `{travel_speed}`

Yes, it is implemented, I have uploaded test file several times successfuly,  I am even able to upload 1.3MB file without error:
[image]

But I need to strees more, because I got at the begining, several times, an issue with Marlin FW but I am not able to reproduce now:
when start using M28
normal output is (this is without checksum /numbering for demo):
```
M28 /test.gco
echo:Now fresh file: /test.gco
Writing to file: /test.gco
ok
```
but I got several times :
```
M28 /test.gco
ok
```
so file is not created and following commands to be saved are actually seen as direct command

I need to check now repetier / smoothieware / MarlinKimbra to verify if I can secure such issue as for now I only check `ok` answer

luc-github, 2018-09-24:
Repetier is now properly tuned :
communication is easier to handle but need to handle extra answer `wait` so cannot really purge serial as always have `wait`
```
Purge Serial

wait
wait

Purge Serial
N5836 G1 X112 Y152 E1073.97502*98
ok 5836

got ok
Purge Serial
Purge Serial
N5837 G1 X112 Y144.45 E1074.39364*78
ok 5837

got ok
Purge Serial
Purge Serial
N5838 G1 X116 Y144.45 E1074.61543*75
ok 5838

```
also sometime the number come after the `ok` so need to cummulate answers to handle properly:
```
N432 G0 F3600 X131.877 Y146.019*78
ok  <- first part not recognised because no line number
ok 432 -<first part + second part

got ok <- ok + line number is recognised
```
I need to redo Marlin with latest code  then Marlin Kimbra  and Smoothieware

luc-github, 2018-09-24:
For Marlin currently it need to comment :
```
#if ENABLED(SDSUPPORT)
        else if (card.saving) {
          gcode_line_error(PSTR(MSG_ERR_NO_CHECKSUM));
          return;
        }
      #endif
```
because M29 should not use checksum/line number or you cannot recover

luc-github, 2018-09-24:
Looks like Marlin Kimbra has same issue

luc-github, 2018-09-25:
Smoothieware does not have such issue but reset Numbering seems working only once

theman00011, 2018-09-25:
There must be a way to do it in ESP3D to fix it since Simplify3D was able to upload just fine without editing Marlin, which I would rather not have to do.

luc-github, 2018-09-26:
Ok for smoothieware documentation is wrong the correct syntax to reset is :
```
N{New Line Position} M110
```
so need to adjust code to follow :
1 - smoothieware 
which is quiet (no` wait` when idle), use `N{New Line Position} M110` to reset, and resend error is `rs <line number>`, `M29` does not need checksum, send `ok` without line number

2 - Marlin Kimbra
which is not quiet (send `wait` when idle), use `N{New Line Position} M110` to reset, resend error is `Resend:<line number>` ,  `M29`  need checksum,  `ok` does not use line number

3 - Marlin 
which is quiet (no `wait` when idle), use `N{New Line Position} M110` to reset, resend error is `Resend:<line number>` ,  `M29`  need checksum,  `ok` does not use line number

4 - Repetier
which is not quiet (send `wait` when idle), use `N{New Line Position} M110` to reset, resend error is `Resend:<line number>` ,  `M29`  does not need checksum,  `ok` use line number

luc-github, 2018-09-26:
additionally I will add a detection in close function using M29 to catch what is the expected number, in case of issue, and use this number / checksum for M29, this will be also added to an [ESP4XX] command to clear any faulty upload manualy - so all situations should be handled

luc-github, 2018-10-03:
I am travelling right now, so this slow down a bit my availability to code
I have already identified almost all impact/actions to be done, still need to see about async webserver but will start with sync because better not to use async on ESP3D with ESP8266 currently
So should have some code update soon - I hope

luc-github, 2018-10-28:
I am back and will work on it this week - sorry for the delay but vacations make me busy  ^_^

AdrienHunter, 2018-10-28:
Many thanks for your status update, right in time as my new printer will be ready soon :-) Appreciate your work there. For me, except of the SD-Card upload, everything else is working fine :-)

luc-github, 2018-10-30:
Code is done and I am now testing it -  for some reason M110 N0 seems not working during my tests on Marlin Kimbra - so I changed the approach, instead of resetting the line number, I do a line number detection using a preleminary  `N1 M29 Nx` command to get the correct expected line number send back by error message. 
Doing like this I am able to do several upload without issue

I will update my marlin / marlin Kimbra / repetier / smoothieware system tests with latest FW and check the fix, if Ok I will push the code to repo

luc-github, 2018-10-31:
For repetier there are 2 cases : 
`Resend:XXX` and `skip XXX` 
Resend:XXX is used when checksum is incorrect and skip XXX is used when Numbering is incorrect...

luc-github, 2018-11-01:
Ok after some runs on different FW I reach a working way for upload
1 - closing with M29 and checksum and line numbering detection, for repetier if failed do M29 without checksum
if failed abort upload
2 - Reset line numbering to 0
if failed abort upload
3 - Send M28 command with line 1 and check sum
no autodection of line numbering as defined by reset
if failed abort upload
4 - send each line with incremented line number and checksum
no autodection of line numbering as defined by reset
if failed abort upload
5 - send M29 with line numbering and checksum 
if failed for repetier do M29 without line numbering/checksum

So it should cover all case, the trickiest is repetier as more test case than others FW
So far so good
Need to pass all FW test and should be Ok

theman00011, 2018-11-03:
Trying to upload a gcode in 8.3 format I get 
`    Error 0 : `
and no gcode file ever appears

Even worse, every 10 or so seconds I get
`Connection lost for more than 20s`

When using the latest 2.1 webui .gz file and no gcode file ever shows up.

Also, the printer seems to be getting spammed with debug information.
[image]

luc-github, 2018-11-03:
why did you updated to Web-UI 2.1 ? the Fix in in ESP3D firmware not in WebUI
ESP3D 2.1 has a heartbeat feature but it is not in public repo,  only on GRBL_ESP32 repo, this what you seem, to disable it go to WebUI preferences and disable monitoring.

Please confirm your Marlin Version and your ESP3D Version

theman00011, 2018-11-03:
Because it was the only UI that had an index.html.gz file updated in the past 5 months. Which UI should I be using?

ESP3D: `UI: 2.1b8 / FW: 2.0.0.c21`
Marlin: Marlin bugfix-1.1.x

luc-github, 2018-11-03:
But good catch seems WebSocket do debug message and it is not disabled by default -
 it is weird I do not have it on my side

luc-github, 2018-11-03:
can you give Marlin version ? not branch 
I tested on Marlin 1.1.9

luc-github, 2018-11-03:
Also what size of the file you upload - so I can reproduce

theman00011, 2018-11-03:
I believe it's on Marlin-bugfix-1.1.9 since that is what is in my Arduino library. M115 just gives Marlin bugfix-1.1.x

I downgraded to 2.0 to fix the disconnecting problem but that still leaves the debug message spam and upload error. The file is only 310KB and it's the same error on UI 2.0.

luc-github, 2018-11-03:
Debug message come from Websocket library - I just pushed a commit disable it in code and websocket library itself - can you give a try to see if you still have these debug message - I cannot see them

I just uploaded a 1.16MB file on Marlin 1.1.9 - 1.1.9 was released 2 month ago (31th July)  so bug fix 1.1.X should identical as last update was 1 month ago - this is the one you use ?

Did you do any specifics settings for SD in configuration ?

luc-github, 2018-11-03:
I am guessing the issue come from the debug message not Marlin - so in theory once no more debug output it should be Ok
I have enclosed my 2 tests files to verify also issue do not come from unexpected line in GCODE file
[test.zip](https://github.com/luc-github/ESP3D/files/2544758/test.zip)

theman00011, 2018-11-03:
Still getting the debug messages on version UI: 2.0c30 / FW: 2.0.0.c22

luc-github, 2018-11-03:
did you updated the arduinoWebSockets library ?

theman00011, 2018-11-03:
It's version 2.2.1

luc-github, 2018-11-03:
In theory should be 2.1.2 so I am not sure where this 2.2.1 come from.

Anyway as wrote in my readme, please use the one in my repo.
Also just now I have just hard coded the define to be sure is not enabled
[image]

theman00011, 2018-11-03:
I meant 2.1.1 and edited it after

luc-github, 2018-11-03:
please use the one from my repo - it is the 2.1.2 with also the `#define NODEBUG_WEBSOCKETS` to be sure

theman00011, 2018-11-03:
Using the libraries in the repo fixed the debug problem. You might want to mention that in the install instructions since I don't see anywhere where it says to copy those libraries over. 

But I still get the `Error 0 :` with your files.

luc-github, 2018-11-03:
https://github.com/luc-github/ESP3D/tree/2.0#installation but there are some typo 

I do not get it - i woks on my side

theman00011, 2018-11-03:
Yeah, I'm not sure. Everything else seems to work, I can send commands and all but I can't upload files. Is it supposed to be uploading them to SPIFFS or to the printer SD card?

luc-github, 2018-11-03:
last way to check : 
in config.h enable debug and SPIFFS serial
[image]

1 - flash the ESP
2 - upload the t1.g I sent you
it is 1 line of GCODE - it is enough to check
3 - once failed go to ESP3D FS and download log.txt
[image]

send to me the log.txt

luc-github, 2018-11-03:
The upload is on SD card

[44 more replies]

closed 2018-11-04 by luc-github