luc-github/ESP3D #254: How to connect esp3d to SD card of full graphic smart lcd controller Rep rap?

amrithmmh, 2018-10-04:
I have a full graphic smart controller is it possible to browse , upload the file and print using the sd card connected to the full graphic controller?

I loved esp3d but adding an extra sd card makes it kind of unusable for me

luc-github, 2018-10-04:
Direct connection is not possible as SD reader is used by printer board but there is a small board under development from panucatt devices that allows that, it works very well. 
I cannot tell when it will be sell, because I have no idea, but I think it should be soon as latest prototype I got was perfect. 
Check their web site time to time it should popup soon.

amrithmmh, 2018-10-04:
what if i disable the sd card in marlin firmware and use the spi pins to connect to esp8266 nodemcu?
I am new to all this trying to learn as much 
thanks again for sharing esp3d!

luc-github, 2018-10-04:
If you disable SD then you cannot print from SD ^_^
ESP3D is not a host yet

luc-github, 2018-10-06:
Can issue be closed ?

closed 2018-10-06 by amrithmmh