luc-github/ESP3D #258: ESP3D Isnt working on Wemos D1 Pro Mini

Baileyjw13, 2018-10-09:
So I was browsing through git hub when I found the esp3d for the ESP8266 boards.  I followed the install instructions on the GitHub wiki for the esp3d but when it says to "Find the AP called 'ESP8266'" I cant. On my phone and computer I can only find a AP called "ESP_809600". I connect to this and follow the rest of the instructions but at 7.2 of this page: https://github.com/luc-github/ESP3D/wiki/Install-Instructions  
it says to connect using the default password of "12345678" but my AP never asks for a password. 
I have downloaded all the right software and plug-ins.
On the Arduino IDE I have the ESP by Community, and have selected :

Board : LOLIN(WEMOS)D1 Pro Mini(the board Im using)

Can someone please help me as I am lost.

Bailey

p.s the code uploads properly and does the sort-job

luc-github, 2018-10-09:
when your ESP is connected to your computer, what the serial output say ?
the D1 Pro Mini is the 16MB flash - I am correct ?

Baileyjw13, 2018-10-09:
well ive found out what the problem was. So I have my wemos d1 mini connected to my pc via a random micro usb i found in the draw. So, next there is the bigger problem is what do i use to program the wemos? can i use this?
https://www.jaycar.co.nz/arduino-compatible-usb-to-serial-adaptor-module/p/XC4464

Bailey

luc-github, 2018-10-09:
I do not remember the pro mini has a flash button, you will have to gnd the GPIO 0 manually if you use serial adaptor on rx/tx pins I think

Baileyjw13, 2018-10-09:
ok, what is a flash button

luc-github, 2018-10-09:
it put your esp in upload mode

Baileyjw13, 2018-10-09:
Am i able to use the Wemos D1 Mini Pro? There is no wemos D1 mini pro, there is only (LOLIN)wemos D1 Mini Pro

Baileyjw13, 2018-10-09:
Am i able to program it using An arduino Uno as isp?

luc-github, 2018-10-09:
yes everything is possible - google it : you will have tutorial for each way to upload / flash

Baileyjw13, 2018-10-09:
I cant find any tutorials

luc-github, 2018-10-09:
you mean you typed in google : "flash esp8266 arduino" or even more difficult sentence "flash esp8266 with arduino uno" and you do not get anything ?
Are you serious ?

d-a-v, 2018-10-09:
On d1 mini (standard / lite / pro) there is no flash button. 
Just select your board in the tools menu and press the compile+upload button.
documentation:
https://arduino-esp8266.readthedocs.io/en/latest/
In case of issues, please go to
https://github.com/esp8266/Arduino/

luc-github, 2018-10-09:
@d-a-v Yes using usb connector, but he was asking how to flash using usb 2 serial adaptor

d-a-v, 2018-10-10:
Sorry :]
The USB adaptor's DTR pin should be then connected to its equivalent on the wemos's one (on the serial chip - ch341 on wemos/lolin d1 mini), requiring soldering.

Why not using the microusb connector on the lolin/wemos d1 mini ?

Baileyjw13, 2018-10-10:
This topic can be closed as I bought a new cable as the old one had fried
wires