luc-github/ESP3D #259: ESP8266 Arduino Compilation Error

cervecero2004, 2018-10-09:
When I try to compile I have a series of compilation errors of this style. The bookstores have already been loaded.
I do not know what else to do.

C:\Users\jpocetti\Documents\Arduino\libraries\ESP8266WiFi\src/WiFiClient.h:45:7: note:   since type 'WiFiClient' has pure virtual functions
 class WiFiClient : public Client, public SList<WiFiClient> {
       ^
invalid abstract type 'WiFiClient' for 'serverClients'

C:\Users\jpocetti\Documents\Arduino\libraries\ESP8266WebServer\src/ESP8266WebServer.h:97:40: error: cannot allocate an object of abstract type 'WiFiClient'

   virtual WiFiClient client() { return _currentClient; }

C:\Users\jpocetti\Documents\Arduino\libraries\ESP8266WebServer\src/ESP8266WebServer.h:97:22: error: invalid abstract return type for member function 'virtual WiFiClient ESP8266WebServer::client()'

   virtual WiFiClient client() { return _currentClient; }

luc-github, 2018-10-10:
What version of esp core do you use?

luc-github, 2018-10-10:
How did you installed ESP core ?

cervecero2004, 2018-10-10:
The IDEs load them in the same way as indicated in the tutoring. The
libraries were copied and pasted in the library folder. These were
recognized by the Arduino Software Is there another Flashing method of
ESP8266 that is not with Arduino? Like some .bin file?

luc-github, 2018-10-10:
I try to understand how `ESP8266WiFi` can be in `C:\Users\jpocetti\Documents\Arduino\libraries` it is not the right place to be so I am guessing you do not installed the ESP core properly

What library did you copied manually ? (do not answer you followed the tutorial it does not help, describe me what you did step by step)

cervecero2004, 2018-10-11:
Ok, my English is not the best.
1. Perform the procedure of "Additional Boards Manager URL field. (Attached
image)
2. Select the esp8266 by ESP8266 Community ESP8266
(Attached image)
3. Extract the content of ESP8266FS-0.2.0.zip into Tools (Attached image)
4. Download, decompressed and Open and compiled of the latest version of
ESP3D-1
5. I configure the Board, but when I later want to compile, I will come up
with several compilation errors.

The ESP8266 module has already been used in other applications and with
other firmware, without problems.
I use Arduino for years without problems, but with this firmware I do not
know where to continue.
}
I appreciate your help, and apologies for my English. This is not my native
language.
Im live in Argentina.

Thanks

cervecero2004, 2018-10-11:
> Puedes escribir espanol lo entiendo.
> No veo las imagenes.
> Puedo ver lo que hay en \Arduino\libraries\ ?

A. Mucho mejor en español.

Ante todo. Agradezco su ayuda. Y estoy sorprendido de la muy buena
predisposición que tiene.

Le comento que al intentar guardar las librerías al Arduino me dice que no
es un archivo válido. Por lo que estas las copie a la la ruta donde están
instaladas.
Al reiniciar. Estas aparecen como librería válidas.
Una de las alternativas en borrar todo el software y reinstalar de cero
todo.
Igualmente, cuando pueda Le mando una imagen de la ruta donde están las
carpetas de las Librerías.

Muchas gracias.

luc-github, 2018-10-11:
porque estas sorprendido ? con la gente amable soy un angel 👼 

yo creo que hay un problema en su instalacion
lo mejor es de borrar todo el software y reinstalar de cero todo come lo dices

viste esta video ? https://www.youtube.com/watch?v=pJGBRriNc9I 
esta in ingles pero es muy simple, describe como hacer la installacion para la version 0.99 / 1.0 y esp8266|
no le necessita altras librerías 
Diga me si le necessita ayuda

cervecero2004, 2018-10-15:
Ya logre funcionar. Borre arduino y todos sus registros.
Ahora conociendo el software. Pero tiene mucho potencias

luc-github, 2018-10-15:
Entonces podemos cerrar el problema?

cervecero2004, 2018-10-15:
Si. Cerrado

closed 2018-10-15 by luc-github