luc-github/ESP3D #260: Error when using Arduino uno as usb-to-serial

Baileyjw13, 2018-10-09:
So Im using an arduino uno as as a usb to serial devive. This is achieved by putting a jumper from reset to gnd on the uno. Then the Rx goes to the Rx of the wemos And the tx to the tx of the wemos, And lastly gnd goes to gnd. I get this error when I try to upload it: 

> warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

Im not sure what causes this issue, but hope it can get fixed.

Bailey.

p.s 
I have tried changing the cpu speed from 80MHz to 160MHz

Again,
Bailey

luc-github, 2018-10-10:
Look like your esp is not in upload mode . is it ?

Baileyjw13, 2018-10-10:
This is on the semis d1 pro mini

luc-github, 2018-10-10:
How did you put your esp in upload mode?

Baileyjw13, 2018-10-10:
Never mind I got it working. Was using the wrong micro USB

luc-github, 2018-10-10:
ok

closed 2018-10-10 by luc-github