luc-github/ESP3D #261: 8.3 filename

Baileyjw13, 2018-10-10:
Hi, 
Long Story short: 

when I try to upload a .gcode file from cura it gives me an error that says 
"Please use 8.3 filename only."
I think that this is an issue that needs to be resolved-and fast

Bailey

Also can someone link me a good IP camera to use from Banggood.com???

luc-github, 2018-10-11:
any IP camera that can be connect by web browser should works
Sorry I do not use Banggood.com

closed 2018-10-11 by luc-github

Baileyjw13, 2018-10-11:
Why you close it? It’s a problem that needs to be fixed

luc-github, 2018-10-11:
You want me to fix Marlin not accepting filename others than 8.3 ?

Baileyjw13, 2018-10-11:
Yes please as it is a pain changing names

luc-github, 2018-10-11:
I am not in charge of Marlin FW - I have enough with my own projects
 Why you do not ask them to fix it if it is a concern for you ?

luc-github, 2018-10-11:
Yes I support Marlin / Repetier / Smoothieware but I do not develop Marlin / Repetier /Smoothieware
Each one has their limitations/features
8.3 file name is Marlin feature/limitation not ESP3D limitation - I have added a message telling user to use 8.3 format for Marlin

Baileyjw13, 2018-10-11:
Why did you tell him to use it as 90% of the world doesn’t know how to
change file names from normal to 8.3

luc-github, 2018-10-11:
because it was raised before and they do not want to change  so I do not see the pooint to fight

Baileyjw13, 2018-10-11:
I’d love to stay and support esp3d but with all o it’s problems it’s just
not worth it

Bailey

iz3man, 2019-04-24:
WOW. Just read this. @luc-github I solute for your patience with this guy ;)
It's totally clear that this 8.3 thing is a Marlin issue, but do you maybe know which part of Marlin is responsible for that stuff, so i know how to address this issue to the Marlin developers in the most appropriate way? Thanks!

luc-github, 2019-04-24:
@iz3man it was raised already on the past and no plan to be solved because it is not a bug but a feature : https://github.com/MarlinFirmware/Marlin/issues/6165
The part is the code of SD Card support - and even 2.X has issue last time I checked
if you want long name try MarlinKimbra : https://www.marlinkimbra.it/

iz3man, 2019-04-24:
Thanks. I digged in it a little now as well. I checked MK4DUO as it's called now. Looks good, but i'm not sure yet how up2date/feature rich it is compared to marlin-2.0.x and how big the community is. 
Too sad Marlin is not addressing that 8.3 feature. It's SO old school that probably 90% of the users are not old enough to even know that stuff first hand ;)

iz3man, 2019-04-24:
Btw : ESP3D looks great!! Any ideas on adding a fileserver feature? Or is this too much for the little esp? But sorry, this is the wrong thread ....

luc-github, 2019-04-24:
what you mean by fileserver feature?

iz3man, 2019-04-25:
Some CIFS server that you can mount from a remote pc to upload your gcode to. So you have the SD card of your printer as a network share.

luc-github, 2019-04-25:
Like https://github.com/ardyesp/ESPWebDAV ?
Was in todo list for a while actually but in low priority ,curl support to automate slicer process looks covering part of the need so I forget, I need to recheck now 3.x allow more easily any addon

iz3man, 2019-04-25:
#https://github.com/luc-github/ESP3D/issues/320

iz3man, 2019-04-25:
No not like webdav, but a real CIFS fileshare you can mount on your remote PC. 
I opened a new thread to keep it organized: https://github.com/luc-github/ESP3D/issues/320

luc-github, 2019-04-25:
with webdav you can do this 
https://www2.le.ac.uk/offices/itservices/ithelp/my-computer/files-and-security/work-off-campus/webdav/webdav-on-windows-10