luc-github/ESP3D #263: ESP3D Wireless

Baileyjw13, 2018-10-10:
Is there a way to print straight from cura to esp3d? As this can just save the hassle of getting the g-code, finding the g-code, then finding out that it has to be in 8.3(long story)

I have noticed that cura has some plug-ins that could work but i just need the API key for it to work.
@luc-github do you know what the api key is so that i can test if any of the plug-ins work???

Bailey

Zefram88, 2018-10-10:
The plugin you are talking about is for Octoprint, it can't work except with it.
I'm trying to figure out the same thing, if you are on a windows machine you can always make the gcode in CURA, saving it and opening the resulting gcode on repetier-host

luc-github, 2018-10-10:
there is another projet that should meet your needs https://github.com/probonopd/WirelessPrinting/

closed 2018-10-10 by luc-github

Baileyjw13, 2018-10-10:
Does it still have the same features as esp3d

Baileyjw13, 2018-10-11:
Why can’t you add the features into esp3d?

luc-github, 2018-10-11:
why ? this project already exists
ESP3D is different

Baileyjw13, 2018-10-11:
Close this topic then

luc-github, 2018-10-11:
done already