luc-github/ESP3D #264: Can't print trough USB on CURA while ESP is connected

Zefram88, 2018-10-10:
As described in the title, I've noticed that I can't properly start a print job from CURA trough USB while the ESP module is connected and ESP3D is running.
ESP3D version 2.0.0.c20
WEB-UI version 2.1b4
Printer's firmware: MK4DUO 4.3.6

luc-github, 2018-10-10:
it depend how you connected the ESP to your board and what board you are using
For example: USB is usualy connected to Serial
if you also connected ESP to Serial (using AUX1 for example) then some boards disable the USB if USB is connected after - not all boards do
If it is the case then try to use another port for ESP but do not forget to add this port in your printer FW configuration

luc-github, 2018-10-17:
here the answer for Marlin and Mega Board https://github.com/MarlinFirmware/Marlin/issues/12064#issuecomment-430547688

I close issue as no update

closed 2018-10-17 by luc-github