luc-github/ESP3D #269: Slow & Intermittent Serial Communication

AustinSaintAubin, 2018-10-17:
@luc-github looks like I am having an issue with serial communication for the [Wemos® Nodemcu ESP8266](https://www.banggood.com/Wemos-Nodemcu-Wifi-And-ESP8266-NodeMCU-1_3-Inch-OLED-Board-White-p-1160048.html?rmmds=myorder&cur_warehouse=CN) and [Wemos® ESP32](https://www.banggood.com/Wemos-ESP32-OLED-Module-For-Arduino-ESP32-OLED-WiFi-Bluetooth-Dual-ESP-32-ESP-32S-ESP8266-p-1148119.html?rmmds=myorder&cur_warehouse=CN) that i am using. Serial communication is slow to send a receive, even at high bauds like 115200. Tested hardware with simple serial sketches... does not seem to be a hardware issue... not sure if/what I am doing wrong.

luc-github, 2018-10-17:
I do not have such issue 
Can you give all details of your configuration : hardware / software 
All fw versions and core version 
Need also all steps to reproduce issue

AustinSaintAubin, 2018-10-18:
ESP Modules Tested:
- ESP8266: [Wemos® Nodemcu ESP8266](https://www.banggood.com/Wemos-Nodemcu-Wifi-And-ESP8266-NodeMCU-1_3-Inch-OLED-Board-White-p-1160048.html?rmmds=myorder&cur_warehouse=CN) w/ [ESP3D v1.0](https://github.com/luc-github/ESP3D/releases/tag/1.0) & [ESP3D-WEBUI v1.0](https://github.com/luc-github/ESP3D-WEBUI/releases/tag/v1.0)
- ESP32: [Wemos® ESP32](https://www.banggood.com/Wemos-ESP32-OLED-Module-For-Arduino-ESP32-OLED-WiFi-Bluetooth-Dual-ESP-32-ESP-32S-ESP8266-p-1148119.html?rmmds=myorder&cur_warehouse=CN) w/ [ESP3D v2.0](https://github.com/luc-github/ESP3D/tree/2.0) & [ESP3D-WEBUI v2.0](https://github.com/luc-github/ESP3D-WEBUI/tree/2.0)

Printer:
- [Prusa i3 MK3](https://www.prusa3d.com/original-prusa-i3-mk3/) w/ [firmware v3.4.1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.4.1), uses [Einsy Rambo 1.1b](https://ultimachine.com/products/einsy-rambo-1-1)

Workstation:
- MacBook Pro (Retina, 15-inch, Late 2013) w/ macOS v10.13.6 (17G65)
  - Google Chrome Version 70.0.3538.67 (Official Build) (64-bit)

-------------

Reporduce issue:

- Connect ESP32 or ESP8266 to [Prusa i3 MK3](https://www.prusa3d.com/original-prusa-i3-mk3/) [Raspberry Pi Port](https://help.prusa3d.com/l/en/article/Loz15FAgEk-octo-print-raspberry-pi-zero-w). 
- Connect to ESP32/ESP8266 and configure ESP3D.
- Power Cycle Printer, and ESP32/ESP8266.
- Connect to ESP32/ESP8266, commands will set slugesh, UI will not update with stats from printer. Seems like commands are not being properly parsed by ESP3D, and that ESP3D is waiting for 'ok' from printer before sending next command... but something is not working as expected...

luc-github, 2018-10-18:
What ESP3D settings are you using ? 
For example did you enabled oled but oled is not connected.

luc-github, 2018-10-18:
As you use different cores and different ESP3D versions I am little bit confuse. What is the root cause but the only common part is the printer FW
So can you try use telnet and see what is happening ?
Send M105 command and see what happen
Same for commands not working in UI

AustinSaintAubin, 2018-10-18:
### Settings

[image]
[image]

----------------

### ESP3D Status
Chip ID:  XXXXX
CPU Frequency:  240Mhz
CPU Temperature:  96.1C
Free memory:  234.33 KB
SDK:  v3.2-dev-1055-g3276a1316
Flash Size:  4.00 MB
Available Size for update:  1.25 MB(Ok)
Available Size for SPIFFS:  1.31 MB
Baud rate:  115200
Sleep mode:  None
Channel:  9
Phy Mode:  11g
Web port:  80
Data port:  8888
Hostname:  ESP3D
Active Mode:  Client Station (XXXXX)
Connected to:  now
Signal:  28%
IP Mode:  DHCP
IP:  XXXXX
Gateway:  XXXXX
Mask:  255.255.255.0
DNS:  XXXXX
Disabled Mode:  Access Point (XXXXX)
Captive portal:  Enabled
SSDP:  Enabled
NetBios:  Enabled
mDNS:  Enabled
Web Update:  Enabled
Pin Recovery:  Disabled
Authentication:  Disabled
Target Firmware:  Marlin
M117 output:  Enabled
Oled output:  Enabled
Serial output:  Enabled
Web socket output:  Enabled
TCP output:  Enabled
FW version:  2.0.0.c20 ESP32

----------------

### Testing

Can send command `M117 Test01` though telnet `telnet ip_address 8888`, printer screen display `Test01`. 
Can send command `M117 Test02` though ESP3D web UI 'Commands', printer screen display `Test02`. 

When I send `M105` though ether method, I do not see a response.... no 'ok', nothing.
I have tested the serial input on both ESP32 and ESP8266 devices that I am testing with, all hardware working.

luc-github, 2018-10-18:
Telnet way is just a bridge. So if no response on telnet it is weird

luc-github, 2018-10-18:
Another test would be to connect the printer port to serial 2 usb adapter toconfirm there is no response to commands sent
is M105 supported by printer FW ?
Also disable oled output to see if affect response time

luc-github, 2018-10-18:
Also I see signal is only 28% which is pretty low. That could also be part of the issue

luc-github, 2018-10-18:
also I see this : https://github.com/prusa3d/Prusa-Firmware/issues/1232
seems there is issue with baudrate - going back to 3.4.0 solve issue

AustinSaintAubin, 2018-10-19:
Installed ESP-Link on my [Wemos® Nodemcu ESP8266](https://www.banggood.com/Wemos-Nodemcu-Wifi-And-ESP8266-NodeMCU-1_3-Inch-OLED-Board-White-p-1160048.html?rmmds=myorder&cur_warehouse=CN).
No issues at all send g-code commands and see that they where received. (tested with `M117 TESt01`)
Did not receive a single reply from the [Prusa i3 MK3](https://www.prusa3d.com/original-prusa-i3-mk3/)'s [Einsy Rambo 1.1](https://github.com/ultimachine/Einsy-Rambo/blob/1.1a/board/Project%20Outputs/Schematic%20Prints_Einsy%20Rambo_1.1a.PDF) board.
Crossed the ESPs RX/TX, saw echo of commands sent, so ESP hardware seems fine.

Connected a 3.3v TTL-USB adapter to the TX of the Prusa, saw the replies to the commands I was testing with (`M117 TEST01`, and `M105`)...  
Looks like the an issues with serial TX on the Prusa i3 MK3 Einsy Rambo 1.1 board and these particular ty of ESP8266 & ESP32 moduals... thoughts?

luc-github, 2018-10-19:
May be 5v tolerance decrease the signal too much ?

AustinSaintAubin, 2018-10-19:
Thats what I am thinking. But not sure how to fix / resolve safely.
[image]

luc-github, 2018-10-19:
First I would try with another board which is not 5v tolerant to confirm root cause
But I am not electronician expert so my guess is just newbie one ^_^

AustinSaintAubin, 2018-10-19:
Looks like the [ESP8266 I/O might be 5V tolerant](https://www.ba0sh1.com/blog/2016/08/03/is-esp8266-io-really-5v-tolerant/).
With that it looks like [ESPs might have snap back / clamping diodes on the IO pins](https://hackaday.io/page/2024-esp8266-is-5v-tolerant-after-all), that could be dropping the voltage.
Guess I will try a logic level converter and see what happens / I am able to determine.

AustinSaintAubin, 2018-10-26:
Ordered the hardware a week ago... tacking shows its expected to arrive in mid November....
I am still doing research into how to get the ESP Wemos modals that I have to work. 
How to have time this weekend for it.

AustinSaintAubin, 2018-11-07:
Received my [Geekcreit® ESP-M2 WiFi Module](https://www.banggood.com/Geekcreit-DT-06-Wireless-WiFi-Serial-Port-Transparent-Transmission-Module-TTL-To-WiFi-p-1141047.html?rmmds=myorder&cur_warehouse=CN) today.
https://github.com/SmartArduino/SZDOITWiKi/wiki/Basic-Manual
https://github.com/SmartArduino/ESP/blob/master/UserManualForESP-M_doitV1.pdf

Flashed Firmware to it: UI: 2.1b8 / FW: 2.0.0.c20
Still having odd issues with communication. 

```
start
echo: 3.4.0-1327
echo: Last Updated: Sep 12 2018 09:43:43 | Author: (none, default config)
Compiled: Sep 12 2018
echo: Free Memory: 1914  PlannerBufferBytes: 1392
echo:Hardcoded Default Settings Loaded
adc_init
CrashDetect ENABLED!
PAT9125_RES_X=0
PAT9125_RES_Y=240
PAT9125_init:1
PAT9125_RES_X=0
PAT9125_RES_Y=240
PAT9125_init:1
FSensor ENABLED
echo:SD card ok
echo:busy: processing
fsensor_autoload_check_start - autoload ENABLED

&��21.2 A:23.8
ok T:40.3 /0.0 B:20.0 /0.0 T0:40.3 /0.0 @:0 B@:0 P:21.2 A:23.7
r�z�r��0 B@:0 P:21.2 A:23.9
P'�0 P:21.0 A:23.7
M104 S40 T0
Қ�r�z�r��0 B@:0 P:21.1 A:23.7
MMU not responding - DISABLED
�T0:39.6 /40.0 @:4 B@:0 P:21.2 A:23.6
start
echo: 3.4.0-1327
echo: Last Updated: Sep 12 2018 09:43:43 | Author: (none, default config)
Compiled: Sep 12 2018
echo: Free Memory: 1914  PlannerBufferBytes: 1392
echo:Hardcoded Default Settings Loaded
adc_init
CrashDetect ENABLED!
PAT9125_RES_X=0
PAT9125_RES_Y=240
PAT9125_init:1
PAT9125_RES_X=0
PAT9125_RES_Y=240
PAT9125_init:1
FSensor ENABLED
echo:SD card ok
echo:busy: processing
fsensor_autoload_check_start - autoload ENABLED

ok
fsensor_autoload_check_stop - autoload DISABLED
ok
�	��:39.2 /0.0 @:0 B@:0 P:21.2 A:24.0
��:39.0 /0.0 @:0 B@:0 P:21.2 A:23.7
M104 S40 T0
Қ�r�z�r��0 B@:0 P:21.2 A:23.8
��r�z��r��13 B@:0 P:21.4 A:24.2
ok
ok
ok T:40.3 /40.0 B:19.8 /0.0 T0:40.3 /40.0 @:0 B@:0 P:21.3 A:23.7
L��:40.4 /40.0 @:0 B@:0 P:21.2 A:24.4
ok T:40.4 /40.0 B:19.8 /0.0 T0:40.4 /40.0 @:0 B@:0 P:21.2 A:24.1
r���:40.3 /40.0 @:0 B@:0 P:21.5 A:24.2
ok T:40.2 /40.0 B:19.7 /0.0 T0:40.2 /40.0 @:1 B@:0 P:21.4 A:24.4
ok T:40.1 /40.0 B:19.8 /0.0 T0:40.1 /40.0 @:2 B@:0 P:21.5 A:24.4
z��r��3 B@:0 P:21.4 A:24.5
r��1 B@:0 P:21.3 A:24.1
	��:40.3 /40.0 @:0 B@:0 P:21.4 A:24.5
'�0 P:21.4 A:24.7
$Ң�r�z��r��1 B@:0 P:21.5 A:24.8
ok T:40.1 /40.0 B:19.7 /0.0 T0:40.1 /40.0 @:2 B@:0 P:21.3 A:24.5
��:40.1 /40.0 @:2 B@:0 P:21.4 A:24.1
���:40.1 /40.0 @:1 B@:0 P:21.5 A:24.6
�r�z��r��0 B@:0 P:21.4 A:24.4
K��:40.1 /40.0 @:1 B@:0 P:21.6 A:24.3
Ң�r�z��r��1 B@:0 P:21.3 A:24.7
�Ң�r�z��r��1 B@:0 P:21.4 A:24.4
ok T:40.0 /40.0 B:19.8 /0.0 T0:40.0 /40.0 @:2 B@:0 P:21.3 A:24.7
ok T:40.0 /40.0 B:19.9 /0.0 T0:40.0 /40.0 @:1 B@:0 P:21.2 A:24.3
ok T:40.1 /40.0 B:19.9 /0.0 T0:40.1 /40.0 @:0 B@:0 P:21.5 A:24.4
	��:40.0 /40.0 @:1 B@:0 P:21.3 A:24.6
ok T:40.0 /40.0 B:19.9 /0.0 T0:40.0 /40.0 @:1 B@:0 P:21.4 A:24.4
� T0:40.0 /40.0 @:2 B@:0 P:21.5 A:24.4
```

AustinSaintAubin, 2018-11-07:
AH! I figure it out.... cable cross-talk.
[Having twisted wire is important.](https://youtu.be/0TTEFF0D8SA?t=328)....

closed 2018-11-07 by AustinSaintAubin