luc-github/ESP3D #273: Marlin: SD Card Filenames Must be Lowercase

AustinSaintAubin, 2018-11-07:
It appears that for the marlin firmware the filenames must be converted to lowercase when sent.
Though, this is not indicated in [the documenation](http://marlinfw.org/docs/gcode/M023.html). 
It appears that the [Prusa firmware for the LCD screen converts to lower case](https://github.com/prusa3d/Prusa-Firmware/blob/688c10247e2780f9bfcccf7dcfa774b0f666689d/Firmware/cardreader.cpp#L582), and is why printing from SD card using the LCD screen works.

Tested the following on Prusa i3 MK3 w/ FW v3.4.1

### [Select file from SD Card:](http://marlinfw.org/docs/gcode/M023.html)
```
M23 test~1.gco
> echo:Now fresh file: test~1.gco
> File opened: test~1.gco Size: 1919151
> File selected
```
If the command is sent as `M23 TEST~1.GCO`, the command will fail.

### [Print selected file from SD Card:](http://marlinfw.org/docs/gcode/M024.html)
```
M24
```

luc-github, 2018-11-07:
But what M20 command display?

AustinSaintAubin, 2018-11-08:
Looks like there is a lot of case sensitivity with Prusa's Marlin firmware....
```
m20
echo:Unknown command: "m20"(2)
```

```
M20
Begin file list
/STOCK/MK3_PL~1.GCO 319820
/STOCK/MK3_PL~2.GCO 23525291
/STOCK/MK3_PL~3.GCO 10256022
/STOCK/MK3_PL~4.GCO 1163600
/STOCK/MK3_PL~5.GCO 3771043
/STOCK/MK3_PR~1.GCO 409041
/STOCK/MK3_P~33.GCO 17819561
/STOCK/MK3_P~38.GCO 224162
/STOCK/MK3_P~42.GCO 25074366
/STOCK/MK3_P~46.GCO 5966830
/STOCK/MK3_P~50.GCO 9991779
/STOCK/MK3_P~54.GCO 839525
/STOCK/MK3_P~58.GCO 21788066
/STOCK/MK3_P~62.GCO 2206259
/STOCK/MK3_P~66.GCO 3552905
/STOCK/MK3_P~70.GCO 7586128
PRINT~1.GCO 12739152
/BENCHM~1/3DBENC~1.GCO 3670093
/BENCHM~1/TEST_Y~1.GCO 1247554
/BENCHM~1/RIPPLE~1.GCO 1064312
/CONTAI~1/SD_HOL~1.GCO 130043
/CONTAI~1/SD_ADA~1.GCO 1132984
/CONTAI~1/SOAP_H~1.GCO 5925818
/CONTAI~1/MASTER~1/MASTER~1.GCO 16236540
/CONTAI~1/MASTER~1/MASTER~2.GCO 9633713
/PARTS-~1/HEATED~1/REPRIN~1.GCO 1308244
/PARTS-~1/HEATED~1/PRUSA_~1.GCO 1666128
End file list
```

AustinSaintAubin, 2018-11-08:
Also worth notting is that the tempature output for the Prusa i3 firmware is different when heating up for a print.

```
M23 /print~1.gco
ok
M24
ok
fsensor_autoload_check_stop - autoload DISABLED
echo:Now fresh file: /print~1.gco
File opened: print~1.gco Size: 2452280
File selected
T:19.37 E:0 B:18.0
T:19.46 E:0 B:18.0
T:19.91 E:0 B:18.0
T:20.66 E:0 B:17.9
```

luc-github, 2018-11-08:
Make no real sense to list files names upper case but accept only lower case file name

luc-github, 2018-11-08:
So it never display target temperature ?

luc-github, 2018-11-08:
Not recognising Gcode with lower case is not prusa specific from another hand many FW behave same

AustinSaintAubin, 2018-11-08:
> Make no real sense to list files names upper case but accept only lower case file name

I agree... I do not know why this is... makes no sense to be case sensitive.

------------

> So it never display target temperature ?

It does after the target temp is reached.

```
ok T:240.2 /240.0 B:90.0 /90.0 T0:240.2 /240.0 @:48 B@:50 P:43.1 A:38.2
ok T:240.2 /240.0 B:90.1 /90.0 T0:240.2 /240.0 @:48 B@:39 P:43.1 A:38.1
ok T:240.0 /240.0 B:90.2 /90.0 T0:240.0 /240.0 @:52 B@:34 P:43.1 A:38.1
```

-------------

> Not recognising Gcode with lower case is not prusa specific from another hand many FW behave same

GRBL is not this way. Thats what I am used to.
Whats the best workaround / solution ?

luc-github, 2018-11-08:
Need to add a complete definition of prusa FW like Marlin or Smoothieware to handle all exceptions as I guess more to come.
I do not know prusa FW so what is the gain of using it vs Marlin FW ?

AustinSaintAubin, 2018-11-09:
SD card printing, and I imagine other gains.

luc-github, 2018-11-09:
Do they plan to fix this upper/lower case issue ? M20 should be fixed or it should affect several function like M30 and M28 I guess

luc-github, 2018-12-08:
Looks like no update - did you handled ?

AustinSaintAubin, 2018-12-08:
No solution yet... no communication from Prusa....
Also, latest firmware update from Prusa for the MK3 did not contain a fix for this issue... 
Not sure if best to wait...

luc-github, 2018-12-09:
Ok if you do not mind I close issue  then - let's re open if any update

closed 2018-12-09 by luc-github