luc-github/ESP3D #274: No Printer information

thucar, 2018-11-13:
I'm probably missing something here, but not sure what.

I just installed a fresh Marlin on my RepRap printer. Installed ESP3D on a Sonoff POW and got it working. I have the UI up and running, but no information on the Printer.

My ESP3D firmware: UI: 2.0c30 / FW: 2.0.0.c23
RAMPS 1.4 + Mega2560

Baud rate matches Marlin setup (115200)
Sleep Mode: Modem
Marlin Serial port is 0

What else am I missing or overlooking?

luc-github, 2018-11-13:
Is hou

luc-github, 2018-11-13:
Is your printer also connected by USB to PC ?

thucar, 2018-11-13:
No, it is not. Just hooked up GND and TX-RX lines to Aux-1 of RAMPS as shown in the wiki

luc-github, 2018-11-13:
No command from webui works or only no feedback?

luc-github, 2018-11-13:
Looks like communication issue : 
* May be invert RX and TX on Aux1
* You can try to use another GND 
* Be sure ESP and Ramps RX/TX/GND work with an USB 2 Serial converter to check if individually communication is Ok

luc-github, 2018-11-28:
wow ! you are lucky so