luc-github/ESP3D #275: Automated card refresh

ModMike, 2018-11-21:
Hi Luc,

I have been away from my printer for a few months and have just started printing again using the Azteeg X5 mini WiFi via the WEB UI running on Marlin 2.0.x.

It is frankly amazing and I haven't felt the need to hook up Octoprint for anything whatsoever.  The only niggle I have is the the workflow is a little tedious and I was wondering if you would consider some minor changes:

1) It would be great if WEB UI would send an automatic SD card initialize command to Marlin a few seconds after startup to avoid having to do it on the printer.  WEB UI will not see the card contents until that is done.

2) It would be equally great if WEB UI would send a card refresh every time it finishes uploading or deleting a file to sync what Marlin and WEB UI see.

3) Uploading files can become tedious when each fie name needs to fit 8.3.  Is there anyway that you can auto truncate filenames?  Ex:  When selecting THE123456789.gcode for upload it would be automatically renamed  2345678.GCO before uploading.

4) I know we discussed this before but I would also like to be able to add a post printing script to simplify3Ds post processing to automatically upload a file to WEB UI when I save tool paths to disk, like I used to do for Octoprint.

I know I am asking fro a lot but it would sure make life easier when printing in Marlin.

Merci!

luc-github, 2018-11-21:
1 - as said this should be done by Printer not webUI - but this could be done of course when checking if SDcard is present -  this is because no SD Detect Pin ?

2 - the issue is same as 1 - so same solution init SD card `M21`

3 - It is not so simple - truncate follow rule : XXXX~000.XXX so when truncate need to check there is no same file and if any increase 000 until free slot - additionaly how user know `THE123456789.gcode` is actually `2345678.GCO` ? That is why I ask user to change name itself

4 - that should be already possible in theory using curl, something like :
 `curl -i -X POST -H "Content-Type: multipart/form-data" 
-F "data=@test.gco" http://mysuperserver/upload`
There is also WebDav support in plan 
But did not have time yet to check both

ModMike, 2018-11-21:
Do you have an updated version of the Azteeg firmware that includes the automator function that you could send me?

Also can I add M21 to the commands myself?  They seem to work like the macros, correct?

luc-github, 2018-11-21:
I won't comment Marlin decisions as I do not use Marlin on my printers but for what I remember SD init is also added in LCD part so I am guessing you do not have LCD neither, is that correct ?

And it seems common to have printer without SD Detect, at least `Robo3D R1` was reported here(https://github.com/luc-github/ESP3D/issues/251).

About adding M21, I will add it as option in WebUI  and in ESP3D FW 
 * for SD refresh 
 * Before upload
 At the end of upload there is a refresh so this case should be cover also 

The curl should run to 8.3 for safety,  originaly 8.3 is for M23/M29 protocol not direct SD upload - as Marlin was not the target  for this FW so I did not tested long name for Marlin, so you can give a try the 8.3 restriction is defined in Web-UI not in ESP3D

I did not worked on it since I sent you the test version 2.0 and as you can see Autostart macro is planned for 2.1  but I did not worked on it yet - because busy with others projects - hope to work on it begining of december. It is not key feature until now

You can create a M21 in macro yes - you still need to press the button in WebUI to send it

ModMike, 2018-11-26:
When using curl, how do i trigger the upload?  I imagine I put it in post processing and then do save to desktop in s3d to trigger upload?

ModMike, 2018-11-26:
> I won't comment Marlin decisions as I do not use Marlin on my printers but for what I remember SD init is also added in LCD part so I am guessing you do not have LCD neither, is that correct ?
> 

I have an LCD just trying to streamline process.  So many clicks every print fell like I am launching a space shuttle..

Works super nice though.

luc-github, 2018-11-27:
I never used and looks like documentation is hard to find (did not find), but when you press 'prepare to print' the gcode file is generated in temporary location (I do not know where) and next step is stream the file or save is specific location
so in theory as the post processing is a copy of the file to SD, I think no other action is necessary , but you may want to save the file on your PC also
Slic3r allow such post processing script also, and I did not test neither.

luc-github, 2018-12-09:
Because on topic is already addressed here https://github.com/luc-github/ESP3D/issues/270
and another is now here https://github.com/luc-github/ESP3D/issues/279

I close this as duplicate

closed 2018-12-09 by luc-github