luc-github/ESP3D #276: Excruciatingly slow WiFi upload

ModMike, 2018-11-21:
I know this has been discussed nearly 2 years ago but has there been any improvements?  Are any coming?

luc-github, 2018-11-21:
What is slow ?  What reference can you give ?
On my side it is Ok

luc-github, 2018-11-21:
Serial upload is using M28/M29 protocol over wifi
Direct SD is direct access from ESP to SD over wifi

On my side 63.54MB take 4:30 min with SPI speed 4X on SD class 10, I did not bench since a while to be honest.

I have tried Flash air Upload and it is indeed faster as it took 1:30 min, but it is different hardware and software. 

The upload to SD only depend on SD library,  SPI Speed and SD card type as the code itself  is minimal - need to test to see what parameter to tune to increase speed if possible

ModMike, 2018-11-26:
Would increasing baud rate help in any way or am I completely off base?

luc-github, 2018-11-27:
ESP is directly connected to SD card on Azteeg WiFi - serial is used for communication with board, as ESP directly write to SD, changing baud rate won't affect upload to SD

luc-github, 2018-12-09:
There are a lot of change coming with ESP8266 core 2.5 - need to test when done as currently there are too many daily change to validate anything - better to wait the release
Also this issue does not happen to this repo (no direct SD access) - only to customized code which is not here - so issue can be closed

closed 2018-12-09 by luc-github