luc-github/ESP3D #277: ESP3D v2.0 after changing web port, it will stuck at the connecting page and please wait forever.

TheFlash12314, 2018-11-29:
In the version 1, this problem does not occur.

luc-github, 2018-11-29:
reproduced and fixed - it is a bug of WebUI and syncwebserver  websocket -  good catch
you can use http://192.168.1.18:82/?forcefallback=yes to upload latest ESP3D-WEBUI 2.0 or 2.1

TheFlash12314, 2018-11-29:
After update, some of the setting boxes becomes small and blank

TheFlash12314, 2018-11-29:
[image]

luc-github, 2018-11-29:
what web-ui version do you use ? what is your browser and OS ?

TheFlash12314, 2018-11-29:
the latest web ui v2.0 chrome and firefox all same

TheFlash12314, 2018-11-29:
windows8.1

luc-github, 2018-11-29:
I do not have windows 8.1 ...

can you try the web-ui 2.1 ? It should be fixed on it 
once done go to preferences and uncheck connection monitoring or you will be disconnected after 20s
[image]

TheFlash12314, 2018-11-29:
the version v2.1 is working. thanks

closed 2018-11-29 by TheFlash12314