luc-github/ESP3D #278: Guru Meditation Error on NodeMCU-32s

kthordarson, 2018-12-03:
Everything compiles and uploads but it crashes and reboots with this error:
Other Arduino sketches work fine. Any ideas ?

load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x401262d6  PS      : 0x00060730  A0      : 0x800dd329  A1      : 0x3ffb1dc0  
A2      : 0x3ffc6400  A3      : 0x00000002  A4      : 0x3ffc5860  A5      : 0x3ffb1f60  
A6      : 0x3ffc3d20  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffb1db0  
A10     : 0x00000001  A11     : 0x3ffb1e0a  A12     : 0x3ffc3db4  A13     : 0x400f6590  
A14     : 0x3ffb239c  A15     : 0x0000000a  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000138  LBEG    : 0x400012e5  LEND    : 0x40001309  LCOUNT  : 0x800f2439  

Backtrace: 0x401262d6:0x3ffb1dc0 0x400dd326:0x3ffb1df0 0x400d7ff3:0x3ffb1f80 0x40140077:0x3ffb1fa0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4

kthordarson, 2018-12-03:
I tried changed Arduino ide settings to get esp32 core from the dev branch and that seems to work better and boots. I used this url: https://dl.espressif.com/dl/package_esp32_dev_index.json in the Arduino ide.

luc-github, 2018-12-03:
also what version of Firmware are compiling ?

kthordarson, 2018-12-03:
What firmware do you mean? I'm using esp3d from branch master, if thats what you mean.

luc-github, 2018-12-03:
master branch is 1.0 and it is not optimised for esp32
for esp32 better to use  2.0 (2.0 branch) & ESP-WEBUI (2.0 branch) or2.1 for latest updates

luc-github, 2018-12-08:
I guess issue can be closed

closed 2018-12-08 by luc-github

kthordarson, 2018-12-08:
Working fine off branch 2.0 :)