luc-github/ESP3D #280: Please add external notification

MirkoUgoliniDev, 2018-12-22:
Hi Luc,

It would be interesting to receive push notification based on errors or end of printing.
Like octoprint that has plug in for phusover. What do you think?

luc-github, 2018-12-23:
Using something like push is possible the issue is the trigger
Octoprint is the host, it is the one which knows print/error status.
ESP3D is not the host and even know what is the error status if print from SD. or even from host on some printer as error go only to host serial not necessary to ESP3D serial
The print status could be know by polling regularly a M27 
The error may be by M408 but no reason and not supported by all FW
So not easy if ESP3D is just the monitoring

From another point - 3.0 will have ESP32 acting has host so this should be part of process, good idea

For notification I would prefer to promote a free service instead of commercial one like pushover, as I do not have reason to promote a not free tool, may be using email at first, TBD
Need to dig more on push solutions

AustinSaintAubin, 2019-01-12:
Supporting Email, MQTT, and Web API POST would allow users to use just about any method of notification including [Pushover](https://pushover.net/api), [IFTTT](https://ifttt.com/), or just plan email.

luc-github, 2019-01-21:
I may go to this - as line is free - not sure for the token generation TDC - https://github.com/TridentTD/TridentTD_LineNotify

luc-github, 2019-02-26:
fo email : https://www.arduinolibraries.info/libraries/esp-mail

luc-github, 2019-03-02:
I did a POC to push notification to pushover from esp8266 and esp32 
[image]
When it is easy to implement in GRBL_esp32 and Marlin_ESP32 as they are hosts.

I still wonder how to trigger the sending of message from ESP3D standalone ?

if adding a command in slicer end script would be the easiest way - I do not see how to catch this command coming from SD / USB. 
 I am not aware of a GCODE sending command to serial for external device.

One solution would be M42 command but it does not exists on Smoothieware...
with something like `M42 PXX S255` a pin turning high or low (TBD) would trigger the send of message 
But about errors not sure can catch any if not acting as host

If someone has suggestion/idea  ?

AustinSaintAubin, 2019-03-03:
[M118 - Serial print](http://marlinfw.org/docs/gcode/M118.html) could be an option.

However, I would focus on ESP3D's detection of events, like Print Finished.

luc-github, 2019-03-11:
Ok now I have code to send notification:
1 - using pushover ( https://pushover.net/) and it cost around 5$ depending on country
need to setup APPToken + UserToken, this is mono user and mono platform
2 - using email (cost free but because cannot use TLS, only https on port 465 so on GMAIL need to lower security protection to allow less secure applications)
need smtp server + port, + login + password + destination emails
3 - using Line (https://line.me) it is fully free 
need to setup user token which include destination users

I have tested each one on ESP8266 and first on ESP32 but others on ESP32 should not be a problem as all 3 only use https, on ESP8266 just need to use latest GIT and set secure client as insecure to not add a fingerprint, like AxTLS did before and like ESP32 do.
So now just need to prepare code 😸

luc-github, 2019-03-11:
[image]

luc-github, 2019-03-14:
Pushover, Line and Email integration is now also working - I got some issue with ESP8266  and its default WiFIClientSecure : BEARSSL,,  this one seems actually unusable within a program, it can be used only in standalone because it use more than 21KB of memory... and this seems normal...

I have switched to the deprecated axTLS and it is now working properly within ESP3D

I will write the documentation on Wiki for the 3 possible notifications way and push the 3.X code  to github tomorrow for notifications

luc-github, 2019-03-15:
Wiki done: https://github.com/luc-github/ESP3D/wiki/Notifications-in-3.X
Code pushed : https://github.com/luc-github/ESP3D/commit/3b9d5302a100f2abde71bde430e1382354e9e74b

Next step is basic host for autostart and macro

closed 2019-03-15 by luc-github