luc-github/ESP3D #283: Webcam Future

mralperem, 2019-01-04:
Hello can someone please explain me how to connect my camera to ESP8266

luc-github, 2019-01-04:
You already asked question here : https://github.com/luc-github/ESP3D-WEBUI/issues/43
and currently there is no direct connection support for camera in ESP3D only IP using WebUI

luc-github, 2019-01-04:
Also ESP8266 is not great candidat for direct connection - have a look to ESP32 instead

closed 2019-01-04 by luc-github

mralperem, 2019-01-04:
Yes you are right I am sorry to ask it here also I just mixed up the links.I couldnt find the other link to reply. What I actually wanted to ask is that if it is possible to directly connect camera to ESP32 or ESP8266. Because of the price tag of IP cameras

luc-github, 2019-01-04:
have several using google you will find them average price is 10$ which is a price of cheap IP cam:
https://www.aliexpress.com/item/ESP32-CAM-WiFi-Bluetooth-module-ESP32-serial-to-WiFi-with-camera-module-ESP32-CAM-development-board/32931586411.html
https://www.aliexpress.com/item/ESP32-CAM-ESP32CAM-Module-OV2640-Camera/32956799822.html

mralperem, 2019-01-04:
I understand I will try the both options. Thank you for your reply. Really appreciated for the great work

fred2088, 2019-01-27:
will this esp32-cam work as ip-cam and usable by ESP3D ?

> have several, using google you will find them, average price is 10$ which is a price of cheap IP cam: [...]

luc-github, 2019-01-27:
in theory yes = they are IP cam that can be seen in browser so they meet the requirement

luc-github, 2019-01-27:
Need to try but video use a lot of power. Not sure can do video and handle ESP3D feature at once.