luc-github/ESP3D #284: Azteeg X5 mini Wifi not connecting

nater7, 2019-01-06:
Hi, I am trying to set up an Azteeg X5 wifi but I cannot connect.  I have seen that there is a custom ESP3D version for this board, but I cannot find any of the files.  Can you point me to the custom version so I can start over on a fresh SD card?

Thanks

luc-github, 2019-01-06:
Sorry, I do not provide the binary image/sources  for Azteeg WiFi please contact http://www.panucatt.com/ also they provide documentation : http://panucattdevices.freshdesk.com/support/solutions/articles/1000258070-setting-up-wireless-connection-on-x5mini-wifi

nater7, 2019-01-07:
Thanks a ton for getting back to me...do you know if I can just use the webif add on to create the web interface for that board running smoothiewear?

-Nate

luc-github, 2019-01-07:
No, they are not compatible, webif is for smoothieware  not esp8266
I am sure panucatt will give you the esp image

luc-github, 2019-01-13:
I guess issue can be closed

closed 2019-01-13 by luc-github