luc-github/ESP3D #285: How to install on esp32

delchrys, 2019-01-06:
HI,

i love this project but i cant seem to understand what i exactly need to do to upload esp3d to my ESP32.
i Have installed arduino
i have installed the board and tested it with a Wifiscan sketch. that works
But now i have to figure out how to upload ESSP3d sketch to my ESP32.
Cant seem to figure out where to place the folders/libraries and which package i need to download. (ESP3d or V2).

Can someone write some small steps please.
I have arduino experience but no ESP32 experience yet and cant figure it out.

Kind regards,

Bert

luc-github, 2019-01-06:
For esp32 use 2.0 branch is better.
Read the readme of 2.0 branch it describe what need to do

delchrys, 2019-01-06:
i tried but i just cant figure it out.
opened the example ino file in arduino but it doesnt compile

delchrys, 2019-01-06:
C:\Users\bertb\AppData\Local\Temp\arduino_build_893025\sketch\basicesp3d.ino.cpp.o:(.literal._Z5setupv+0x8): undefined reference to `Esp3D::begin(unsigned short, unsigned short)'

C:\Users\bertb\AppData\Local\Temp\arduino_build_893025\sketch\basicesp3d.ino.cpp.o:(.literal._Z4loopv+0x0): undefined reference to `Esp3D::process()'

C:\Users\bertb\AppData\Local\Temp\arduino_build_893025\sketch\basicesp3d.ino.cpp.o:(.literal.startup._GLOBAL__sub_I_myesp3d+0x4): undefined reference to `Esp3D::Esp3D()'

C:\Users\bertb\AppData\Local\Temp\arduino_build_893025\sketch\basicesp3d.ino.cpp.o: In function `setup()':

C:\Users\bertb\Downloads\ESP3D-2.0\ESP3D-2.0\examples\basicesp3d/basicesp3d.ino:29: undefined reference to `Esp3D::begin(unsigned short, unsigned short)'

C:\Users\bertb\AppData\Local\Temp\arduino_build_893025\sketch\basicesp3d.ino.cpp.o: In function `loop()':

C:\Users\bertb\Downloads\ESP3D-2.0\ESP3D-2.0\examples\basicesp3d/basicesp3d.ino:35: undefined reference to `Esp3D::process()'

C:\Users\bertb\AppData\Local\Temp\arduino_build_893025\sketch\basicesp3d.ino.cpp.o: In function `_GLOBAL__sub_I_myesp3d':

C:\Users\bertb\Downloads\ESP3D-2.0\ESP3D-2.0\examples\basicesp3d/basicesp3d.ino:24: undefined reference to `Esp3D::Esp3D()'

collect2.exe: error: ld returned 1 exit status

exit status 1
Fout bij het compileren voor board DOIT ESP32 DEVKIT V1

delchrys, 2019-01-06:
the libraries i have to copy to my library directory of arduino.
where do i place the src folder???

delchrys, 2019-01-06:
what first bullet?

luc-github, 2019-01-06:
First bullet of installation process in readme

delchrys, 2019-01-06:
Please follow installation of the ESP core you want to use : ESP8266 core version or ESP32 core version

delchrys, 2019-01-06:
never mind, i give up. Going for a Raspi

luc-github, 2019-01-06:
This is first line,  bullet is point after
* this is bullet

delchrys, 2019-01-06:
DNSServer (from https://github.com/bbx10/DNSServer_tng)

delchrys, 2019-01-06:
those arent added in the git?

luc-github, 2019-01-06:
That is why I ask you , you do not read 2.0 branch read me but master one

luc-github, 2019-01-06:
The 2.0 branch read me explain better for esp32 and esp3d library

delchrys, 2019-01-06:
ESP3D because it is now a library so copy project to your arduino library folder.

delchrys, 2019-01-06:
so now i copied the library folders to my library folder in C:\program files\arduino\libraries.

luc-github, 2019-01-06:
Yes this is the correct read me, esp3d is a library so copy it in library folder

delchrys, 2019-01-06:
and now i open????:::::???

luc-github, 2019-01-06:
No library folder is in my document/Arduino/libraries on windows

delchrys, 2019-01-06:
yeah sorry i meat that folder....now what?? i need to open a *.ino file toupload??

luc-github, 2019-01-06:
Yes everything is explained in readme
You can also check the travis.yml script which build everything the path are under linux but it is easy to understand all actions

delchrys, 2019-01-06:
readme is a bit difficult for a noob..lol
do i need to upload all files and folders manually????

luc-github, 2019-01-06:
You just need index.html.gz

delchrys, 2019-01-06:
thanks for the help...
So when a new version comes, can i just update the firmware through a file

luc-github, 2019-01-06:
Yes

delchrys, 2019-01-06:
so now all i have to do is to cennect to my ender 3 board.

luc-github, 2019-01-13:
I guess issue can be closed

closed 2019-01-13 by luc-github