luc-github/ESP3D #286: Control PWM pin

shaddow501, 2019-01-09:
Hello Luc

First it is not really an issue...

I am talking with the 2.1 version.
How can I implement PWM option to control a pin?
I see that there is an option only for digitalWrite 0/1, is it possible to have a guidance how to control PWM pin?

Also you set that the maximum pins are <16, I think it is better to have all the pins available <27.

luc-github, 2019-01-09:
PWM is no implemented, there was no need when I added pin control - it need additional parameter to tell FW to use analogWrite instead of digitalWrite  I think - I can add it in 3.0

yes I am rewriting 3.0 and this part is already implemented:
```
bool Hal::is_pin_usable(uint pin)
{
#ifdef ARDUINO_ARCH_ESP8266
    if  ((pin <= 5) || ((pin >= 12) && (pin <= 16))) {
        return true;
    } else {
        return false;
    }
#endif //ARDUINO_ARCH_ESP8266
#ifdef ARDUINO_ARCH_ESP32
    if  ((pin <= 5) || ((pin >= 12) && (pin <= 39))) {
        return true;
    } else {
        return false;
    }
#endif //ARDUINO_ARCH_ESP32
}
```

shaddow501, 2019-01-10:
Dear Luc

I have implemented the PWM feature if you would like to use it/check it.

the command to control the pin (if you would like to use PWM pin) is now  [ESP201]P<pin> PWM=YES V<value>.
If you would like to use it as a digital pin you can use it as before. [ESP201]P<pin> V<value>.

Also there was a need to add analogWrite lib, since ESP32 default lib doesnt have this function.
I took it from here -  https://platformio.org/lib/show/5819/ESP32%20AnalogWrite/installation
and added it to the src lib in the esp3d 2.1 folder.

The updated code is (only the part that is relevant):
#if defined(ARDUINO_ARCH_ESP32)
#include "SPIFFS.h"
**#include "analogWrite.h"**
#define MAX_GPIO 26
#else
#define MAX_GPIO 37
#endif

#ifdef DIRECT_PIN_FEATURE
    //Get/Set pin value
    //[ESP201]P<pin> V<value> [PULLUP=YES RAW=YES]pwd=<admin password>
    case 201:
        parameter = get_param (cmd_params, "", true);
#ifdef AUTHENTICATION_FEATURE
        if (auth_type == LEVEL_GUEST) {
            ESPCOM::println (INCORRECT_CMD_MSG, output, espresponse);
            response = false;
        } else
#endif
        {
			int PWM = 0;
            //check if have pin
            parameter = get_param (cmd_params, "P", false);
            LOG ("Pin:")
            LOG (parameter)
            LOG ("\r\n")
            if (parameter == "") {
                ESPCOM::println (INCORRECT_CMD_MSG, output, espresponse);
                response = false;
            } else {
                int pin = parameter.toInt();
                //check pin is valid and not serial used pins
                if ( (pin >= 0) && (pin <= 26) && ! ( (pin == 1) || (pin == 3) ) ) {
			
							parameter = get_param (cmd_params, "PWM=", false);
                            if (parameter == "YES") {
								PWM = 1;
							if (parameter != "YES") {
								PWM = 0;
							}
							}
										
                    //check if is set or get
                    parameter = get_param (cmd_params, "V", false);
                    //it is a get
                    if (parameter == "") {
                        //this is to not set pin mode
                        parameter = get_param (cmd_params, "RAW=", false);
                        if (parameter != "YES") {
                            parameter = get_param (cmd_params, "PULLUP=", false);
                            if (parameter == "YES") {
                                //GPIO16 is different than others
                                if (pin < MAX_GPIO) {
                                    LOG ("Set as input pull up\r\n")
                                    pinMode (pin, INPUT_PULLUP);
                                }
#ifdef ARDUINO_ARCH_ESP8266
                                else {
                                    LOG ("Set as input pull down 16\r\n")
                                    pinMode (pin, INPUT_PULLDOWN_16);
                                }
#endif
                            } else {
                                LOG ("Set as input\r\n")
                                pinMode (pin, INPUT);
                            }
                        }
						int value;
						if (PWM == 0) {
                        value = digitalRead (pin);
						}
						if (PWM == 1) {
						value = analogRead (pin);
						}
                        LOG ("Read:");
                        LOG (String (value).c_str() )
                        LOG ("\n");
                        ESPCOM::println (String (value).c_str(), output, espresponse);
                    } else {

                        //it is a set
                        int value = parameter.toInt();
                        
						if (PWM == 1){ 
						  if ((value >=0) && (value <= 255))  {
                            pinMode (pin, OUTPUT);
                            LOG ("Set:")
                            LOG (String ( (value == 0) ? LOW : HIGH) )
                            LOG ("\r\n")
                            analogWrite (pin, value);
							ESPCOM::println (OK_CMD_MSG, output, espresponse);
							} else {
                            ESPCOM::println (INCORRECT_CMD_MSG, output, espresponse);
                            response = false;
							}
						} else {
						//verify it is a 0 or a 1
							if ( (value == 0) || (value == 1) ) {
                            pinMode (pin, OUTPUT);
                            LOG ("Set:")
                            LOG (String ( (value == 0) ? LOW : HIGH) )
                            LOG ("\r\n")
                            digitalWrite (pin, (value == 0) ? LOW : HIGH);
                            ESPCOM::println (OK_CMD_MSG, output, espresponse);
							} else {
                            ESPCOM::println (INCORRECT_CMD_MSG, output, espresponse);
                            response = false;
							}
						}
					}
                } else {
                    ESPCOM::println (INCORRECT_CMD_MSG, output, espresponse);
                    response = false;
                }
            }
        }
        break;

luc-github, 2019-01-10:
the lib you mention is just a wrapper - right ?
Is it the one mentioned here ? :https://github.com/espressif/arduino-esp32/issues/4

shaddow501, 2019-01-11:
I  just searched the web for analogWrite ESP32- this is the info of the lib;

This library provides an analogWrite function polyfill for ESP32 Arduino framework by wrapping the [ledc](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-ledc.c) library.

shaddow501, 2019-01-11:
If you use authentication to enter the web page of the ESP3D is there an option to extend the time till it logs you out?

luc-github, 2019-01-11:
the only way is to do something like check the box for temperature or position or any action before the 3 minutes  time out happen

luc-github, 2019-01-12:
if you increase the session timeout, do not do several login in same time as currently number of active login are limited to 10 for the period defined and only clean when timeout is over

I will add better session management in 3.0, the timeout will be defined in preference and sent at login and should  be able to be disabled  as you suggested

I guess issue can be closed ?

shaddow501, 2019-01-13:
Yes, sure

closed 2019-01-13 by luc-github