luc-github/ESP3D #287: sd card Ender 3

zeppy14, 2019-01-11:
Hi,
I want use ESP3D with my Ender 3.
All work fine, move axe, control  temperature …
But I have a problem with the SD card. I can't read the SD card. I have test two different SD card.
I tried to format in FAT and FAT32… I see no file !!
If I upload a file, the upload indicate 100% but it never finish. After the upload ESP3D is dead and I can see this message in the windows Commands : 
Error 503 :Communication locked!

I have flash my Wemos with the ESP3D firmware 1.0 (https://github.com/luc-github/ESP3D/releases/tag/1.0)
I have a board creality 3d v1.1.2 with the original firmware.
The M115 command give me this information : 
FIRMWARE_NAME:Marlin V1; Sprinter/grbl mashup for gen6 FIRMWARE_URL:http://www.mendel-parts.com PROTOCOL_VERSION:1.0 MACHINE_TYPE:www.creality3d.cn EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000

Can you help me ?

[image]

luc-github, 2019-01-11:
What is the  M20 command output when you have files on SD ?

zeppy14, 2019-01-11:
It look good !
[image]
[image]

luc-github, 2019-01-11:
Why did you closed ? Did you solved issue ?

zeppy14, 2019-01-11:
Sorry, I dont

luc-github, 2019-01-11:
Ok in that case I think issue come from Marlin support in 1.0
I suggest you to use ESP3D 2.0 and ESP3D-WEBUI 2.1 they have lot of fixes for Marlin 
Once in WebUI 2.1 go to preferences to disable `Connection monitoring` or you will be disconnect after 20s

zeppy14, 2019-01-12:
I test theESP3D 2.0 and ESP3D-WEBUI 2.1 ...it is not better

[image]
[image]

luc-github, 2019-01-12:
Ok I just tested on my test system with Marlin 1.1.9 and it works :
[image]

Look like the M20 on your system does not give file size so file name is not found as not matching the expected patern
Can you try the enclosed webUI and feedback 
[index.html.gz](https://github.com/luc-github/ESP3D/files/2751877/index.html.gz)

zeppy14, 2019-01-12:
Hi Luc, 
It look better but not good ... I see the file, I can do nothing with : no print, no delete, no updload : 
[image]

luc-github, 2019-01-12:
Ok lets check what your FW is sensitive to and able to do
The upload will be for the end as it is the more complex
try delete using command and let see which one is working:
M30 /TEST.GCO
M30 TEST.GCO

luc-github, 2019-01-12:
for printing try 
M23 TEST.GCO 
then
M24

zeppy14, 2019-01-12:
M30 /TEST.GCO
M30 TEST.GCO

Don't work => Deletion(bin) failed.

M23 TEST.GCO
M24

Do nothing.

What do you mind ? It is  better to update my Marlyn ?

luc-github, 2019-01-12:
Well I do not have ender3 so I can just comment on your feedback 
if M30 do not work with error message it means issue with SD file system  as message state command exists but failed - so I would format SD using proper tools instead of PC : https://www.sdcard.org/downloads/formatter_4/

about printing command if M23  which select file and M24 which launch print , these commands are in theory present if I read well : https://github.com/Creality3DPrinting/Ender-3
But I am not sure why it does not work on your printer as you can see launch manually seems doing nothing or you do not have the open source version of the FW

Upgrading to Marlin is more on you - I do not use Marlin as none of my Printers use Marlin, I just have it on test boards, so I cannot comment what is the best for your Printer. I suggest you to ask on Forum what would be the benefit and also the possible drawback.

Bon Courage ^_^

zeppy14, 2019-01-13:
Ok, I have format my SD card with SD Card Formatter : No change 
Thank for our help ... perhaps I will use octoprint but the raspeberry hasn't same cost that a esp8866. 

Merci,

luc-github, 2019-01-13:
No problem - as your FW does not support the basic GCODE commands there is not much ESP3D can do sorry

closed 2019-01-13 by luc-github

zeppy14, 2019-01-15:
Hi I upgrade my FW to marlin 1.1.19.
Thanks

IInickII, 2019-01-16:
Hi, I made same try yesterday and with same result. Is there any chance to make it working?

luc-github, 2019-01-16:
Ender FW does not provide the SD commands there is no choice but change ender FW

IInickII, 2019-01-16:
Thank you for information, do you have some link to guide how to do it on Ender 3?

zeppy14, 2019-01-16:
Https://youtu.be/p5rUR8Skp1c
Sorry it IS un french.

IInickII, 2019-01-16:
Thank you, I will check it for inspiration. Do you know how I can check actual firmware version in my printer?

zeppy14, 2019-01-16:
with the command M115

zeppy14, 2019-01-16:
Finally, I will use another solution because the transfert of the SD Card are to slow. 
See the issue 290.

IInickII, 2019-01-16:
Can you give me more info? I am very interested in this topic or can I contact you in any other way? Thank you

zeppy14, 2019-01-16:
@IInickII 
Look at this octoprint.org