luc-github/ESP3D #288: ESP32 flash issue

davidcgu, 2019-01-13:
Good morning
I tried to flash latest version on a ESP32 dev  and can not compile, I get errors with all related to WIFI_MODEM_SLEEP.

I tried to comment all those lines and flash anyway however afterwards then I can not configure to detect my home network, on setup process ask me to set the language and then the other tabs are moving forward but not showing anything else what makes me impossible to configure my network.

I tried also to set the SSID and PASS manually on arduino config, while doing this I need to set the default mode to station instead of access point to force to connect to my wifi, however I tried to change line #define DEFAULT_WIFI_MODE      AP_MODE  to #define DEFAULT_WIFI_MODE      STA_MODE but then tells me this is not correct and I have no more ideas.

Any advice is welcome.

Thanks in advance

luc-github, 2019-01-13:
what branch of ESP3D do you use ?
what core ESP32 version do you use?

davidcgu, 2019-01-13:
CP2102 ESP-32/ ESP-32S
ESP3D-1,0
Arduino 1.8.5

luc-github, 2019-01-13:
Please  use git version https://github.com/luc-github/ESP3D/tree/2.0 as 1.0 as old support for ESP32
For ESP32 core please use git version  https://github.com/espressif/arduino-esp32/tree/master

davidcgu, 2019-01-13:
I don't know what I'm missing....

When downloading version 1.0 I unzip the full folder and inside is a folder called ESP3D  with all the arduino staff however if I download 2.0 from the link you provided me I can not find where is the arduino scketh inside and on the web I can not find any download link for the 2.0 version

luc-github, 2019-01-13:
Please read the read me the installation is different
https://github.com/luc-github/ESP3D/tree/2.0#installation

davidcgu, 2019-01-13:
I tried but did not understand exactly how to proceed and was not succesfull with any attemps, tells to compile but Arduino gives me error with all lines that had WIFI_MODEM_SLEEP and can't move forward.

Can you advice me how to change to default station mode changing line DEFAULT_WIFI_MODE AP_MODE ?

Maybe that can do the trick for me

luc-github, 2019-01-13:
No this is not your issue and compilation work fine. I have just tested i
What is the compilation output ?

davidcgu, 2019-01-13:
Arduino: 1.8.5 (Windows 10), Board: "Node32s, 80MHz, 921600"

In file included from sketch\bridge.h:24:0,

                 from sketch\command.h:24,

                 from sketch\command.cpp:21:

sketch\command.cpp: In static member function 'static bool COMMAND::execute_command(int, String, tpipe, level_authenticate_type)':

config.h:33: error: 'WIFI_PS_MAX_MODEM' was not declared in this scope

 #define WIFI_MODEM_SLEEP WIFI_PS_MAX_MODEM

                          ^

sketch\command.cpp:568:58: note: in expansion of macro 'WIFI_MODEM_SLEEP'

             BRIDGE::print((const char *)CONFIG::intTostr(WIFI_MODEM_SLEEP), output);

                                                          ^

exit status 1
'WIFI_PS_MAX_MODEM' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

luc-github, 2019-01-13:
please enable verbose compilation and give the full output
[image]

davidcgu, 2019-01-13:
where i can download the binaries?

luc-github, 2019-01-13:
There no binaries as there are too many possible boards and options

davidcgu, 2019-01-13:
I've being able to upload V1.0 to an ESP8266 as don't understand how to proceed with V2, I will wait patiently some more detailed installation instructions some day.

Thanks for your time

luc-github, 2019-01-13:
Did you read the instructions In read me ? What do you don t understand in installation steps?

closed 2019-01-28 by luc-github