luc-github/ESP3D #289: SD card pins for ESP32 TTGO T8

neo799, 2019-01-13:
The SD cards pins on the TTGO-T8 V1.7 board seems different than in the esp32 standard library. Is it possible to implement something special for these boards or where can I change that myself. I didn't find anything. Here the link to the board. https://github.com/LilyGO/TTGO-T8-ESP32

luc-github, 2019-01-13:
Currently ESP3D does not support Direct SD - what do you want to do ?
to change pins for ESP32 SD you must specify them at with `SPI.begin(int8_t sck=-1, int8_t miso=-1, int8_t mosi=-1, int8_t ss=-1)`

neo799, 2019-01-13:
Hi, I want to upload my gcodes files to the SD card and then print from there via the serial connection. I'm using a lerdge board and access to the on board sd card is not possible.
Thanks for the fast answer.

luc-github, 2019-01-13:
But how you plan to print file from your esp sd card ?
Currently ESP3D does not have any host capability

neo799, 2019-01-13:
I must have misunderstood something. I went to through the wiki and the install instructions and saw the SD card support for the ESP8266. I I though I can store the Gcode file on the SD and then print via my lerdge or ramps board.

luc-github, 2019-01-13:
The SD card is the printer SD card and print is launched using M23 and M24 GCODE commands
The ESP3D 3.0 will have host capability but I am on begining for ghe moment

neo799, 2019-01-13:
Sounds great. Great work.