luc-github/ESP3D #290: Upload Speed

zeppy14, 2019-01-15:
Hi,
the Upload file to the SD card are very very very slow.
It's work but a 585 kb file, I wait more that 10 mn : it's normal ?

The communication speed is set to 115200. For me it is the max speed for my Ender 3 with a motherboard V1.1.2.

It is possible to improve file transfer rate ?

[image]

luc-github, 2019-01-15:
Yes it is normal
The bottleneck is not the baud rate but the protocol itself

luc-github, 2019-01-15:
This serial transfert is slow like hell

zeppy14, 2019-01-15:
You think use a SD card direct connect at the ESP.
Do ESP3D support this ? have this solution approved ?

zeppy14, 2019-01-15:
Ok thank.
So I close this issue.

closed 2019-01-15 by zeppy14