luc-github/ESP3D #292: Why SD upload is so slow ? How to increase speed ?

luc-github, 2019-01-21:
# Why SD upload is so slow ?
This is due to upload protocol itself using M28/M29 gcode commands, this protocol actually send command by command, verifying command is correct before writing it to SD, this speed is even slower when checksum is used (adding more calculation and time) and when GCODE is converted from text to binary. So increase baud rate will increase transfert a little but not making the upload 'fast'

<H2>This is not related to ESP3D. But to the fact that the protocol is not a file protocol but a command protocol</H2>

# How to increase speed ?
As explained above ESP3D is not the root cause and may not be the solution
* Use WiFi SD card, they are super fast for upload , but use proper one most of them allow only download not upload - I use the Toshiba flash air with this https://github.com/luc-github/Flashair-FM
* Use a board which share the SD card between ESP and Main board like azteeg wifi : http://www.panucatt.com/azteeg_X5_mini_reprap_3d_printer_controller_p/ax5mini.htm
there may have soon some boards allowing same mechanism for most of DIY printer but they are not available yet
* ESP3D 3.0 with ESP32 with SD card reader will act as host, so upload will be faster but file won't be on SD printer card, only on ESP SD card so no SD Print but host like printing, which mean print will be launch from Web browser or from supported touch screen

closed 2019-01-21 by luc-github

kazooless, 2020-08-31:
Hi there. Thank you for your hard work on this software.

I converted a FLSUN QQ-S Pro to Marlin. It has a modified MKS Robin Mini motherboard with an ESP8266 wifi card. With Repetier, the wifi card worked super fast. I could upload gcode to it via Cura and the file would show up on the sdcard file system. It was accessible from the TFT screen and the mobile app.

I mention this because I think that would indicate the hardware is capable. If that is the case, is it Marlin that is preventing this type of function?

Thank you,

kazooless

luc-github, 2020-08-31:
the wifi sd upload speed is due to fact that the wifi has direct access to SD
esp board connected by serial  do not have direct access to sd and serial tranfert protocole is not a fast one
you cannot compare both

kazooless, 2020-08-31:
So here is my question, is this because of Marlin? Physically, nothing has changed on my hardware. It is the same motherboard and the same ESP connected the same way. The only thing I changed was the firmware on the motherboard and the ESP. 

So what I am saying is, the hardware setup the way it is, works super fast when Repetier is installed with their own ESP firmware. If they can program it to be fast, why can't the Marlin setup be fast too? 

Does that make sense?

luc-github, 2020-08-31:
you use which esp wifi are you using? which fw does it use?

kazooless, 2020-08-31:
I followed the instructions given on the readme here: 
https://github.com/Foxies-CSTL/Marlin2.0.x-FLSun_QQ-S/tree/FLSUN_QQS-PRO-MULTI/Marlin/Firmwares 

and used the files found here:
https://github.com/Foxies-CSTL/Marlin2.0.x-FLSun_QQ-S/tree/FLSUN_QQS-PRO-MULTI/Marlin/Firmwares/ESP3DV2 

The board expects a particular name which is why they are named such. The MksWifi.bin file uploads and installs your firmware version 2.1.0.b39 and the interface is 2.1b39.

The Original-MksWifi.bin file is the one that works perfectly with Repetier. I don't know what version or from which repository it is found.

luc-github, 2020-08-31:
as you mentioned fw is not public so I do not know what solution they use to get fast upload but there are several solutions , all of them need esp board connected to sd card reader : one is this : https://github.com/ardyesp/ESPWebDAV
and it is not depeendant of printer fw