luc-github/ESP3D #294: Oled in version 2.1

shaddow501, 2019-01-22:
Dear Luc

I have a problem with the OLED graphics, when first load it seems that it write correctly the info, but some times it write and erase a part of the writing and sometimes after a while it draw lines, please have a look at the attached picture.
What can I do to fix it?
[image]

luc-github, 2019-01-22:
I guess you are using ESP3D 2.0 FW  with ESP3D-WEBUI 2.1 not ESP3D 2.1
did you do any code modification ?
what is this green wire ?

shaddow501, 2019-01-22:
Dear Luc

I use 2.1 FW code,  with ESP3D-WebUI 2.0, but strange, in the firmware page it shows me UI:2.0c29 and FW:2.0.0.c23, so I have no idea why it write that I use the 2.0 firmware...

The green wire connected to GOIO 17 (that allows me to use another Serial).

luc-github, 2019-01-22:
Ok first do not use 2.1 it is writen in big letter
[image]

Now what I see looks communication issue may be some noise - that is why I asked what is your green wire - it may bring some perturbation
beside that I have no idea as I never got such issue

shaddow501, 2019-01-22:
Ok so now basically i use 2.0 firmware with 2.1 webUI, yes I did some code modification about blocking Serial and allow second Serial.

Basically the thinking is to use the Serial TX that comes from the USB to Serial on the printer board and forward it to the RX line on the ESP32 module, when GPIO 12 go high it stop the function of sending the TX command that come from the ESP3D and instead getting from the USB the commands and inject then to the Serial0.
this allows me to be able to use both USB connection and Wireless connection.

luc-github, 2019-01-28:
` myesp3d.process(); ` is the main function, it handle all process webserver / OLED/ etc...
 if it is disabled even OLED will not be updated, actually nothing will work

luc-github, 2019-01-28:
yes if no LCD update - LCD is fine it make sense
there is another location where the LCD is updated but the occurence is not matching what you get 
it is : onWiFiEvent which is triggered if IP is disconnected / reconnected
instead of main function disable only onWiFiEvent onces

shaddow501, 2019-01-28:
Well I see the error of the LCD happen with 1 second interval so it does match this - f (now_oled - last_oled_update > 1000) {
but I dont know where to look if something else send anything to the LCD (without calling the update_lcd) and just fill the lcd buffer, so when calling the update_lcd all of the buffer getting in...

luc-github, 2019-01-28:
Did you read my previous comment ? I wrote where to disable LCD

shaddow501, 2019-01-28:
Where is it in the code??

shaddow501, 2019-01-28:
I am really sorry that your code is complicated, I am trying to help with a bug, the only thing that fixes the issue is disabling  this part:

if ((WiFi.getMode() == WIFI_OFF) || !wifi_config.WiFi_on) OLED_DISPLAY::display_signal(-1);
else OLED_DISPLAY::display_signal(wifi_config.getSignal (WiFi.RSSI ()));
//if line 0 is > 85 refresh
if(OLED_DISPLAY::L0_size >85)OLED_DISPLAY::display_text(OLED_DISPLAY::L0.c_str(), 0, 0, 85);
//if line 1 is > 128 refresh
if(OLED_DISPLAY::L1_size >128) OLED_DISPLAY::display_text(OLED_DISPLAY::L1.c_str(), 0, 16, 128);
//if line 2 is > 128 refresh
if(OLED_DISPLAY::L2_size >128) OLED_DISPLAY::display_text(OLED_DISPLAY::L2.c_str(), 0, 32, 128);
//if line 3 is > 128 refresh
if(OLED_DISPLAY::L3_size >128) OLED_DISPLAY::display_text(OLED_DISPLAY::L3.c_str(), 0, 48, 128);
**OLED_DISPLAY::update_lcd();**

the more I dig into your code the more it get complex!
sorry I am giving up.

closed 2019-01-28 by shaddow501

luc-github, 2019-01-28:
This part is the updatelcd itself 

Updating Line 0 1 2 and 3 of text.   
So disabling it means you have no lcd update. So no issue as no data is sent.
I cannot reproduce your issue it does not mean there no issue but I cannot fix what I cannot reproduce.

shaddow501, 2019-01-28:
Well I wasnt able to go very far everything I have tried was not successful, the only thing that stopped it was the comment  https://github.com/luc-github/ESP3D/issues/294#issuecomment-458102068.
I dont know why!

shaddow501, 2019-01-28:
Update;
It seems when I attach the task to "Core" number 1 I dont see any problems, when the task runs under core 0 it shows the problem of the writing:

void Task2code( void * parameter) {
  for(;;) {
     myesp3d.process();
     TIMERG0.wdt_wprotect=TIMG_WDT_WKEY_VALUE;
     TIMERG0.wdt_feed=1;
     TIMERG0.wdt_wprotect=0;
     }  
  }

 xTaskCreatePinnedToCore(
      Task2code, /* Function to implement the task */
      "Task2", /* Name of the task */
      10000,  /* Stack size in words */
      NULL,  /* Task input parameter */
      1,  /* Priority of the task */
      &Task2,  /* Task handle. */
      1); /* Core where the task should run */

luc-github, 2019-01-28:
`myesp3d.process();` is in loop which is core 1 in sample code

so you mean your code use this `Task2code` attach to core 0  since the begining ?

luc-github, 2019-01-28:
So it is normal I cannot reproduce issue, you never mentioned this...
core 0 is already used by WIFI so you shared a core with several tasks - current Oled code does not support it - this is the origin of your corrupted  data

shaddow501, 2019-01-28:
Well at the beginning all processes were running on core 0. I didnt see anywhere any core assignment...
Was seems to me that core 1 is unused.