luc-github/ESP3D #296: Bug in file upload error handling

A13xund3r, 2019-02-02:
Just experimenting with high baudrate serial transfer which is quite unstable. Noted that after 5 "resend" (or assumed not ok) answers ESP does abort upload, hope this is 5 per line not 5 in total. Issue is when marlin request "Resend: 18", ESP send it, marlin is happy with new data saying "ok", but ESP is sending the same line again instead of next one, suspect line number is not increased after successful resend or successful read is incorrectly interpreted.

```
from ESP				from marlin
M21
					echo:SD card ok
					ok
M110 N0
					ok
M117 Uploading...
					ok
N1 M28 /xyzCal1.gco*86
					echo:Now fresh file: /xyzCal1.gco
					Writing to file: /xyzCal1.gco
					ok
N2 M82 *59
					ok
N3 G21 *57
					ok
N4 G90 *52
					ok
N5 M82 *60
					ok
-----------cut nice and tidy comm----------
N37 G1 X58.107 Y55.934 E3.79049*83
					ok
N38 G1 F2100 E0.79049*62
					Error:checksum mismatch, Last Line: 37
					Resend: 38
					ok
N38 G1 F2100 E0.79049*62
					ok   <- data accepted, should go ahead
N38 G1 F2100 E0.79049*62	<- line 38 again? Missed previous ok?
					Error:Line Number is not Last Line Number+1, Last Line: 38
					Resend: 39
					ok
N38 G1 F2100 E0.79049*62
					Error:Line Number is not Last Line Number+1, Last Line: 38
					Resend: 39
					ok
N38 G1 F2100 E0.79049*62
					Error:Line Number is not Last Line Number+1, Last Line: 38
					Resend: 39
					ok
N38 M29*35
					Error:Line Number is not Last Line Number+1, Last Line: 38
					Resend: 39
					ok
N39 M29*34	<- correct line number with file close command
					Done saving file.
					ok
M30 /xyzCal1.gco
					File deleted:xyzCal1.gco
					ok
M117 SD upload failed
					ok
N39 M29*34	<- garbage from buffer sent?
					Error:Line Number is not Last Line Number+1, Last Line: 39
					Resend: 40
					ok
N40 M29*44	<- finaly correctly interpreted which line is needed
					Error:Line Number is not Last Line Number+1, Last Line: 39
					Resend: 40
					ok
M30 /xyzCal1.gco
					ok										
M117 SD upload failed
					Deletion failed, File: xyzCal1.gco.ok
					ok
```

A13xund3r, 2019-02-03:
Seen this function earlier but from the code was not able to catch if it is reset every new line or not, sorry I'm old assembler guy and feel sick if have to dig in unintuitive language like this. 
Please find attached log file, problem starts from line 507.
Line https://github.com/luc-github/ESP3D/blob/11d50db1b6fd5e4b266899b6d3c6bacf2f02001c/src/webinterface.cpp#L157 and next one are executed every single transmission? 
[log.txt](https://github.com/luc-github/ESP3D/files/2825736/log.txt)

luc-github, 2019-02-03:
yes reset is done here https://github.com/luc-github/ESP3D/blob/11d50db1b6fd5e4b266899b6d3c6bacf2f02001c/src/webinterface.cpp#L177

What version of Marlin do you use ? 
What baud rate do you use ?

A13xund3r, 2019-02-03:
Marlin latest 1,1,9, testing 2Mb with aiming for 8Mb. I've modified your code so default is 2Mb and in general it is working fine except long data transmission.

luc-github, 2019-02-03:
Is that speed in supported specs of ESP8266 ?

A13xund3r, 2019-02-03:
According to https://www.espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf page 20 it support up to 4.5Mb (so I can forget about 8Mb).

> The data transmission speed via UART interfaces reaches 115200 x 40 (4.5 Mbps).

luc-github, 2019-02-03:
Will do that, as now looking at log I should say some response log are missing from it to understand what part failed

luc-github, 2019-02-04:
@A13xund3r I think I found the issue - the response buffer was not reset - so resend was always in response
I have also added ******* to see what printer send as it could be several lines response is in between `*********`
I have also decreased a delay from 50ms to 5ms to fit your baudrate 
I cannot reproduce issue on my side - so please let me know if it fix issue  - log should be more clear in theory

luc-github, 2019-02-04:
Thank you . take your time I am traveling right now.

A13xund3r, 2019-02-04:
Sorry for late report but upload with enabled SPIFF log is painfully slow and I can't find which port is used for TCP log.
In short: perfect, every single error is correctly handled now, for 4348 lines there was 116 retries where 4 times marlin requested second retry, never third. 
Unfortunately as expected real transfer rate is at about 15kb, just over 2MB file uploaded in about 24min. Pointless. At least you have fixed quite important bug. I'm going in to mixing your code with https://github.com/ardyesp/ESPWebDAV

```
Send line N30 G1 X113.108 Y119.671 E2.42988*95
Response:
************
Error:No Checksum with line number, Last Line: 29
Resend: 30
ok

************
Resend detected
Line requested is 30
Purge Serial
Purge done
Resend 1
Response:
************
ok

************
Got ok
Purge Serial
Purge done
Empy line
Send line G1 X112.878 Y119.923 E2.44123  NB:31
Purge Serial
Purge done
Send line N31 G1 X112.878 Y119.923 E2.44123*86
Response:
************
ok

************
Got ok
```

luc-github, 2019-02-05:
Great that is good news that issue is fixed
yes serial upload is damned slow https://github.com/luc-github/ESP3D/issues/292
Yes I know this project 

I guess issue can be closed now

luc-github, 2019-02-05:
FYI: Looks like a board allowing SD sharing from panucatt will be available soon after Chinese New Year

closed 2019-02-05 by luc-github

luc-github, 2019-02-05:
you mean you get same as https://github.com/luc-github/ESP3D/issues/297 when you print from SD ?
Did you compare the 2 files to see if there is a difference ? in theory the checksum should be enough to check integrity
Currently there is no way to abort the upload, is it 100% handled by browser itself- it is planned in 3.0 to get better control but it is not done yet
The pop up message should happen on which criteria per your opinion ? you said you got in average 1 transfert issue very 37 lines, is reducing speed would really solve the issue ?
Well the board is for people like me who do not soldering ;) there are of always others solutions.

Yes knowning if print in progress is ongoing is on plan but not only for file upload - also jog command and even restarting ESP may reset some printer boards if connected ESP is connected to Serial0
today there is a 'lock UI' option in preferences but I far from what it should,
also sending M27 before each command is little bit overkill - I still do not know how to handle this properly - as currently I mostly rely on User ...

A13xund3r, 2019-02-05:
Yes, something like that, I have only temperature monitoring so sometime query is distorted apparently.
Not compared yet, ESP is stripping empty lines and comments out so I need to learn how to use diff under linux for that purpose, usually i use compare dirs feature from total commander.
IMHO popup should be displayed instantly on first error as this clearly indicating unreliable comm line, we don't know how many lines accidentally got corrupted data with correct checksum. 
In regards speed reduction, not sure how to solve it, Atmel chips have clear rules which baudrate will be fine, even calculators are available, basically crystal divided by 2 and so on will be fine, no idea how function ESP baud clock generator, suspect it should be multiplication of 115200, also ESP clock selection possibly does affect it.
Another thing,  #297 failing at 115200, this mean something else is talking on the same pins, too much load on ESP TX line dragging signal too low or chip receiving it (don't know this Prusa controller board) is 5V and not dealing with 3.3V signal too well, considering last option in my case too as never had issue with data from marlin to ESP.

> Well the board is for people like me who do not soldering

You are absolutely right here, however this will require quite simple PCB with just ESP and SD card socket (micro or mini), it should be possible to offer such PCBs or ready made module for people not familiar with PCB etching. 
Not sure got it right about problem with M27, it is needed only if we want access SD card right? And we really don't want to interrupt if printing is in progress so better if we use it, but why 

> sending M27 before each command

luc-github, 2019-02-05:
here the diff tool I use on linux and window : http://meldmerge.org/ it is easy and powerful

Thank you for the popup suggestion - it is really a good one, you are right, at first issue it should be sent - I will work on it, I will also test again how to properly cancel an upload

Actually for any reason you can press home button on web interface or any jog button so if you are printing this will bring a big mess - so file commands are not the only ones that could cause problem when printing

A13xund3r, 2019-02-05:
Thank you for meldmerge, will use it and report here.
Which upload you want to cancel? 
With error case? just change max error number to 1. 
Just on user request? use same way as in error handling, close file with M29 and checksum in data payload then delete file with M30.

luc-github, 2019-02-05:
actually cancel upload method was not always working when I tested a while ago- may be I saw issue with ESP32 or asynwebserver  so if `client().stopAll();` is working well now on esp8266 - it is good news, I have hard time to test everything each time there is an ESP core update.

Issue I meet was not on Printer / SD, but with upload method itself- even using close method  the next packet was opening a new client connection and so continue to receive packet. Anyway with 3.0 I am rewriting/ testing every part and upload is not yet done - so I will implement all of these -thanks to your feedback and suggestions

And yes I want also to allow user cancel if user uploaded wrong file for any reason and may not want upload to finish which on serial upload can take lot of time ^_^

luc-github, 2019-02-06:
I did a quick python script to calculate check sum:
```
('line 1 :', 'G1 X110.602 Y112.112 E362.44809')
('checksum 1 :', 62)
('line 2 :', 'G1 X110.602 Y112.2 E362.44809')
('checksum 2 :', 62)
***********************
('line 1 :', 'G1 X110.079 Y97 E363.32694')
('checksum 1 :', 30)
('line 2 :', 'G1 X0.079 Y97 E363.32694')
('checksum 2 :', 30)
***********************
('line 1 :', 'G1 X83.485 Y102.46 E365.83377')
('checksum 1 :', 55)
('line 2 :', 'G1 X83.485 Y102.46 E365.877')
('checksum 2 :', 55)
***********************
('line 1 :', 'G1 X92.865 Y119.835 E370.99708')
('checksum 1 :', 0)
('line 2 :', 'G1 X92.865 Y9.835 E370.99708')
('checksum 2 :', 0)
***********************
('line 1 :', 'G0 F6000 X92.616 Y113.898')
('checksum 1 :', 38)
('line 2 :', 'G0 F60 X92.616 Y113.898')
('checksum 2 :', 38)
***********************
('line 1 :', 'G1 F2000 X107.369 Y110.991 E374.68289')
('checksum 1 :', 94)
('line 2 :', 'G1 F2000 X107.369 Y110.1 E374.68289')
('checksum 2 :', 94)
***********************
('line 1 :', 'G1 X111.115 Y85.688 E375.87264')
('checksum 1 :', 13)
('line 2 :', 'G1 X1.115 Y85.688 E375.87264')
('checksum 2 :', 13)
***********************
('line 1 :', 'G1 F2400 X78.388 Y118.719 E532.65561')
('checksum 1 :', 98)
('line 2 :', 'G1 F2400 X78.388 Y118.719 E532.661')
('checksum 2 :', 98)
***********************
('line 1 :', 'G1 F3000 X80.324 Y98.149 E540.39121')
('checksum 1 :', 80)
('line 2 :', 'G1 F30 X80.324 Y98.149 E540.39121')
('checksum 2 :', 80)
***********************
('line 1 :', 'G1 F2000 X87.604 Y107.399 E548.78299')
('checksum 1 :', 100)
('line 2 :', 'G1 F2000 X87.604 Y107.399 E548.782')
('checksum 2 :', 100)
```
for different line we actually get same check sum -  not really reliable...

A13xund3r, 2019-02-06:
That's obvious. 2 bytes cannod represent all combinations of 16 bytes. If there is second checksum calculated from same bits in all bytes (vertical) it would catch more errors but stll not all and without information where wrong data is.

luc-github, 2019-02-07:
not only but also all others FW repetier, smoothiware, Marlin Kimbra, etc, and also all hosts : repetier host, pronterface, etc...
it is big change