luc-github/ESP3D #297: Printing Stops Because of Arbitrary Character/Newline

AustinSaintAubin, 2019-02-03:
Prusa i3 MK3 stops printing because of `echo:Unknown command: ""(2)` or something else unsee. 
Really need verbose logging, [like to a syslog server](https://github.com/arcao/Syslog).

Able to get the printer running by some valid gcode.
Here I am turning off and on my light strip connect to the printer.
Notice the "echo:Unknown command: ""(2)".
```
[ESP700]/gcode/light-off.gcode
echo:Unknown command: ""(2)
ok
ok
ok
[ESP700]/gcode/light-on.gcode
ok
ok
ok
NORMAL MODE: Percent done: 73; print time remaining in mins: 75
SILENT MODE: Percent done: 73; print time remaining in mins: 76
```

Also, just to validate.
They light on/off code is good, I ran it again later.
```
[ESP700]/gcode/light-off.gcode
ok
ok
ok
[ESP700]/gcode/light-on.gcode
ok
ok
ok
```

AustinSaintAubin, 2019-02-05:
[Geekcreit® DT-06 ESP-M2](https://www.banggood.com/Geekcreit-DT-06-Wireless-WiFi-Serial-Port-Transparent-Transmission-Module-TTL-To-WiFi-p-1141047.html?cur_warehouse=CN)
Connected to headers directly (see pic).
Baudrate: 115200

[image]

luc-github, 2019-02-05:
Also I see it is an esp8285 modified with 1MB flash - I did not test on such MCU since a while may be there are some compatibility issue in core

Can you confirm :
* not doing any query  in webUI you have the issue when printing,
* your printer FW stop printing for any `Unknown command:` 
* what is your printer FW and FW version

AustinSaintAubin, 2019-02-07:
Looks like there might be a

Discovered a major bug with the [Prusa Firmware v3.5.1-1778](http://prusa3dprinters.vshcdn.net/downloads/firmware/prusa3d_fw_3_5_1_MK3.zip)
Tested by connecting to ESP3D and sending commands over telnet.

Any one of the following commands will halt the printer.
```
{ perl -e' print "\x03"'; sleep 8; } | telnet esp3d 8888
{ perl -e' print "\x00"'; sleep 8; } | telnet esp3d 8888
{ perl -e' print "\x10"'; sleep 8; } | telnet esp3d 8888
{ perl -e' print "just some simple-text with no newline at the end"'; sleep 8; } | telnet esp3d 8888
```
-------

It seems that the printer firmware will halt/wait to parse a statement until it has received the newline '\n' character.

Any of the following statements work without issue / will resume printing because they send the newline character.
```
{ perl -e' print "\n"'; sleep 8; } | telnet esp3d 8888
{ perl -e' print "\x03\n"'; sleep 8; } | telnet esp3d 8888 
{ perl -e' print "just some simple-text with a newline char at the end\n"'; sleep 8; } | telnet esp3d 8888
```
-------

My best guess is that ESP3D is sending a statement at some point that does not have a new line... or there is some noise on the serial connection that is being interpreted as input by the printer.

Posting this issue to Prusa support. 
Thanks for your help.

luc-github, 2019-02-07:
Man you are not lucky with your printer. You always found strange issue.
I will check again but in theory all comnand have an ending \n to be catched by printer FW. Only GRBL accept single char command without \n

AustinSaintAubin, 2019-02-07:
> Man you are not lucky with your printer. You always found strange issue.

Just my lot in life...

luc-github, 2019-02-07:
Positive thinking : you participate a lot in improvement :)

A13xund3r, 2019-02-12:
Unknown command issue can be resolved if your firmware does have debug command like M111 from Marlin http://marlinfw.org/docs/gcode/M111.html M111 S7 does the trick.
To confirm comm bug put 8s wait between text and \n. If printer return to work after 8s, you found critical bug in firmware.

AustinSaintAubin, 2019-02-12:
@A13xund3r thanks for your feedback.
Can you explain in a little more detail.
I don’t fully understand how I can use ‘M111 S7’ to debug the issue / confirm a bug.

A13xund3r, 2019-02-12:
It was about unknown command error
M111 S7 in marlin enables debug mode. Particularly it activates bit 1,2 and 3. 
1 is echo mode, so it will repeat anything it will receive on serial port, in case of unknown command you will see what it was.
2 will set marlin to be more verbose
3 is giving some extra info on errors
But again, prusa firmware must support this command.

AustinSaintAubin, 2019-02-15:
Same results after entering `M111 S7`.
Ran commend `{ perl -e' print "just some simple-text with no newline at the end"'; sleep 8; } | telnet esp3d 8888`
printing halted/froze. Received no output.

Ran command `{ perl -e' print "just some simple-text with no newline at the end"'; sleep 8; } | telnet esp3d 8888`
Printing resumed and received the message:
`echo:Unknown command: "just some simple-text with no newline at the endjust some simple-text with no newline at the en"(2)
ok`

----------

Running latest firmware to-date: v3.5.2-1999

A13xund3r, 2019-02-15:
There is some misunderstanding, from your initial post I see 2 issues: 
1. unknown command, M111 can help diagnose it as it will return echo with command it received. If it will report some garbage this will mean you have hardware issue on data lines.
2. stop printing possibly due to bug in printer firmware, I don't see new line \n in second command, is it mistype and there was \n sent?  If yes then it seems like serious bug in your printer firmware as you suspected.
And finally you just confirmed M111 is supported in prusa firmware.

luc-github, 2019-03-02:
@AustinSaintAubin any update ?

AustinSaintAubin, 2019-03-03:
@A13xund3r 
  1. Ran a few print starting with `M111 S7`. Did not see any garbage in the console.
  2. You are correct, the 2nd run command DID NOT contain a newline statement. Likely bug in the printer firmware. 

@luc-github I installed [ESP-Link](https://github.com/jeelabs/esp-link) on my [Geekcreit® DT-06 ESP-M2](https://www.banggood.com/Geekcreit-DT-06-Wireless-WiFi-Serial-Port-Transparent-Transmission-Module-TTL-To-WiFi-p-1141047.html?cur_warehouse=CN).
Connecting to printer though ESP-Link with [Pronterface](http://www.pronterface.com/) v1.6.0.
[image]

So far no issues of printer randomly stopping. But still have some longer prints to do.

luc-github, 2019-03-09:
@AustinSaintAubin well I have no clue - may be ESP3D asks too much resource to ESP-M2 (ESP8285) vs ESP-Link

AustinSaintAubin, 2019-03-11:
@luc-github I have been stumped as well.... going to do some more debugging as I have time.

luc-github, 2019-05-12:
@AustinSaintAubin I close issue - feel free to re-open if any update

closed 2019-05-12 by luc-github