luc-github/ESP3D #298: ESP3D 2.0 Cant compile

ffnull, 2019-02-04:
Hello. Im do all by README file. But it whant compile. Here is error log

Archiving built core (caching) in: C:\Users\ffnul\AppData\Local\Temp\arduino_cache_791938\core\core_esp32_esp32_espea32_FlashFreq_40,UploadSpeed_921600_4e960e16d97a7edd219f23bd68dae1e2.a
Linking everything together...
"C:\Users\ffnul\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-gcc" -nostdlib "-LC:\Users\ffnul\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/lib" "-LC:\Users\ffnul\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/ld" -T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,--undefined=uxTopUsedPriority  -u __cxa_guard_dummy -u __cxx_fatal_exception  -Wl,--start-group "C:\Users\ffnul\AppData\Local\Temp\arduino_build_36385\sketch\basicesp3d.ino.cpp.o" "C:\Users\ffnul\AppData\Local\Temp\arduino_build_36385/arduino.ar" -lgcc -lopenssl -lbtdm_app -lfatfs -lwps -lcoexist -lwear_levelling -lesp_http_client -lhal -lnewlib -ldriver -lbootloader_support -lpp -lmesh -lsmartconfig -ljsmn -lwpa -lethernet -lphy -lapp_trace -lconsole -lulp -lwpa_supplicant -lfreertos -lbt -lmicro-ecc -lcxx -lxtensa-debug-module -lmdns -lvfs -lsoc -lcore -lsdmmc -lcoap -ltcpip_adapter -lc_nano -lesp-tls -lrtc -lspi_flash -lwpa2 -lesp32 -lapp_update -lnghttp -lspiffs -lespnow -lnvs_flash -lesp_adc_cal -llog -lsmartconfig_ack -lexpat -lm -lc -lheap -lmbedtls -llwip -lnet80211 -lpthread -ljson  -lstdc++ -Wl,--end-group -Wl,-EL -o "C:\Users\ffnul\AppData\Local\Temp\arduino_build_36385/basicesp3d.ino.elf"
C:\Users\ffnul\AppData\Local\Temp\arduino_build_36385\sketch\basicesp3d.ino.cpp.o:(.literal._Z5setupv+0x8): undefined reference to `Esp3D::begin(unsigned short, unsigned short)'

C:\Users\ffnul\AppData\Local\Temp\arduino_build_36385\sketch\basicesp3d.ino.cpp.o:(.literal._Z4loopv+0x0): undefined reference to `Esp3D::process()'

C:\Users\ffnul\AppData\Local\Temp\arduino_build_36385\sketch\basicesp3d.ino.cpp.o:(.literal.startup._GLOBAL__sub_I_myesp3d+0x4): undefined reference to `Esp3D::Esp3D()'

C:\Users\ffnul\AppData\Local\Temp\arduino_build_36385\sketch\basicesp3d.ino.cpp.o: In function `setup()':

C:\Users\ffnul\AppData\Local\Temp\arduino_modified_sketch_668803/basicesp3d.ino:29: undefined reference to `Esp3D::begin(unsigned short, unsigned short)'

C:\Users\ffnul\AppData\Local\Temp\arduino_build_36385\sketch\basicesp3d.ino.cpp.o: In function `loop()':

C:\Users\ffnul\AppData\Local\Temp\arduino_modified_sketch_668803/basicesp3d.ino:35: undefined reference to `Esp3D::process()'

C:\Users\ffnul\AppData\Local\Temp\arduino_build_36385\sketch\basicesp3d.ino.cpp.o: In function `_GLOBAL__sub_I_myesp3d':

C:\Users\ffnul\AppData\Local\Temp\arduino_modified_sketch_668803/basicesp3d.ino:24: undefined reference to `Esp3D::Esp3D()'

collect2.exe: error: ld returned 1 exit status

exit status 1
Ошибка компиляции для платы ESPea32.

luc-github, 2019-02-05:
Where did you copy esp3d.cpp file and others ?

ffnull, 2019-02-05:
Src folder are in liblary folder

luc-github, 2019-02-05:
You copied src folder only or all project directory ?

ffnull, 2019-02-05:
fixed. Im copy all content from src to examples folder. Then copy from ESP3D library to libray folder. And it compiles. Please fix readme file. Thnx

luc-github, 2019-02-05:
Nope, readme is fine, and esp3d directory must be placed in library directory,
Putting src content is another way to make it work, but then, esp3d is no more a library, travis is also compiling in that way and it works
So the issue is on your side -  some file missing or wrong location, I cannot say.

closed 2019-02-05 by luc-github