luc-github/ESP3D #299: Printrbot Printrhub ESP8266 Touch Screen Controller Hardware Project

drphil3d, 2019-02-12:
Hello, I'm looking to get some feedback on our project.
Essentially we are working to make the Printrbot Simple Pro hardware more accessible and useful by everyone who owns one or wants to use the hardware for their own printer (all the gerber files are published). The Printer is controlled by an ESP8266 touch screen on the front of the machine which is connects via ribbon cable to a power board that supplies the esp with always on power and controls relays for powering on the main board. The screen talks to the mainboard over Uart.

More information here
http://www.appfruits.com/2016/11/printrbot-simple-2016-hardware-pcb-explained/
http://www.appfruits.com/2016/11/printrbot-simple-2016-commstack-explained/
[image]

The main board is modular and can be used in multiple configurations.
You have the option of a ribbon cable from the main board to the printhead or an adapter to plug stepper motors, thermistors, fans in locally. The Mainboard controller is a SAM3XC 32bit processor with drv8825 the board costs about $33 to manufacture.

**More information about the Simple Pro hardware and how it works**
The Simple Pro shipped with g2core firmware which for 3D printing is frustrating to use. The Simple Pro has a touch screen lcd called the Printrhub which was a cloud based controller. The printrbot cloud server was taken offline about 4 months ago. It worked by sending the cloud an stl file (3mb max) to slice and it would then download the gcode to the printer where you could go and start the print. It was a bit ahead of its time and was missing the ability to just send a gcode file locally. Unfortunately due to those major shortcoming It was not well received by the 3d printing community.
https://github.com/Printrbot/printrboardmodernmarlin/wiki/Printrboard-G2-164v3c-Marlin-&-Klipper-Support

After Printrbot shutdown last year a some of us have started working on bringing marlin 2.0 and Klipper to the Printrboard G2. Now that it's pretty close to being complete were looking at the feasibility of replacing the display software with ESP3D and would greatly benefit from some more eyes looking at what we're trying to accomplish. Thank You! 

Here is our thread.
https://github.com/MarlinFirmware/Marlin/issues/13020#issuecomment-462596910

luc-github, 2019-02-12:
hello - I see more details than last PM you sent me and I answered
so here my comments: 
1 - current usage looks what ESP3D is designed to work so for Marlin it should work out of the box, for Klipper I cannot say because I never used it - if use same GCODE commands and same format of GCODE answer as Marlin it should work 
2 - the access of printrbot cloud should be added to ESP3D-WEBUI ans I see no issue based on current information
3 - just one comment about transfering 3MB files using Serial : protocol is slow like hell and some users of ESP3D moved to another solution (octoprint) when they experienced that, but it is not an ESP3D issue it is the normal speed of M28/M29 protocol and using high baud rate does not solve the issue but my even bring some data corruption, a shared access to SD like on Azteeg Wifi would be better IMHO

So my answer will same as last time : Not sure how I can help on this, but let me know.

luc-github, 2019-02-12:
Additional comment for Klipper :
based on https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md, 
the SD transfert command M28/M29 and File delete M30 are not supported in klipper, so ESP3D Serial SD file transfert won't work

luc-github, 2019-02-16:
@drphil3d any update/request ?

luc-github, 2019-02-22:
No update/request so I guess everything is fine, I close issue then

closed 2019-02-22 by luc-github