luc-github/ESP3D #300: Problem with Communicating with the Printer

shaddow501, 2019-02-13:
Dear Luc

Software used is V.2.0.

I am having problems with communicating with the printer.
I use Anycubic i3 Mega printer that I have soldered out the RX / TX lines.

When Connecting USB to TTL as like the https://www.ebay.com/itm/CP2102-USB-2-0-to-TTL-UART-Module-6Pin-Serial-Converter-STC-Replace-FT232-Module-/381374541932

Connect from the CP2102 RX to TX in the printer and the CP2102 TX to the RX in the printer, open Repeter and no issues with communicating with the printer and printing.

When connected with the ESP32 lolin module - RX from lolin go to TX from the printer & TX from lolin goes to the RX on the printer I have strange behavior.

When I ask for a moment (to move the bed for example)  trough the lolin ESP3D website it happens a few seconds after I was asking the command.

When I ask for temperature monitor I dont see anything in the graphs (graphs stay empty) also stay empty when asking for refresh command to the SD card, even when files are exist I cannot see them.

Also reading the printer setting doesnt work. (stay empty)

The firmware chosen is Marlin  (this is the firmware Anycubic also uses).

first I thought that it might be the leveling of the signals on the RX TX lines (on ESP32 it is 3.3V and on the anycubic board it is 5V) So I have added logic level converter on both of those lines, but it didnt solve the issue.

next I thought since I have made some changes in your code I will burn the original code (without any changes) and give it a try, but no it is not the problem.

Then I thought that might be that my ESP32 lolin RX / TX lines maybe have been damaged so I have tested them with other code and verify that my hardware is working ok.

at the moment I dont know where the problem is, any thoughts ?

luc-github, 2019-02-13:
does sending commands by Web UI terminal works ? Like M115 or M20 and [ESP800]
does sending commands by telnet works ? Like M115 or M20  and [ESP800]

shaddow501, 2019-02-13:
I will give you answers later when I will be next to the printer.

shaddow501, 2019-02-13:
Dear Luc

On Repetier host while connected with the USB TO TTL  I get for M20 - list of the SD card files 
M115 give me the firmware version that is - FIRMWARE_NAME:Marlin 1.1.0-RC8

On the Web UI I get:
M20
M115
[ESP800]
FW version:2.0.0.c23 # FW target:repetier # FW HW:Serial SD # primary sd:none # secondary sd:none # authentication:yes # webcommunication:Sync:81 # hostname:Anycubic I3 Mega

It seems that it is not catching the info from the printer.

luc-github, 2019-02-13:
Yes looks like - what about telnet ? 
what serial port of ESP32 do you use ? What baud rate ?

Note : will not solve communication issue but you set repetier instead of Marlin for FW in ESP3D

shaddow501, 2019-02-14:
I dont know the API for telnet commands.. so didnt try it.
I am using the default Serial 0 (GPIO1 ,GPIO3) at 250000 bandwidth.

I did try before setting repetier but it pretty much worked the same.

By default everything like "ack" that the printer send I should see in the Web UI terminal ?

luc-github, 2019-02-14:
You can use putty on window and just connect using your esp IP:8888
This act as remote terminal. As it is just a bridge it remove webui to be a root cause
Yes you should see everthing in webui terminal

luc-github, 2019-02-14:
Your issue looks like this one 
https://github.com/luc-github/ESP3D/issues/269

shaddow501, 2019-02-18:
Dear Luc
I havent got the time to debug it yet I hope soon I will know more, I need to do more tests.

shaddow501, 2019-02-19:
Well this is what I know after today test.

While the ESP32 lolin is off (no power applied).
Using FTDI USB to TTL adapter while the jumper on 5V (sorry I forgot to test it if it works also at 3V) connected to the same lines of the TX/RX/GND and using Repetier host, all works and does everything without any errors.
Disconnecting from Repetier host and open COM port terminal and sending commands like G28 / M20 everything works and I get ACK from the printer.

While the ESP32 lolin is ON(power is applied).
Still connected with the FTDI and sending commands via ESP3D  the commands working with a delay (commands like turn on the BED  / Extruder / moving the XYZ)  but do not get any ACK on the terminal of the ESP3D and not getting anything on the COM that is still connected with the FTDI.

Trying to send commands via the FTDI, it works (on and off)  and no ACK command anymore.

While the FTDI is disconnected sending commands via ESP3D to the printer works as like before but still no ACK.

Next I will try to disassemble the level converter and connect directly and see what I will get now...
(the first setup was without the level converter and it didnt works so if you remember I told you that I have added it and still it doesnt work...)

shaddow501, 2019-02-26:
Dear Luc

After spending a long time with this project and couldnt reach any good results I am ending this. I am moving to an old hardware that I bought before based on the EOL C.H.I.P with  Repetier - server (On Linux).
Proven that it works without any issues.

closed 2019-02-26 by shaddow501