luc-github/ESP3D #301: problem with ip in lcd dont show.

ardyno, 2019-02-15:
hi.
i have the arduino with ramps 1.4 . i was istalled the marlin 2.0.x latest . and esp module with esp3d frameware. it work own without the arduino and ramps . but when i need to controle my printer i can't to see any ip in lcd or any axis work or data tx or rx connect. so what is the probleme.
in addition am using the rx (ramps) to tx (esp) and tx (ramps) to rx (esp).then am using the 3.3volts and gnd from other pins like picture.
[image]

luc-github, 2019-02-16:
So to sum up: full communication is not working with Main board

What info display in webUI : 
[image]

Is baudrate identical on Marlin FW and ESP ?
Do you have mainboard disconnected from USB when starting ?

ardyno, 2019-02-16:
this is the status of esp module:
ESP3D Status
Chip ID:   380784
CPU Frequency:   80Mhz
Free memory:   29.57 KB
SDK:   2.1.0(deb1901)
Flash Size:   1.00 MB
Available Size for update:   184.58 KB(Not enough) 
Available Size for SPIFFS:   471.36 KB
Baud rate:   115200
Sleep mode:   Modem
Channel:   11
Phy Mode:   11g
Web port:   80
Data port:   8888
Active Mode:   Access Point (62:01:94:05:CF:70)
SSID:   DILFIN_LASER
Visible:   Yes
Authentication:   WPA/WPA2
Max Connections:   4
DHCP Server:   Started
IP:   192.168.0.1
Gateway:   192.168.0.1
Mask:   255.255.255.0
Connected clients:   1 
70:  F1:A1:CE:5F:D8 192.168.0.100 
Disabled Mode:   Station (60:01:94:05:CF:70) is disabled
Captive portal:   Enabled
SSDP:   Enabled
NetBios:   Enabled
mDNS:   Enabled
Web Update:   Enabled
Pin Recovery:   Disabled
Authentication:   Disabled
Target Firmware:   Marlin
FW version:   1.0

ardyno, 2019-02-16:
i need to ask you if my wirring is correct to 3.3V and GND like picture?
i was put the baudrate in marlin like esp and define the serial port to 1 value,so that is ok or not?

luc-github, 2019-02-16:
ESP3D is responding so yes wiring is OK but:
* ESP frequency is 80MHz should be 160Mhz - change this setting in IDE and reflash 
* Why set Serial to 1? you use Aux1 = Serial 0 in Marlin that would be the main cause of not communication

ardyno, 2019-02-16:
i was trying change the serial port to 0 but not working.
now am changing the esp frequency .i think the frecuency it not probleme

luc-github, 2019-02-17:
I suggest you to verify your setup is working properly by connecting a USB-TTL adaptor on RX/TX and checking with serial terminal you can communicate with your printer.

ardyno, 2019-02-17:
i need to trying the connect by using the arduino uno .so how can i do this.

ardyno, 2019-02-17:
i need the code from arduino frameware and wiring schema with uno to verify the communication rx tx from ramps then i test the rx tx esp . to define my error where is

luc-github, 2019-02-17:
Sorry I use only 
CP 2102 USB to TTL

ardyno, 2019-02-18:
i was trying connection esp by rx tx gnd from arduino uno without any code the i see the output gcode in serial monitor ,it work fine.
but the board of ramp and arduino don't work ,maybe it preparing heatpad .but i don't have the pad ,i have  just arduino mega 2560,ramps1.4 ,driver ,lcd2004,and one nema17 stepper motor in x axis.
when i start the power it give in lcd (X? Y? Z?) ,why?
so how can i controle the printer without other part .

luc-github, 2019-02-18:
I guess `?` means you did not home axis yet so positions are not defined without rotary encoder or keypad only using GCODE commands by serial will work I think

Sorry I am not ramps specialist it is better to ask help on 3d printers forums they will give you more support than you can get here.

ardyno, 2019-02-18:
thank you very much.

luc-github, 2019-02-18:
Or Marlin as it also could be a Marlin configuration issue

luc-github, 2019-02-18:
For AVR like ramps/mega marlin 1.1.9 is better than 2.x I think

ardyno, 2019-02-20:
i have some problem in esp when i reload the file . am not have the probleme an upload the framware but when i upload by using sketch data in ide from arduino .it give:
Arduino : 1.8.5 (Windows 7), Carte : "Generic ESP8266 Module, 80 MHz, 26 MHz, 40MHz, QIO, 115200, 1M (512K SPIFFS), ck, Disabled, None"

[SPIFFS] data   : C:\Users\pc\Downloads\ESP3D-1.0\ESP3D-1.0\esp3d\data
[SPIFFS] size   : 512
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/404.htm

/favicon.ico

/index.html

/index.html.gz

SPIFFS_write error(-10001): File system is full.

error adding file!

SPIFFS Create Failed!
[SPIFFS] data   : C:\Users\pc\Downloads\ESP3D-1.0\ESP3D-1.0\esp3d\data
[SPIFFS] size   : 512
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/404.htm

/favicon.ico

/index.html

/index.html.gz

SPIFFS_write error(-10001): File system is full.

error adding file!

SPIFFS Create Failed!
[SPIFFS] data   : C:\Users\pc\Downloads\ESP3D-1.0\ESP3D-1.0\esp3d\data
[SPIFFS] size   : 512
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/404.htm

/favicon.ico

/index.html

/index.html.gz

SPIFFS_write error(-10001): File system is full.

error adding file!

SPIFFS Create Failed!
[SPIFFS] data   : C:\Users\pc\Downloads\ESP3D-1.0\ESP3D-1.0\esp3d\data
[SPIFFS] size   : 512
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/404.htm

/favicon.ico

/index.html

/index.html.gz

SPIFFS_write error(-10001): File system is full.

error adding file!

SPIFFS Create Failed!
[SPIFFS] data   : C:\Users\pc\Downloads\ESP3D-1.0\ESP3D-1.0\esp3d\data
[SPIFFS] size   : 512
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/404.htm

/favicon.ico

/index.html

/index.html.gz

SPIFFS_write error(-10001): File system is full.

error adding file!

SPIFFS Create Failed!
[SPIFFS] data   : C:\Users\pc\Downloads\ESP3D-1.0\ESP3D-1.0\esp3d\data
[SPIFFS] size   : 512
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/404.htm

/favicon.ico

/index.html

/index.html.gz

SPIFFS_write error(-10001): File system is full.

error adding file!

SPIFFS Create Failed!

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.

last time it work fine .but i need to work with 160mhz cpu because it now just 80mhz like you side.
i was trying flash memory but the probleme it stay. a using the ESP_DOWNLOAD_TOOL_V2.3 .

luc-github, 2019-02-20:
Je passe en Francais ca sera plus simple ;)
pour le SPIFFS pas besoin de l'uploadeur externe - par contre il faut mieux formater le SPIFFS pour avoir un filesystem propre
[ESP710]FORMAT
ensuite aller sur la page par default http://<ip> cela affichera l'espace disponible sur SPIFFS
dernier point : certain ESP01 on un soucis avec le SPIFFS  et je ne me rappel pas si le patch est deja dans ESP8266 https://github.com/luc-github/ESP3D/issues/229

ardyno, 2019-02-20:
oui j'ai passer tout c'est etape mais les problem il reste.
pour upload dans http c'est tres long temps et apres comleté le http pas afficher les dossier.
j'ai preferer upload avec arduino sketch data.

luc-github, 2019-02-20:
C est pour ca qu il faut etre a 160MHz. C est plus rapide
 Si apres un formatage on ne peut toujours mettre a jour le spiffs c est que le chip est un puya. Et il faut un patch dans le core esp8266

ardyno, 2019-02-20:
comment faire un patch pour esp module.

luc-github, 2019-02-20:
a priori le patch est deja dans la version 2.5 du core

luc-github, 2019-02-20:
a priori is faut l'activer avec un define car il est desactive par default
dans ESP.h
```
#ifndef PUYA_SUPPORT
  #define PUYA_SUPPORT 0
#endif
```

ardyno, 2019-02-20:
thank you
i will to try

ardyno, 2019-02-20:
now it work fine
thank you very much

luc-github, 2019-02-20:
Good ! Can we close issue then?

luc-github, 2019-02-22:
I guess yes

closed 2019-02-22 by luc-github